ev3dev-lang-java/ev3dev-lang-java #6: Study the project autogen

jabrena, 2015-10-31:
https://github.com/ev3dev/ev3dev-lang/tree/develop/autogen

jabrena, 2015-11-01:
Local configuration:

``` java
    "java": {
        "files": [
            "../ev3dev-lang-java/ev3dev.java",
            "../ev3dev-lang-java/spec_version.java"
        ],
        "templateDir": "java/templates/"
    },
```

jabrena, 2015-11-01:
https://github.com/ev3dev/ev3dev-lang/issues/114

rhempel, 2015-11-16:
So here's what you need to do - for changes to anything outside the `java` folder, clone the official `ev3dev-lang` repo and issue pull requests against it.
If the `java` code is actually different from `js` (I don't know the difference :-)) then coordinate with the `js` owner - @wasabifan
If this is a new language, I will create a `java` subfolder in the `ev3dev-lang` repo and point it at your repo as a submodule. You control the code - the `master branch` is the latest validated code, you should use `develop` for running changes and ongoing development, then merge into `master` for a new release. We'll update the `ev3dev-lang` repo to point at the new commit in `master`

jabrena, 2015-11-16:
Yes Sir!!!

rhempel, 2015-11-16:
Seems like lots of process, but as we add more language bindings I'd like to keep things organized...

jabrena, 2015-11-16:
Dont't worry. in the office, I use the same kind of procedure. I prefer the processes to be all clean.

jabrena, 2015-11-22:
@rhempel, I did my first pull request to the project.

I am thinking to create a java test project to test the physical references included in clases to a class generated dynamically with the templates.

How to get all methods from a tacho motor?

Cheers

ddemidov, 2015-11-23:
Properties of tacho motor are listed in `spec.json` file: https://github.com/ev3dev/ev3dev-lang/blob/develop/autogen/spec.json#L19.

Here is an example of using the spec to generate C++ methods corresponding to the properties: https://github.com/ev3dev/ev3dev-lang/blob/develop/cpp/templates/generic-get-set.liquid. 

This template generates this section of the C++ header, corresponding to tacho-motor: https://github.com/ev3dev/ev3dev-lang/blob/develop/cpp/ev3dev.h#L586-L834 (as well as methods of other devices).

Same section is used to generate the test report here: https://github.com/ev3dev/ev3dev-lang/blob/develop/cpp/ev3dev.h#L586-L834 with the following template: https://github.com/ev3dev/ev3dev-lang/blob/develop/cpp/templates/generic_report_status.liquid.

jabrena, 2015-11-23:
Many thanks @ddemidov this week, I will try to generate this structure using your example.

jabrena, 2015-12-11:
@rhempel Tomorrow, I will continue with the process to generate a class using autogen system.

jabrena, 2015-12-13:
Command to generate files from autogen:

``` bash
node autogen.js java
```

jabrena, 2015-12-13:
How to update git submodules:

``` bash
git submodule foreach git pull origin master
git pull --recurse-submodules
git submodule update --recursive
```

jabrena, 2015-12-13:
Generated ev3dev version.
I continue with attributes about tacho-motors

jabrena, 2015-12-13:
Reading examples to improve template system for Java Binding:

Devices supported:
https://github.com/WasabiFan/ev3dev-lang-js/blob/master/motors.ts

Sensors:
https://github.com/WasabiFan/ev3dev-lang-js/blob/master/sensors.ts

WasabiFan, 2015-12-13:
> How to update git submodules:

I would recommend not doing it this way if you are running that command on the whole ev3dev-lang repo. It is best to only update your submodule and not the others.

jabrena, 2015-12-13:
Hi @WasabiFan, yes i am executing only the submodule about java:

``` bash
node autogen.js java
```

jabrena, 2015-12-13:
Now, the class: "EV3DevConstants" is useful to be used for the classes in the project. If the project has to release, it is necessary to generate the class in order to check if something changed in spec.json.

closed 2015-12-27 by jabrena