ev3dev-lang-java/ev3dev-lang-java #14: Analyze LeJOS Class design

jabrena, 2015-11-01:


jabrena, 2015-11-01:
The project ev3classes are organized the following main packages:

* lejos.hardware
* lejos.internals
* lejos.remote
* lejos.robotics

rhempel, 2015-11-14:
@jabrena - please try to follow the binding spec that we have introduced in `spec.json` so that the java binding has the same features as the other language bindings will have.

I'm getting pretty good at writing the templates, and if you get the basic `Device` class sorted out, then add a subclass (like `Motor`) and a sample getter and setter written I can do the templates to get you started.

jabrena, 2015-11-14:
Hi mate, 
tomorrow i will update the clases.

Cheers

jabrena, 2015-11-15:
autogen.js change:

``` js
'.java': { start: cStyleAutogenStart, end: cStyleAutogenEnd }
```

jabrena, 2015-11-15:
autoge-list.json changes:

``` json
    "java": {
        "files": [
            "../ev3dev-lang-java/ev3dev.java",
            "../ev3dev-lang-java/spec_version.java"
        ],
        "templateDir": "../ev3dev-lang-java/templates/"
    },
```

jabrena, 2015-11-15:
.gitmodules changes:

``` bash

[submodule "java"]
	path = java
	url = https://github.com/jabrena/ev3dev-lang-java
	branch = master

```

jabrena, 2015-11-15:
Good morning @rhempel I think that I catched your idea.
I am implementing Device.java and Motor.java using your idea from:
https://github.com/rhempel/ev3dev-lang-python/blob/master/ev3dev/core.py

rhempel, 2015-11-15:
Forgive my ignorance, but is Java much different than JavaScript?

Also, to suggest code changes to top level files in `autogen`, please clone [https://github.com/ev3dev/ev3dev-lang] and issue pull requests against it. That way we just merge the changes if they make sense.

You can read about git pull requests here: [https://help.github.com/articles/using-pull-requests/]

closed 2015-11-15 by jabrena

jabrena, 2015-11-15:
Hi @rhempel, this is the issue about autogen analysis.
https://github.com/jabrena/ev3dev-lang-java/issues/6

I run motors in a easy way. If you like, we could use the issue #6 for autogen