ev3dev-lang-java/ev3dev-lang-java #33: OpenCV for Java

jabrena, 2015-11-15:
http://stefanshacks.blogspot.com.es/2015/05/build-opencv-with-java-bindings-on_6.html

jabrena, 2015-11-15:
http://electronsfree.blogspot.com.es/2015/02/opencv-and-java-on-raspberry-pi-2-b.html

jabrena, 2015-11-15:
export JAVA_HOME=/usr/lib/jvm/java-7-openjdk-armel

jabrena, 2015-11-15:
http://salaboy.com/2013/06/14/using-javacv-in-the-raspberry-pi-linux-arm/

jabrena, 2015-11-15:
http://docs.opencv.org/2.4/doc/tutorials/introduction/desktop_java/java_dev_intro.html

jabrena, 2015-11-15:
http://opencv-java-tutorials.readthedocs.org/en/latest/

jabrena, 2015-11-15:
http://www.tutorialspoint.com/java_dip/introduction_to_opencv.htm

jabrena, 2015-11-15:
http://es.slideshare.net/luigidr/introduction-to-open-cv-28728435

jabrena, 2015-11-15:
https://www.packtpub.com/application-development/opencv-java-javacv

jabrena, 2015-11-15:
http://cvgl.stanford.edu/BAVM14/slides/Bradski.pdf

jabrena, 2015-11-15:
http://opencv.org/wp-content/uploads/2013/icvs/ICVS2013%20-%20OpenCV%203.0.pdf

jabrena, 2015-11-15:
http://image2measure.net/files/Mastering_OpenCV.pdf

jabrena, 2015-11-16:
Compiling on EV3Dev.
After 6h. processing, 13% of compilation.

dlech, 2015-11-16:
Tip: Use brickstrap to compile.

https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug

jabrena, 2015-11-16:
@dlech I like so muchs this small details. Many thanks for the link. I will link. I hope to finish the compilation tomorrow... :p 

For next compilation, I will use brickstrap.

Many thanks @dlech 

Cheers

jabrena, 2015-11-16:
First round: KO.
I will repeat with "Brickstrap"

```bash
[ 16%] Built target opencv_highgui_pch_dephelp
Scanning dependencies of target pch_Generate_opencv_highgui
[ 16%] Generating precomp.hpp
[ 16%] Generating precomp.hpp.gch/opencv_highgui_RELEASE.gch
[ 16%] Built target pch_Generate_opencv_highgui
Scanning dependencies of target opencv_highgui
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap_images.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/loadsave.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/utils.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/window.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/window_gtk.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap_v4l.cpp.o
[ 16%] Building CXX object modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap_ffmpeg.cpp.o
In file included from /home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg_impl.hpp:60:0,
                 from /home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg.cpp:45:
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:104:7: error: 'CODEC_ID_H264' was not declared in this scope
     { CODEC_ID_H264, MKTAG('H', '2', '6', '4') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:105:7: error: 'CODEC_ID_H264' was not declared in this scope
     { CODEC_ID_H264, MKTAG('h', '2', '6', '4') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:106:7: error: 'CODEC_ID_H264' was not declared in this scope
     { CODEC_ID_H264, MKTAG('X', '2', '6', '4') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:107:7: error: 'CODEC_ID_H264' was not declared in this scope
     { CODEC_ID_H264, MKTAG('x', '2', '6', '4') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:108:7: error: 'CODEC_ID_H264' was not declared in this scope
     { CODEC_ID_H264, MKTAG('a', 'v', 'c', '1') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:109:7: error: 'CODEC_ID_H264' was not declared in this scope
     { CODEC_ID_H264, MKTAG('V', 'S', 'S', 'H') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:111:7: error: 'CODEC_ID_H263' was not declared in this scope
     { CODEC_ID_H263, MKTAG('H', '2', '6', '3') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:112:7: error: 'CODEC_ID_H263P' was not declared in this scope
     { CODEC_ID_H263P, MKTAG('H', '2', '6', '3') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:113:7: error: 'CODEC_ID_H263I' was not declared in this scope
     { CODEC_ID_H263I, MKTAG('I', '2', '6', '3') }, /* intel h263 */
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:114:7: error: 'CODEC_ID_H261' was not declared in this scope
     { CODEC_ID_H261, MKTAG('H', '2', '6', '1') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:117:7: error: 'CODEC_ID_H263P' was not declared in this scope
     { CODEC_ID_H263P, MKTAG('U', '2', '6', '3') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:118:7: error: 'CODEC_ID_H263P' was not declared in this scope
     { CODEC_ID_H263P, MKTAG('v', 'i', 'v', '1') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:120:7: error: 'CODEC_ID_MPEG4' was not declared in this scope
     { CODEC_ID_MPEG4, MKTAG('F', 'M', 'P', '4') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:121:7: error: 'CODEC_ID_MPEG4' was not declared in this scope
     { CODEC_ID_MPEG4, MKTAG('D', 'I', 'V', 'X') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:122:7: error: 'CODEC_ID_MPEG4' was not declared in this scope
     { CODEC_ID_MPEG4, MKTAG('D', 'X', '5', '0') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:123:7: error: 'CODEC_ID_MPEG4' was not declared in this scope
     { CODEC_ID_MPEG4, MKTAG('X', 'V', 'I', 'D') },
       ^
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:124:7: error: 'CODEC_ID_MPEG4' was not declared in this scope
     { CODEC_ID_MPEG4, MKTAG('M', 'P', '4', 'S') },
[... 334 lines cut ...]
/home/opencv-2.4.10/modules/highgui/src/ffmpeg_codecs.hpp:258:7: error: 'CODEC_ID_NONE' was not declared in this scope
     { CODEC_ID_NONE, 0 },
       ^
In file included from /home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg.cpp:45:0:
/home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg_impl.hpp: In member function 'double CvCapture_FFMPEG::getProperty(int)':
/home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg_impl.hpp:774:33: error: 'AVStream' has no member named 'r_frame_rate'
         return av_q2d(video_st->r_frame_rate);
                                 ^
/home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg_impl.hpp: In member function 'double CvCapture_FFMPEG::get_fps()':
/home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg_impl.hpp:821:49: error: 'AVStream' has no member named 'r_frame_rate'
     double fps = r2d(ic->streams[video_stream]->r_frame_rate);
                                                 ^
In file included from /home/opencv-2.4.10/modules/highgui/src/cap_ffmpeg.cpp:45:0:
[... cut ...]
modules/highgui/CMakeFiles/opencv_highgui.dir/build.make:215: recipe for target 'modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap_ffmpeg.cpp.o' failed
make[2]: *** [modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap_ffmpeg.cpp.o] Error 1
CMakeFiles/Makefile2:1496: recipe for target 'modules/highgui/CMakeFiles/opencv_highgui.dir/all' failed
make[1]: *** [modules/highgui/CMakeFiles/opencv_highgui.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
root@ev3dev:/home/opencv-2.4.10/release# 
```

dlech, 2015-11-16:
Debian already have as `libopencv2.4-java` package. It is 2.4.9. Does this work for you?

    sudo apt-get install libopencv2.4-java

https://packages.debian.org/jessie/libopencv2.4-java

jabrena, 2015-11-16:
Installed!!!

I am trying to find a Hello World example to close the circle... :D 
and... open a new big circle... :D 

Cheers

jabrena, 2015-11-16:
http://docs.opencv.org/2.4/doc/tutorials/introduction/java_eclipse/java_eclipse.html#java-eclipse

jabrena, 2015-11-16:
brew tap homebrew/science
brew install opencv --with-java

jabrena, 2015-11-16:
http://stackoverflow.com/questions/22816777/not-finding-opencv-jar-after-installing-with-homebrew

jabrena, 2015-11-16:
Local output:

```bash
➜  target git:(master) ✗ java -Djava.library.path=/usr/local/Cellar/opencv/2.4.11_1/share/OpenCV/java/ -cp HelloCV.jar:/usr/local/Cellar/opencv/2.4.11_1/share/OpenCV/java/opencv-2411.jar:/usr/local/Cellar/opencv/2.4.11_1/share/OpenCV/java/ Main 
Welcome to OpenCV 2.4.11.0
/usr/local/Cellar/opencv/2.4.11_1/share/OpenCV/java/
m = [1, 0, 0;
  0, 1, 0;
  0, 0, 1]
```

jabrena, 2015-11-16:
Hi @dlech do you know where OpenCV store libraries?

If I find the path for .so, I could execute the example:

``` bash
java -Djava.library.path=/usr/share/OpenCV/java -cp HelloCV2.jar:/usr/share/OpenCV/java/opencv-249.jar Main
Welcome to OpenCV 2.4.9.1
/usr/share/OpenCV/java
Exception in thread "main" java.lang.UnsatisfiedLinkError: no opencv_java249 in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1886)
	at java.lang.Runtime.loadLibrary0(Runtime.java:849)
	at java.lang.System.loadLibrary(System.java:1088)
	at Main.main(Main.java:10)
```

jabrena, 2015-11-16:
I only have to update this parameter:

```bash
-Djava.library.path=/usr/share/OpenCV/java
```

jabrena, 2015-11-16:
Solved:

``` bash
find / -name "libopencv_java249.so"
/usr/lib/jni/libopencv_java249.so
root@ev3dev:/home# java -Djava.library.path=/usr/lib/jni/ -cp HelloCV2.jar:/usr/share/OpenCV/java/opencv-249.jar Main
Welcome to OpenCV 2.4.9.1
/usr/lib/jni/
m = [1, 0, 0;
  0, 1, 0;
  0, 0, 1]
```

jabrena, 2015-11-16:
I run OpenCV for Java on EV3Dev

closed 2015-11-16 by jabrena