ev3dev-lang-java/ev3dev-lang-java #36: [OpenCV] Detect Color Blob

jabrena, 2015-11-16:
https://github.com/Itseez/opencv/blob/master/samples/android/color-blob-detection/src/org/opencv/samples/colorblobdetect/ColorBlobDetector.java

jabrena, 2015-11-16:
http://stackoverflow.com/questions/9129037/javacv-blob-detection

jabrena, 2015-11-16:
https://www.youtube.com/watch?v=bVNYkpH66Hc

jabrena, 2015-11-16:
http://www.learnopencv.com/blob-detection-using-opencv-python-c/

jabrena, 2015-11-16:
http://www.technical-recipes.com/2011/track-colored-objects-in-video-using-opencv-and-cvblobslib/

jabrena, 2015-11-16:
http://m.youtube.com/watch?v=hstdTgiSv8Y

jabrena, 2015-11-16:
http://8a52labs.com/2011/05/24/detecting-blobs-using-cvblobs-library/

jabrena, 2015-11-16:
http://answers.opencv.org/question/23207/trying-to-find-center-of-contour-using-color-blob-detection/

jabrena, 2015-11-16:
https://www.quora.com/How-do-I-track-a-car-using-color-detection-and-blob-detection-What-are-the-steps-I-need-to-follow

jabrena, 2015-11-19:
https://github.com/VMMobility/OpenCV-Color-Blob-Detection/blob/master/OpenCV%20Sample%20-%20color-blob-detection/src/org/opencv/samples/colorblobdetect/ColorBlobDetector.java

jabrena, 2015-11-19:
http://akaifi.github.io/MultiObjectTrackingBasedOnColor/

jabrena, 2015-11-19:
https://code.google.com/p/dronelander/source/browse/src/eu/hrasko/dronelander/BlobDetector.java?r=3a4f2f65e377e44d08d7f7691e5165756d3da6da

jabrena, 2015-11-19:
https://df97e5c8f580f6ac75b80d841f004d5ca19d1b5c.googledrive.com/host/0By56j-W37_RAYS1DcVl3MnFRZGc/index.html

jabrena, 2015-11-20:
https://github.com/AllNamesAreTaken/BlobDetection/tree/master/BlobDetection/src

jabrena, 2015-11-20:
https://github.com/Itseez/opencv/blob/master/samples/android/color-blob-detection/src/org/opencv/samples/colorblobdetect/ColorBlobDetector.java

jabrena, 2015-11-20:
https://github.com/loncaa/Improved-Blob-detection-sample-OpenCV-/blob/master/src/org/opencv/samples/colorblobdetect/ColorBlobDetector.java

jabrena, 2015-11-20:
https://github.com/CSE455ResistorReader/RedYellowDetection/blob/master/app/src/main/java/com/example/abe/resistorreader/MainActivity.java

jabrena, 2015-11-30:
http://cell0907.blogspot.com.es/2013/07/tracking-ball-with-javaopencv.html

jabrena, 2015-11-30:
http://berenoune.blogspot.com.es/2012/10/opencv-android-java-color-detection.html

jabrena, 2015-11-30:
http://berenoune.blogspot.com.es/2012/10/opencv-c-triangle-detection.html

ameysutavani, 2016-02-10:
see votchallenge.net also

jabrena, 2016-12-05:
REJECTED BY LEAN

closed 2016-12-05 by jabrena