ev3dev-lang-java/ev3dev-lang-java #56: Install LeJOS firmware to a RCX brick from EV3 brick with EV3Dev

jabrena, 2015-12-13:
A EV3 Brick with EV3Dev is a full featured Linux OS so, it is necessary to test if it possible to install LeJOS firmware to a RCX Brick from EV3 following this document:

jabrena, 2015-12-13:
http://mindsensors.ddns.net/index.php?module=pagemaster&PAGE_user_op=view_page&PAGE_id=59

jabrena, 2015-12-13:
http://www.lejos.org/p_technologies/nxt/nxj/api/lejos/nxt/RCXLink.html

jabrena, 2015-12-13:
http://www.lejos.org/nxt/nxj/tutorial/Communications/Communications.htm

jabrena, 2015-12-13:
http://www.drvegetable.com/irremote_protocol.html

jabrena, 2015-12-13:
http://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=nil1046

jabrena, 2015-12-13:
http://www.mindsensors.com/ev3-and-nxt/123-pf-motor-controller-for-nxt-or-ev3-pfmate

jabrena, 2015-12-13:
http://blog.hsh.one.pl/?p=210

jabrena, 2015-12-13:
http://www.lejos.org/forum/viewtopic.php?t=7633

jabrena, 2015-12-14:
https://www.cs.kent.ac.uk/pubs/2002/1517/content.pdf
ftp://ftp.oreilly.com/pub/conference/java2001/Knudsen_leJOSpresent.pdf
https://www.usenix.org/legacy/event/usenix2000/freenix/nikander/nikander.pdf
http://scripts.cac.psu.edu/users/r/e/ree5047/research/lejos/lejosProgramming.pdf
ftp://ftp.dca.fee.unicamp.br/pub/docs/gudwin/LEGO%20Mindstorms/Syngress%20-%20Programming%20Lego%20Mindstorms%20with%20Java.pdf
http://willsmc.com/robotics/rcx/rcx.html

jabrena, 2015-12-28:
http://clark.cementhorizon.com/RCX-web-page-2013-09-29.html

jabrena, 2015-12-28:
http://cs.brown.edu/courses/cs148/old/2004fall/brickOS/HOWTO/lnp.html

jabrena, 2015-12-28:
https://minordiscoveries.wordpress.com

jabrena, 2015-12-28:
http://www.extremenxt.com/lego.htm#sensors

jabrena, 2015-12-28:
http://www.mralligator.com/rcx/index.html

jabrena, 2015-12-28:
http://www.philohome.com/sensors/gp2d12.htm

jabrena, 2015-12-28:
http://www.cs.ru.nl/lab/rcx/experiments/1.Proximity/Proximity.html

jabrena, 2015-12-28:
http://www.cs.ru.nl/lab/rcx/

jabrena, 2015-12-28:
http://www.techno-stuff.com/IRRD.htm

jabrena, 2015-12-28:
https://seriousrobotics.wordpress.com/2011/11/20/using-rcx-lego-sensors-with-arduino/

jabrena, 2015-12-28:
http://www.instructables.com/id/Use-LEGO-Mindstorm-RCX-Light-Sensor-with-Arduino-P/?ALLSTEPS

jabrena, 2015-12-28:
http://www.philohome.com/nxtcables/nxtcable.htm

jabrena, 2015-12-28:
http://battle-bot.blogspot.com.es/2012/10/unobot.html

ameysutavani, 2016-02-10:
Hello,
I can confirm that it is definitely possible to load lejos firmware to RCX from raspberry pi(running raspbian jessie). I have tried this with standard lego usb IR tower and also with a self made serial tower (using 555 for generating 38kHz pulse and FTDI cable for usb serial port). I can share the circuit diagram if anyone is interested.

However, I was not able to compile lejos programs (error being not able to locate java.lang classes). I can try and replicate the error again and share console log if anyone wants to try.

As for ev3dev, I think it should yield same outcome as that of raspbian as they are quite similar.

jabrena, 2016-02-10:
Hijo @ameysutavani,

Congratultions! 

If you could test with EV3Dev,  it could be great. In a near future,  I will add mini site to add docs and experiences with this project. If you test it,  we could share the experience with other mates. 

Congrats!! 

Cheers

jabrena, 2016-02-10:
In what parts of the project,  are you interested?

ameysutavani, 2016-02-24:
Hello @jabrena 
Sorry for the late reply, was travelling.

> If you could test with EV3Dev, it could be great.

Currently I don't have access to mu EV3 brick; I will test on it as soon as possible.

> In what parts of the project, are you interested?

No specific interests as such; however I would like to work in area with overlapping ev3 and android smartphones.

jabrena, 2016-02-24:
Yes,  web could play with Android using Akka. We could create a remote control from Android using Akka. Do you like the challenge?

jabrena, 2016-02-25:
Take a look Akka:
http://es.slideshare.net/mobile/remeniuk/first-glance-at-akka-20

jabrena, 2016-02-25:
http://es.slideshare.net/mobile/frodriguezolivera/concurrent-and-distributed-applications-with-akka-java-and-scala-39130140

jabrena, 2016-02-25:
https://www.google.es/url?sa=t&source=web&rct=j&url=https://courses.cs.ut.ee/MTAT.08.009/2013_fall/uploads/Main/2013DS-L010.pdf&ved=0ahUKEwjT0vu2t5LLAhVEPBQKHQ8HAnk4ChAWCDMwCA&usg=AFQjCNEpY2ITumoeXVzUSlXcQl15yrq9VQ&sig2=avjTHXdtE2PkVp3ORrVi0g

jabrena, 2016-02-25:
Hi @ameysutavani, If you like the idea, I will create a new issue about Remote Control.
I am thinking to create a Github organization to create sub repositories for other developments. One repository could be the development of a remote control using Android.

ameysutavani, 2016-04-10:
Hi @jabrena seems like a good idea. I dont have any prior knowledge of Akka though. I am currently exploring different ways to run an opencv app on android (to do slam) and command a robot made of ev3 accordingly. (I know this can be done with lejos's remote api, however i would like to do something similar in c++). If you like the idea, we can explore further.

jabrena, 2016-04-10:
Take a look this issue:
https://github.com/ev3dev-lang-java/remote/issues/1

jabrena, 2016-04-10:
Later, I will test it, but it is a offtopic issue.

jabrena, 2017-03-07:
https://forum.arduino.cc/index.php?topic=370010.0

jabrena, 2017-03-07:
http://phillipecantin.blogspot.co.uk/2012/08/a-new-life-for-rcx-20.html

jabrena, 2017-03-07:
http://www.crynwr.com/lego-robotics/
https://www.cs.duke.edu/courses/fall05/cps097s/cps004/notes/usinglejos.pdf
http://www.mralligator.com/rcx/opcodes.html
http://www.legoengineering.com/wp-content/uploads/2013/06/RCX-constructopedia-beta-1.0.pdf

jabrena, 2017-03-07:
http://www.convict.lu/Jeunes/ultimate_stuff/Making%20your%20own%20RCX%20sensors_part2.pdf

jabrena, 2017-03-07:
http://www.kovan.ceng.metu.edu.tr/mindstorms/

jabrena, 2017-03-19:
https://pbrick.info/2013/10/configuring-the-lego-usb-tower-on-linux/

jabrena, 2017-03-19:
http://www.technicopedia.com/index.html

jabrena, 2017-03-19:
http://www.lejos.org/rcx/tutorial/getstarted/firstbrick/linux.html

jabrena, 2017-03-19:
https://minordiscoveries.files.wordpress.com/2017/01/rpi_rcx.pdf

jabrena, 2017-03-19:
```
robot@ev3dev:~$ ls -l /dev/usb/
total 0
crw-rw-rw- 1 root root 180, 160 Mar 19 20:35 legousbtower0
```

jabrena, 2017-03-19:
```
sudo vi /etc/udev/rules.d/90-legotower.rules
ATTRS{idVendor}=="0694",ATTRS{idProduct}=="0001",MODE="0666",GROUP="robot"
```

jabrena, 2017-03-19:
``` 
robot@ev3dev:~/rcx/lejos$ export LEJOS_HOME=/home/robot/rcx/lejos
robot@ev3dev:~/rcx/lejos$ export PATH=$PATH:$LEJOS_HOME/bin
robot@ev3dev:~/rcx/lejos$ export CLASSPATH=$CLASSPATH:.:$LEJOS_HOME/lib/classes.jar.:$LEJOS_HOME/lib/pcrcxcomm.jar
robot@ev3dev:~/rcx/lejos$ export RCXTTY=/dev/usb/legousbtower0
robot@ev3dev:~/rcx/lejos$ export JAVA_HOME=/usr/lib/jvm/java-9-openjdk-armhf
robot@ev3dev:~/rcx/lejos$  export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$LEJOS_HOME/bin
robot@ev3dev:~/rcx/lejos$ chmod +x /home/robot/rcx/lejos/bin/*
robot@ev3dev:~/rcx/lejos$ chmod +x /home/robot/rcx/lejos/release/*
```

jabrena, 2017-03-19:
```
robot@ev3dev:~/rcx/lejos/bin$  ./firmdl
read firmware srec
  100%
Installing firmware
Loading native libs
Deleting firmware
Firmware deleted
Downloading firmware
  100%
Firmware downloaded
Unlocking firmware
Firmware unlocked
```

jabrena, 2017-03-19:
robot@ev3dev:~/rcx/lejos/bin$ javac -bootclasspath "$LEJOS_HOME/lib/classes.jar:$LEJOS_HOME/lib/rcxcomm.jar" DistributionSmokeTest.java -target 1.6 -source 1.6

jabrena, 2017-03-19:
```
robot@ev3dev:~/rcx/lejos/bin$ lejos DistributionSmokeTest
linking...
downloading...
read binary
  100%
download binary
Loading native libs
download program
  100%
  100%
```

jabrena, 2017-03-19:
Running nice

closed 2017-03-19 by jabrena