ev3dev-lang-java/ev3dev-lang-java #67: Beginners Guide updates

burf2000, 2015-12-28:
Tonight is my first night trying to get EV3Dev and Lejos to play.  Sadly I am not a Linux or Maven expert

So 
First issue

https://jabrena.github.io/ev3dev-lang-java/#/4/5 
0.2 is not out yet and so anyone following this will get stuck

Second issue
I tried pasting the stuff listed on the github page, <repository> etc in to my pom.xml and it did not like it.  I had to actually use one of the examples. (maybe its white space or something)

Third Issue
if you follow the example, your in a package called package ev3dev.examples.demo; (for nice website you have ev3dev.java.examples;)
You have a class called Test

but  your asked to run java -cp MyFirstRobot-1.0-SNAPSHOT.jar:ev3-lang-java-0.2-SNAPSHOT.jar ev3dev.java.MyFirstRobot.Test?

Where did MyFirstRobot come from?

Forth issue
Where is the ev3-lang-java-0.*-SNAPSHOT located, I do not have it on my machine anywhere (looking for 0.1 or 0.2)

Once we have a easy to follow guide I could do a youtube video?

jabrena, 2015-12-28:
Good morning Simon,

One:
yes, the release 0.2 is coming this week, I hope to release and you will see the .jar in https://jitpack.io

Two:
The example is on the repo.
If you download the repo, you could use the example:

``` bash
git clone https://github.com/jabrena/ev3dev-lang-java.git
cd library
mvn package
```

later, move the jar generated to your brick with EV3Dev using a SFTP tool (For Example: Fillezilla) to path: /home

``` bash
ev3-lang-java-0.2-SNAPSHOT.jar
````

To test the example is:

``` bash
java -cp ev3-lang-java-0.2-SNAPSHOT.jar ev3dev.java.examples.Test
```

jabrena, 2015-12-28:
*Third Issue*
Once I release, all examples will be located here:
https://github.com/jabrena/ev3dev-lang-java/tree/master/examples

Not in the own library.

*Forth issue*
When you compile any project with maven, the default folder to store the java artifact (.jar) is ./target folder.

jabrena, 2015-12-28:
I did this document online:
https://jabrena.github.io/ev3dev-lang-java/

But I will create a guide with more detail with Jekill.

burf2000, 2016-01-04:
I will try this out hopefully this week :)

jabrena, 2016-12-05:
REJECTED BY LEAN

closed 2016-12-05 by jabrena