ev3dev-lang-java/ev3dev-lang-java #85: Add support for RPLIDAR, 2D LIDAR

jabrena, 2016-01-09:


jabrena, 2016-01-17:
Related issue:
https://github.com/ev3dev/ev3dev/issues/491

jabrena, 2016-02-11:
Example:
https://github.com/jmccormack200/RPLidar/blob/master/LidarPrint.py

``` bash
robot@ev3dev:~$ python LidarPrint.py 
Traceback (most recent call last):
  File "LidarPrint.py", line 1, in <module>
    import serial
ImportError: No module named serial
```

jabrena, 2016-02-12:
http://stackoverflow.com/questions/12317576/stable-alternative-to-rxtx

jabrena, 2016-02-12:
http://pi4j.com/example/serial.html

jabrena, 2016-02-12:
Running on PiStorms:

``` bash
robot@ev3dev:~$ python LidarPrint.py
```

jabrena, 2016-02-13:
Comparing results from python Script with official C++ driver to compare results:
http://www.slamtec.com/en/Lidar
http://www.slamtec.com/download/lidar/sdk/rplidar_sdk.1.4.5.7z

jabrena, 2016-02-13:
A better Python script:
https://github.com/jabrena/RPLidar/blob/master/LidarPrint2.py

Way to use:

``` bash
python LidarPrint2.py /dev/ttyUSB0
```

jabrena, 2016-02-13:
Example integrating Python from Java:
https://github.com/jabrena/RPLidar/blob/master/PythonTest1.java

``` bash
robot@ev3dev:~$ java PythonTest1 
Primary argument is :  /dev/ttyUSB0
/dev/ttyUSB0
Connecting
...
515, 291
488, 292
490, 293
472, 295
470, 296
470, 297
470, 298
470, 299
472, 300
473, 301
475, 302
477, 303
480, 304
484, 305
485, 306
482, 307
480, 309
480, 310
477, 311
478, 312
476, 313
473, 314
470, 315
469, 316
467, 317
465, 319
462, 319
461, 321
461, 322
460, 323
459, 324
462, 325
276, 328
242, 329
227, 330
213, 332
211, 334
202, 335
199, 336
198, 337
196, 338
191, 339
189, 342
189, 342
189, 342
184, 344
183, 347
183, 347
184, 347
182, 348
180, 350
181, 352
182, 351
180, 354
179, 354
179, 354
180, 357
181, 358
181, 359
182, 1
182, 2
[... 9663 lines cut ...]
879, 141
867, 142
854, 143
842, 144
832, 145
821, 146
810, 147
801, 148
791, 149
784, 150
774, 151
766, 152
759, 154
752, 154
745, 156
739, 157
734, 158
728, 159
721, 160
717, 161
711, 162
708, 163
704, 164
700, 165
```

jabrena, 2016-02-13:
Now, it is necessary to play a bit Java Threads in order to process 360 degrees and returns data on demand.

jabrena, 2016-02-13:
http://winterbe.com/posts/2015/04/07/java8-concurrency-tutorial-thread-executor-examples/

jabrena, 2016-02-13:
https://docs.oracle.com/javase/tutorial/essential/concurrency/simple.html

jabrena, 2016-02-13:
http://winterbe.com/posts/2015/04/30/java8-concurrency-tutorial-synchronized-locks-examples/

jabrena, 2016-02-13:
http://winterbe.com/posts/2015/05/22/java8-concurrency-tutorial-atomic-concurrent-map-examples/

jabrena, 2016-02-13:
http://www.codejava.net/java-core/the-java-language/java-8-lambda-runnable-example

jabrena, 2016-02-13:
http://coopsoft.com/ar/Calamity2Article.html

jabrena, 2016-02-15:
Schema to show data:
https://github.com/jabrena/livingrobots/blob/master/chapter8/www/Radar/amcharts/polar.json

jabrena, 2016-12-05:
REJECTED BY LEAN

sarandaegus, 2016-12-18:
Hi @jabrena, I am Sergio, student of UPC in Catalonia.

I have a problem when i run the"LidarPrint.py"code, it appears :

```
/dev/ttyUSB0
Connecting
...
```

i change only this part in the original code:

```
if __name__ == "__main__":
    #COM4 was used on my computer, this will change based on
    #your setup and whether you're on Windows/Mac/Linux
    #port = '/dev/ttyUSB0'
    #port = 20
    port = "/dev/ttyUSB0"
    ser = serial.Serial(port, 115200, timeout = 5)
    ser.setDTR(False)
    print ser.name
    #Create a Lidar instance, this will immidiately start printing.
    #To edit where the data is sent, edit the GetPoints Method
    lidar = Lidar(ser)
```

sorry for insist, but you are the only person who has worked with python in raspberry, so whatis the problem here?

Sorry but my english, I am from Peru studying a master degree in Spain.

I had to write in the post too: 
[https://github.com/ev3dev/ev3dev/issues/491](url)

Thanks for all

Sergio Aranda

jabrena, 2016-12-20:
Hi @sarandaegus,

I have just read the email. I will write your comments here and on EV3Dev.
I will try to help you. The Scripts run.

sarandaegus, 2016-12-23:
Hi. i write a new post in this link: 

https://github.com/ev3dev/ev3dev/issues/491

Thank you.

Sergio Aranda

jabrena, 2017-01-08:
Hi @sarandaegus, 

many thanks for the link. Do you have the script updated in some place?

jabrena, 2017-01-15:
Closing this issue here.

Related issue:
https://github.com/ev3dev-lang-java/RPLidar4J/issues/1

jabrena, 2017-03-11:
https://packages.debian.org/jessie/librxtx-java

jabrena, 2017-03-11:
http://www.planet-rcs.de/article/mac_serial_port/

jabrena, 2017-03-19:
Store data in JSON format:
https://github.com/jabrena/livingrobots/blob/master/chapter3/ev3/java/RPLIDARTest/src/RPLIDARTest.java

jabrena, 2017-03-19:
Showing polar chart from LIDAR on a Web interface:
https://github.com/jabrena/livingrobots/tree/master/chapter8/www/Radar/amcharts

jabrena, 2017-03-20:
http://pastebin.com/daLc6JTH

jabrena, 2017-03-22:
First release is out.
https://jitpack.io/#ev3dev-lang-java/RPLidar4J/v0.1.0-SNAPSHOT

closed 2017-03-22 by jabrena

YuHanChen04, 2018-04-17:
Hello @jabrena :
I want to try the script of LidarPrint2.py  on my EV3 with RPLidar A2 .
Could you give some advice and the script ,and let me to try 

Thank you so much .
Sorry for my English !