ev3dev-lang-java/ev3dev-lang-java #86: Apply Singleton design pattern for Battery & Sound classes

jabrena, 2016-01-11:


dlech, 2016-01-16:
I don't think this is a good idea for the battery class. In the future, there could be more than one battery. For example, some sensor multiplexers and motor controllers have their own battery that provides information back to the EV3. Instead, it would be better to add a factory method to get the "System" battery - the battery whose "scope" attribute (POWER_SUPPLY_PROP_SCOPE property) returns "System". Other batteries will have scope of "Device" or "Unknown".

jabrena, 2016-01-17:
Sorry, I was going to close the issue but I am going to study in detail your idea because it has sense and it is useful your opinion. Feel free to discuss anything. This after lunch, I will study your comment. The rest of the API was tested and it runs.

closed 2016-01-17 by jabrena