ev3dev-lang-java/ev3dev-lang-java #87: Apply Factory Design pattern for Battery.

jabrena, 2016-01-17:


jabrena, 2016-01-17:
Related issue: https://github.com/jabrena/ev3dev-lang-java/issues/86

jabrena, 2016-01-17:
@dlech I have implemented the factory. 
Do you know some device in the market which you can manage the battery?
Are you thinking in this Power Option?
http://www.mindsensors.com/rpi/41-uninterruptible-power-supply-for-raspberry-pi

jabrena, 2016-01-17:
Take a look the example:

https://github.com/jabrena/ev3dev-lang-java/blob/master/library/src/test/java/ev3dev/java/examples/BatteryDemo.java

jabrena, 2016-01-17:
Try to improve the method handling no input parameter:

**Idea:**
http://stackoverflow.com/questions/2330942/java-variable-number-or-arguments-for-a-method

dlech, 2016-01-17:
> Do you know some device in the market which you can manage the battery?

I did not know of any specifically for robotics, but the example that you gave is a good one. We don't have a driver for it currently, but it would be easy to write one.

jabrena, 2016-01-17:
Yes, it was interesting.

I am going to research this Device for v0.3.0
http://www.mindsensors.com/rpi/41-uninterruptible-power-supply-for-raspberry-pi

Anyway, this way will be consolidated for v0.3.0 not v0.2.0. 

This next week, I will close the number of features for v0.2.0 and I will begin with v0.3.0. Besides, I will use a Git branch to not work in master. (Bad habit, I recognize :( )

Many thanks

jabrena, 2016-12-05:
REJECTED BY LEAN

closed 2016-12-05 by jabrena