ev3dev-lang-java/ev3dev-lang-java #126: Unresponsive Motor

pambrose, 2016-04-04:
Hello,

I am not getting any response from my motors.  I am able to read sensor and the battery values.
I have tried it with version 0.2.0 and the next-release branch. The python ev3dev bindings
work fine.  My brick is running this version of ev3dev:
Linux ev3dev 3.16.7-ckt21-9-ev3dev-ev3 #1 PREEMPT Tue Dec 15 15:16:17 CST 2015 armv5tejl GNU/Linux

Any ideas on what I am doing wrong would be much appreciated.
 
``` java
public class Simple {
  public static void main(String[] args)
      throws InterruptedException {
    System.out.println("Starting motor on A");
    final EV3LargeRegulatedMotor mA = new EV3LargeRegulatedMotor(MotorPort.A);
    mA.setSpeed(500);
    mA.forward();
    System.out.println(String.format("Large Motor is moving: %s at speed %d", mA.isMoving(), mA.getSpeed()));
    Delay.msDelay(2000);
    mA.stop();
    System.out.println("Stopped motor");
    System.out.println("Battery: " + Battery.getInstance().getVoltage());
    Sound.getInstance().playTone(1000, 100);
    System.exit(0);
  }
}
```

jabrena, 2016-04-04:
Good afternoon,

The first think that I thought is that your Brick doesnt have the latest distro release, but you have it.

``` java
root@ev3dev:~# uname -r
3.16.7-ckt21-9-ev3dev-ev3
```
I am going to to test the example to know what is the problem.

Many thanks for participate in this Java project.

Juan Antonio

jabrena, 2016-04-04:
Hi, 

it is a bit rare, I have just tested with latest development branch: next-release and the behaviour is right. I send you the output from my terminal now:

I have just added the example to the project:

``` java
package examples.motors;

import lejos.utility.Delay;
import ev3dev.hardware.Battery;
import ev3dev.hardware.Sound;
import ev3dev.hardware.port.MotorPort;
import ev3dev.hardware.motor.EV3LargeRegulatedMotor;

public class LargeMotorTest {

    public static void main(String[] args) throws InterruptedException {

        System.out.println("Starting motor on A");
        final EV3LargeRegulatedMotor mA = new EV3LargeRegulatedMotor(MotorPort.A);
        mA.setSpeed(500);
        mA.forward();
        System.out.println(String.format("Large Motor is moving: %s at speed %d", mA.isMoving(), mA.getSpeed()));
        Delay.msDelay(2000);
        mA.stop();
        System.out.println("Stopped motor");
        System.out.println("Battery: " + Battery.getInstance().getVoltage());
        Sound.getInstance().playTone(1000, 100);

        System.exit(0);
    }

}
```

``` bash
root@ev3dev:/home/robot# java -Djava.util.logging.config.file=logging.properties -jar ev3-lang-java-all-0.3.0.jar
Starting motor on A
19:06:28.933 [main] DEBUG ev3dev.hardware.EV3DevSysfs - /sys/class/lego-port
19:06:29.876 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port0/address
19:06:29.948 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port1/address
19:06:29.984 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port2/address
19:06:30.051 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port3/address
19:06:30.100 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port4/address
19:06:30.191 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
19:06:30.323 [main] DEBUG ev3dev.hardware.EV3DevSysfs - /sys/class/tacho-motor
19:06:30.418 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor2/address
19:06:30.503 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo on > /sys/class/tacho-motor/motor2/speed_regulation
19:06:30.623 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 500 > /sys/class/tacho-motor/motor2/speed_sp
19:06:30.724 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 500 > /sys/class/tacho-motor/motor2/speed_sp
19:06:30.815 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo run-forever > /sys/class/tacho-motor/motor2/command
19:06:30.905 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor2/state
19:06:30.963 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor2/speed_sp
Large Motor is moving: true at speed 500
19:06:34.766 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo stop > /sys/class/tacho-motor/motor2/command
Stopped motor
19:06:35.044 [main] DEBUG ev3dev.hardware.Platform - Detected the platform: EV3BRICK
19:06:35.091 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/power_supply/legoev3-battery/voltage_now
Battery: 8366066.0
19:06:35.449 [main] DEBUG ev3dev.hardware.Platform - Detected the platform: EV3BRICK
19:06:35.474 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo  1000 100 > /sys/devices/platform/snd-legoev3/tone
root@ev3dev:/home/robot# 

```

I am going to upload the example in GIT to test by yourself.

in your host, update the IP of the EV3 Brick in the file: build.gradle and type:

``` bash
gradle clean fatJar deploy
```
In your EV3 Brick, type:

``` bash
java -jar ev3-lang-java-all-0.3.0.jar
```

pambrose, 2016-04-04:
I downloaded the LargeMotorTest  example, ran it and got the same debugging output, but the motor was still inactive.

I am able to reach the motor via echo commands from the shell.  I can also reach it via python.
I also wrote something in java that talked directly to the FS that is able to reach the motor.

The only odd thing I am seeing is after running the LargeMotorTest program, the contents of 
/sys/class/tacho-motor changes. Before, it might have motor1, motor2 and after it has motor1, motor3.
If I run it again, the contents change to motor1, motor4. I am not changing the cables and yet the endpoints change.  
That does not happen after running a python app or sending commands via the shell.

Do you see the contents of /sys/class/tacho-motor changing after running LargeMotorTest?

Thanks for your help.

jabrena, 2016-04-04:
mmm, no but it is an interesting case to debug. If you like, tomorrow we could debug this issue to find the solution. The library is young and it could be have some interesting bugs to solve :)

Can you execute the instructions manually using the log traces?
Do you know if Python library is able to offer some log to know what operations they do?

In my brick, I don't have any problem but you have some stuff so It is necessary to investigate a bit :)

jabrena, 2016-04-04:
The initial premise is that you execute well with Python, but you can't with Java lib. Can you past the example and instructions to test with python?

jabrena, 2016-04-04:
Maybe it is this part:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/library/src/main/java/ev3dev/hardware/motor/BaseRegulatedMotor.java#L64-L72

Tomorrow, I will check this piece of code.

Juan Antonio

jabrena, 2016-04-04:
Is it possible that you are using different motors?
I am trying to remember why I added this line:

``` java
this.setStringAttribute(MODE,SYSTEM_CLASS_NAME);
19:06:30.191 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
```

https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/73

pambrose, 2016-04-05:
I will give that a try.

pambrose, 2016-04-05:
I tried the sample from the README:

import ev3dev.ev3 as ev3
m = ev3.LargeMotor('outA’) 
m.run_timed(time_sp=3000, duty_cycle_sp=75)

pambrose, 2016-04-05:
I am using an EV3 motor from the commercial Mindstorm product. I ma not sure if the educational
product has different motors.

pambrose, 2016-04-05:
You were right, if I remove that line, the motor works fine.

jabrena, 2016-04-05:
Mmm, I have to check it. I think that I added this Line to avoid a problem if in a previous interaction a user connect in the same port a dc motor and later, the user connect a tacho motor. I am going to create a new sd card to test with a clean image.

jabrena, 2016-04-05:
Can you continúe with your examples?

jabrena, 2016-04-06:
I found an alternative to current situation. If a user with EV3 Brick change the motors in the same port and the user uses dc-motors in that moment, althought the use disconnect the dc-motor and connect later a ev3 large motor, the lego-port doesn't identify the motor in an automatic way. Once solution in EV3, is to set the mode as auto in the constructor.

I send a session as example:

``` bash
root@ev3dev:/sys/class# ls
backlight  block      dc-motor  gpio      hidraw       i2c-dev    input  lego-port  misc      mtd  power_supply  rfkill  scsi_device  scsi_generic  sound       tacho-motor  udc     vc         watchdog
bdi        bluetooth  dma       graphics  i2c-adapter  ieee80211  leds   mem        mmc_host  net  pwm           rtc     scsi_disk    scsi_host     spi_master  tty          usbmon  vtconsole
root@ev3dev:/sys/class# cd lego-port/
root@ev3dev:/sys/class/lego-port# ls
port0  port1  port2  port3  port4  port5  port6  port7
root@ev3dev:/sys/class/lego-port# cd port4/
root@ev3dev:/sys/class/lego-port/port4# ls
address  device  driver_name  mode  modes  outA:lego-ev3-l-motor  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port4# cat mode
auto
root@ev3dev:/sys/class/lego-port/port4# cat status 
tacho-motor
``` 
I disconnect the motor to connect a dc motor

``` bash
root@ev3dev:/sys/class/lego-port/port4# ls 
address  device  driver_name  mode  modes  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port4# cat status 
no-motor
root@ev3dev:/sys/class/lego-port/port4# cat modes 
auto tacho-motor dc-motor led raw
root@ev3dev:/sys/class/lego-port/port4# echo dc-motor > mode
root@ev3dev:/sys/class/lego-port/port4# ls
address  device  driver_name  mode  modes  outA:rcx-motor  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port4# cat status 
dc-motor
root@ev3dev:/sys/class/lego-port/port4# cd ..
root@ev3dev:/sys/class/lego-port# ls
port0  port1  port2  port3  port4  port5  port6  port7
root@ev3dev:/sys/class/lego-port# cd ..
root@ev3dev:/sys/class# ls
backlight  block      dc-motor  gpio      hidraw       i2c-dev    input  lego-port  misc      mtd  power_supply  rfkill  scsi_device  scsi_generic  sound       tacho-motor  udc     vc         watchdog
bdi        bluetooth  dma       graphics  i2c-adapter  ieee80211  leds   mem        mmc_host  net  pwm           rtc     scsi_disk    scsi_host     spi_master  tty          usbmon  vtconsole
root@ev3dev:/sys/class# cd dc-motor/   
root@ev3dev:/sys/class/dc-motor# ls
motor0
root@ev3dev:/sys/class/dc-motor# cd motor0
root@ev3dev:/sys/class/dc-motor/motor0# ls
address  command  commands  device  driver_name  duty_cycle  duty_cycle_sp  polarity  power  ramp_down_sp  ramp_up_sp  state  stop_command  stop_commands  subsystem  time_sp  uevent
root@ev3dev:/sys/class/dc-motor/motor0# cat driver_name 
rcx-motor
root@ev3dev:/sys/class/dc-motor/motor0# cd ..
root@ev3dev:/sys/class/dc-motor# cd ..
``` 
connect a ev3 motor

``` bash
root@ev3dev:/sys/class# cd tacho-motor/
root@ev3dev:/sys/class/tacho-motor# ls
root@ev3dev:/sys/class/tacho-motor# ls
root@ev3dev:/sys/class/tacho-motor# ls
root@ev3dev:/sys/class/tacho-motor# cd ..
root@ev3dev:/sys/class# cd lego-port/
root@ev3dev:/sys/class/lego-port# ls
port0  port1  port2  port3  port4  port5  port6  port7
root@ev3dev:/sys/class/lego-port# cd port4/
root@ev3dev:/sys/class/lego-port/port4# ls
address  device  driver_name  mode  modes  outA:rcx-motor  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port4# ls
address  device  driver_name  mode  modes  outA:rcx-motor  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port4# cat mode
dc-motor
root@ev3dev:/sys/class/lego-port/port4# cat modes
auto tacho-motor dc-motor led raw
root@ev3dev:/sys/class/lego-port/port4# echo auto > mode
mode   modes  
root@ev3dev:/sys/class/lego-port/port4# echo auto > mode
mode   modes  
root@ev3dev:/sys/class/lego-port/port4# echo auto > mode
root@ev3dev:/sys/class/lego-port/port4# ls
address  device  driver_name  mode  modes  outA:lego-ev3-m-motor  power  set_device  status  subsystem  uevent
root@ev3dev:/sys/class/lego-port/port4# 
``` 

I don't know why you reached that status in that port, but i consider that it is a necessary to check this part. for ev3, the best way is to "reset" the status of the lego port, in this case, using "auto" could be the way. Maybe one reset command in lego-port, could be a better way. @dlech, what is your opinion to have a reset command in /sys/class/lego-port?

Scenario EV3:
EV3 Motor: If the user connectd a ev3 motor, the class set to auto to "reset" the lego-port
DC Motor: If the user connect  dc motor, the class set the mode

Scenario PiBoard: Pending of testing

Juan Antonio

brezinajn, 2016-04-18:
I had the same issue. Update kernel to 3.16.7-ckt26-10-ev3dev-ev3 and you'll be fine.

jabrena, 2016-04-18:
Hi @brezinajn, 

Tomorrow, I will upgrade the brick to that kernel to test.

Cheers