ev3dev-lang-java/ev3dev-lang-java #129: EV3RegulatedMotor fails to start in some simple occasions

brezinajn, 2016-04-19:
In this case motor fails to start. Program just waits 10 seconds and then exits.
```java
EV3MediumRegulatedMotor mA = new EV3MediumRegulatedMotor(MotorPort.A);
mA.setSpeed(500);

mA.forward();
Delay.msDelay(10000);

mA.stop();
System.exit(0);
```
When I prepend the code with another forward() and delay() it magically starts to work.
```java
EV3MediumRegulatedMotor mA = new EV3MediumRegulatedMotor(MotorPort.A);
mA.setSpeed(500);

mA.forward();
Delay.msDelay(100);

mA.forward();
Delay.msDelay(10000);

mA.stop();
System.exit(0);
```
I experience the same weird behavior with rotate function.
Presumably this is just due my inexperience with ev3 :).

jabrena, 2016-04-19:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/128

jabrena, 2016-04-19:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/130

jabrena, 2016-04-19:
Hi, 

You are using latest ev3dev kernel but previous release and current code for v0.3.0 run with previous kernel. I am reading changes in this release and It is easy to fix the changes.

If you want to play meanwhile you could downgrade the kernel:
https://github.com/ev3dev/ev3dev/issues/609

Or wait a bit. PR are welcome.

Cheers

jabrena, 2016-04-19:
I am trying to upgrade the environment at the moment.

jabrena, 2016-04-19:
Tomorrow, I will try to fix the issue. with the new feature to see the traces, it is easier to know the interaction with /sys/class

``` bash
root@ev3dev:/home/robot# java -jar ev3-lang-java-all-0.3.0.jar 
22:58:33.086 [main] DEBUG ev3dev.hardware.Platform - Detected the platform: EV3BRICK
22:58:34.907 [main] DEBUG ev3dev.hardware.EV3DevSysfs - /sys/class/lego-port
22:58:34.937 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port0/address
22:58:35.016 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port1/address
22:58:35.059 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port2/address
22:58:35.110 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port3/address
22:58:35.147 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port4/address
22:58:35.197 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port5/address
22:58:35.261 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port6/address
22:58:35.319 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo tacho-motor > /sys/class/lego-port/port6/mode
22:58:35.521 [main] DEBUG ev3dev.hardware.EV3DevSysfs - /sys/class/tacho-motor
22:58:35.587 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor0/address
22:58:35.689 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/address
22:58:35.797 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo on > /sys/class/tacho-motor/motor1/speed_regulation
22:58:35.874 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 100 > /sys/class/tacho-motor/motor1/speed_sp
22:58:38.727 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/position_sp
0
22:58:38.795 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 90 > /sys/class/tacho-motor/motor1/position_sp
22:58:38.848 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo run-to-rel-pos > /sys/class/tacho-motor/motor1/command
22:58:38.905 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:38.965 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.002 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.674 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.746 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.807 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.842 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.892 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:39.952 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.010 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.074 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.133 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.207 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.266 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.311 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.402 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.456 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.503 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.547 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.601 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.661 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.700 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.749 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.814 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.852 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.918 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:40.958 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:41.010 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
22:58:41.076 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/state
```

Cheers

jabrena, 2016-04-19:
Besides, I checked another example with the methods that you say:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/next-release/library/src/main/java/examples/motors/RegulatedMotorDemo.java

``` bash
root@ev3dev:/home/robot# java -jar ev3-lang-java-all-0.3.0.jar 
23:14:34.838 [main] DEBUG ev3dev.hardware.EV3DevSysfs - /sys/class/lego-port
23:14:35.827 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port0/address
23:14:35.894 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port1/address
23:14:35.931 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port2/address
23:14:35.980 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port3/address
23:14:36.038 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/lego-port/port4/address
23:14:36.117 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
23:14:36.356 [main] DEBUG ev3dev.hardware.EV3DevSysfs - /sys/class/tacho-motor
23:14:36.413 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor1/address
23:14:36.527 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor3/address
23:14:36.600 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo on > /sys/class/tacho-motor/motor3/speed_regulation
23:14:36.691 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 500 > /sys/class/tacho-motor/motor3/speed_sp
23:14:36.794 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 500 > /sys/class/tacho-motor/motor3/speed_sp
23:14:36.850 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo run-forever > /sys/class/tacho-motor/motor3/command
23:14:36.944 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor3/state
true
23:14:38.260 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo stop > /sys/class/tacho-motor/motor3/command
23:14:38.324 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor3/state
false
23:14:38.405 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo -500 > /sys/class/tacho-motor/motor3/speed_sp
23:14:38.452 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo run-forever > /sys/class/tacho-motor/motor3/command
23:14:38.519 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor3/state
true
23:14:39.595 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo stop > /sys/class/tacho-motor/motor3/command
23:14:39.632 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor3/state
false
23:14:39.726 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo 500 > /sys/class/tacho-motor/motor3/speed_sp
23:14:39.776 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo run-forever > /sys/class/tacho-motor/motor3/command
23:14:40.370 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/tacho-motor/motor3/state
true
23:14:41.474 [main] DEBUG ev3dev.hardware.EV3DevSysfs - echo stop > /sys/class/tacho-motor/motor3/command
23:14:41.785 [main] DEBUG ev3dev.hardware.Platform - Detected the platform: EV3BRICK
23:14:41.810 [main] DEBUG ev3dev.hardware.EV3DevSysfs - cat /sys/class/power_supply/legoev3-battery/voltage_now
7353000.0
root@ev3dev:/home/robot#
``` 

What version, are you using?

Anyway, I observed some changes to fix to run better with kernel 10 but forward, backward, rotate run nice. But I have to do a full regression to certificate kernel 10 with this library.

jabrena, 2017-01-02:
Fixed

jabrena, 2017-01-02:
@brezinajn 
Sorry by the delay, did you continue with this kind of projects?

brezinajn, 2017-01-03:
No problem. I had little time for my EV3. Although last week I started new project.

jabrena, 2017-01-03:
Hi @brezinajn,

Is about something related with EV3 Ecosystem?

brezinajn, 2017-01-04:
Yes, It's just simple experiment. Teaching robot with neural network to play tic-tac-toe. Anyway, do you need help with ev3dev-lang-java :)?

jabrena, 2017-01-04:
Maybe a neural network example for the project could be interesting to share with the community if you like. :)

closed 2017-01-07 by jabrena

brezinajn, 2017-01-07:
I'll of course provide my source to community :)

jabrena, 2017-01-07:
If you like, you could do pull request to example repo, if you like

jabrena, 2017-01-07:
https://github.com/ev3dev-lang-java/examples