ev3dev-lang-java/ev3dev-lang-java #176: Install JDK9 Early version on EV3

jabrena, 2017-02-15:
https://jdk9.java.net/download/

jabrena, 2017-02-15:
```
sudo update-alternatives --install "/usr/bin/java" "java" "/usr/lib/jvm/jdk1.8.0/bin/java" 1
sudo update-alternatives --install "/usr/bin/javac" "javac" "/usr/lib/jvm/jdk1.8.0/bin/javac" 1
``` 

http://askubuntu.com/questions/56104/how-can-i-install-sun-oracles-proprietary-java-jdk-6-7-8-or-jre

jabrena, 2017-02-15:
``` 
jdk-9/README.html
jdk-9/bin/appletviewer
jdk-9/bin/idlj
jdk-9/bin/jar
jdk-9/bin/jarsigner
jdk-9/bin/java
jdk-9/bin/javac
jdk-9/bin/javadoc
jdk-9/bin/javah
jdk-9/bin/javap
jdk-9/bin/jcmd
jdk-9/bin/jconsole
jdk-9/bin/jdb
jdk-9/bin/jdeprscan
jdk-9/bin/jdeps
jdk-9/bin/jhsdb
jdk-9/bin/jimage
jdk-9/bin/jinfo
jdk-9/bin/jjs
jdk-9/bin/jlink
jdk-9/bin/jmap
jdk-9/bin/jmod
jdk-9/bin/jps
jdk-9/bin/jrunscript
jdk-9/bin/jshell
jdk-9/bin/jstack
jdk-9/bin/jstat
jdk-9/bin/jstatd
jdk-9/bin/keytool
jdk-9/bin/orbd
jdk-9/bin/pack200
jdk-9/bin/policytool
jdk-9/bin/rmic
jdk-9/bin/rmid
jdk-9/bin/rmiregistry
jdk-9/bin/schemagen
jdk-9/bin/serialver
jdk-9/bin/servertool
jdk-9/bin/tnameserv
jdk-9/bin/unpack200
jdk-9/bin/wsgen
jdk-9/bin/wsimport
jdk-9/bin/xjc
jdk-9/conf/logging.properties
jdk-9/conf/management/jmxremote.access
jdk-9/conf/management/jmxremote.password.template
jdk-9/conf/management/management.properties
jdk-9/conf/management/snmp.acl.template
jdk-9/conf/net.properties
jdk-9/conf/security/java.policy
jdk-9/conf/security/java.security
jdk-9/conf/security/policy/README.txt
jdk-9/conf/security/policy/limited/default_US_export.policy
jdk-9/conf/security/policy/limited/default_local.policy
jdk-9/conf/security/policy/limited/exempt_local.policy
jdk-9/conf/security/policy/unlimited/default_US_export.policy
jdk-9/conf/security/policy/unlimited/default_local.policy
jdk-9/conf/sound.properties
jdk-9/include/classfile_constants.h
jdk-9/include/ir.idl
jdk-9/include/jawt.h
jdk-9/include/jdwpTransport.h
jdk-9/include/jni.h
jdk-9/include/jvmti.h
jdk-9/include/jvmticmlr.h
jdk-9/include/linux/jawt_md.h
jdk-9/include/linux/jni_md.h
jdk-9/include/orb.idl
jdk-9/jmods/java.activation.jmod
[... 246 lines cut ...]
jdk-9/lib/libsunec.so
jdk-9/lib/libt2k.so
jdk-9/lib/libunpack.so
jdk-9/lib/libverify.so
jdk-9/lib/libzip.so
jdk-9/lib/minimal/Xusage.txt
jdk-9/lib/minimal/libjsig.so
jdk-9/lib/minimal/libjvm.so
jdk-9/lib/modules
jdk-9/lib/oblique-fonts/LucidaSansDemiOblique.ttf
jdk-9/lib/oblique-fonts/LucidaSansOblique.ttf
jdk-9/lib/oblique-fonts/LucidaTypewriterBoldOblique.ttf
jdk-9/lib/oblique-fonts/LucidaTypewriterOblique.ttf
jdk-9/lib/oblique-fonts/fonts.dir
jdk-9/lib/psfont.properties.ja
jdk-9/lib/psfontj2d.properties
jdk-9/lib/security/blacklist
jdk-9/lib/security/blacklisted.certs
jdk-9/lib/security/cacerts
jdk-9/lib/security/default.policy
jdk-9/lib/security/trusted.libraries
jdk-9/lib/src.zip
jdk-9/lib/tzdb.dat
jdk-9/release
```

jabrena, 2017-02-15:
http://askubuntu.com/questions/17299/what-do-the-different-colors-mean-in-the-terminal

jabrena, 2017-02-15:
EV3 Test:

``` 
robot@ev3dev:/opt/jdk-9/bin$ ls
appletviewer  javadoc   jdeprscan  jlink       jstack   policytool   servertool
idlj          javah     jdeps      jmap        jstat    rmic         tnameserv
jar           javap     jhsdb      jmod        jstatd   rmid         unpack200
jarsigner     jcmd      jimage     jps         keytool  rmiregistry  wsgen
java          jconsole  jinfo      jrunscript  orbd     schemagen    wsimport
javac         jdb       jjs        jshell      pack200  serialver    xjc
```

jabrena, 2017-02-15:
**Testing the same distro on Raspi 3:**

https://es.wikipedia.org/wiki/Raspberry_Pi

```
Raspberry Pi 3 Modelo B
Broadcom BCM2837 (CPU + GPU + DSP + SDRAM + Puerto USB	
1.2GHz 64-bit quad-core ARMv8
```

jabrena, 2017-02-15:
http://cfp.devoxx.us/2017/talk/XGT-4587/Raspberry_Pi_with_Java_9

jabrena, 2017-02-15:
The zips from website doesn`t work for EV3 and Raspi 3.
I am waiting for some feedback from: 
http://cfp.devoxx.us/2017/talk/XGT-4587/Raspberry_Pi_with_Java_9

jabrena, 2017-02-15:
https://launchpad.net/ubuntu/+source/openjdk-9/9~b155-1/+build/11961038

jabrena, 2017-02-15:
https://packages.debian.org/jessie/ca-certificates-java

jabrena, 2017-02-16:
``` 
robot@ev3dev:~$ sudo dpkg -i openjdk-9-jre-headless_9-b155-1_armhf.deb 
(Reading database ... 31598 files and directories currently installed.)
Preparing to unpack openjdk-9-jre-headless_9-b155-1_armhf.deb ...
Unpacking openjdk-9-jre-headless:armhf (9~b155-1) over (9~b155-1) ...
dpkg: dependency problems prevent configuration of openjdk-9-jre-headless:armhf:
 openjdk-9-jre-headless:armhf depends on libjpeg8 (>= 8c); however:
  Package libjpeg8 is not installed.
 openjdk-9-jre-headless:armhf depends on util-linux (>= 2.26.2-4); however:
  Version of util-linux on system is 2.25.2-6.

dpkg: error processing package openjdk-9-jre-headless:armhf (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openjdk-9-jre-headless:armhf
robot@ev3dev:~$ 
```

jabrena, 2017-02-16:
```
robot@ev3dev:~$ sudo dpkg -i util-linux_2.29.1-1_armhf.deb
dpkg: regarding util-linux_2.29.1-1_armhf.deb containing util-linux:
 util-linux breaks bash-completion (<< 1:2.1-4.1~)
  bash-completion (version 1:2.1-4) is present and installed.

dpkg: error processing archive util-linux_2.29.1-1_armhf.deb (--install):
 installing util-linux would break bash-completion, and
 deconfiguration is not permitted (--auto-deconfigure might help)
Errors were encountered while processing:
 util-linux_2.29.1-1_armhf.deb
```

jabrena, 2017-02-16:
Related issue: https://github.com/ev3dev/ev3dev/issues/843

jabrena, 2017-02-16:
http://askubuntu.com/questions/769467/can-not-install-openjdk-9-jdk-because-it-tries-to-overwrite-file-aready-includ

jabrena, 2017-02-16:
```
Preparing to unpack openjdk-9-jre_9-b155-1_armhf.deb ...
Unpacking openjdk-9-jre:armhf (9~b155-1) over (9~b155-1) ...
Setting up openjdk-9-jre:armhf (9~b155-1) ...
```

jabrena, 2017-02-16:
``` 
Preparing to unpack openjdk-9-jdk_9-b155-1_armhf.deb ...
Unpacking openjdk-9-jdk:armhf (9~b155-1) ...
Setting up openjdk-9-jdk:armhf (9~b155-1) ...
```

jabrena, 2017-02-16:
[image]

jabrena, 2017-02-16:
Installed.

Hax4us, 2017-11-21:
@jabrena from where u downloaded jdk9 for arm64

jabrena, 2017-11-30:
Hi @Hax4us, did you solve the issue?
Sorry by the delay

Juan Antonio

closed 2017-12-31 by jabrena