ev3dev-lang-java/ev3dev-lang-java #209: Adding Unit test

cecile, 2017-03-27:
please add unit test

jabrena, 2017-03-29:
yes, it is something pending. I will add them.

jabrena, 2017-04-23:
It is necessary to research how to add juice in the same way that it is done on xbot and stronghold.
https://github.com/Team488/Shared/blob/master/2015%20Competition%20Season/CompetitionBot/workspace/XbotCommonLib/src/xbot/common/injection/RobotModule.java

jabrena, 2017-04-23:
https://github.com/Team488/SeriouslyCommonLib/blob/master/tests/xbot/common/command/BaseCommandTest.java

jabrena, 2017-04-23:
https://github.com/Team488/SeriouslyCommonLib/blob/master/tests/xbot/common/controls/sensors/AnalogDistanceSensorTest.java

jabrena, 2017-04-24:
https://github.com/google/jimfs/blob/master/README.md

jabrena, 2017-04-24:
http://slides.com/strongback/using-strongback/#/

jabrena, 2017-04-24:
http://stackoverflow.com/questions/6970192/are-there-any-fake-file-system-frameworks-for-java

jabrena, 2017-04-24:
http://mockftpserver.sourceforge.net/apidocs/org/mockftpserver/fake/filesystem/FileSystem.html

jabrena, 2017-04-24:
https://www.google.co.uk/amp/s/garygregory.wordpress.com/2010/01/20/junit-tip-use-rules-to-manage-temporary-files-and-folders/amp/

jabrena, 2017-04-24:
http://www.programcreek.com/java-api-examples/org.junit.rules.TemporaryFolder
http://junit.org/junit4/javadoc/4.12/org/junit/rules/TemporaryFolder.html
http://howtodoinjava.com/junit/junit-creating-temporary-filefolder-using-temporaryfolder-rule/

jabrena, 2017-04-28:
http://stackoverflow.com/questions/21879804/inject-mocks-for-objects-created-by-factory-classes

jabrena, 2017-04-30:
https://github.com/ddemidov/ev3dev-lang-fake-sys

jabrena, 2017-06-25:
Added question to `junit team` about the usage of: File.createTempFile
https://github.com/junit-team/junit4/issues/1469

jabrena, 2017-06-26:
Unblocked with the Master class of @juan-medina about Mocking.

jabrena, 2017-09-10:
Ideas:
https://stackoverflow.com/questions/27558284/java-mock-object-without-dependency-injection

jabrena, 2017-09-10:
FAKE_SYS = os.path.join(os.path.dirname(__file__), 'fake-sys')
https://github.com/rhempel/ev3dev-lang-python/pull/352/files

jabrena, 2018-03-15:
Added base classes to implement testing for Sensors

jabrena, 2018-03-31:
https://docs.gradle.org/current/userguide/jacoco_plugin.html

jabrena, 2018-04-02:
Minimum Level, reached in the last CI build:
https://travis-ci.org/ev3dev-lang-java/ev3dev-lang-java/builds/361212850

```
jacocoTestCoverageVerification {
    violationRules {
        rule {
            limit {
                minimum = 0.75
            }
        }
    }
}
```

Now, it is necessary to review Unit Testing syntax in order to close this issue and the issue about testing infrastructure:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/423

jabrena, 2018-04-05:
Many thanks @juan-medina with this issue.
Now, it is very easy maintain this library for multiple Debian Distros.

Cheers

juan-medina, 2018-04-05:
@jabrena I think this issue should not be close, until you have an acceptable level of coverage > 85%

jabrena, 2018-04-05:
hehehe, Oki... :D 

Some days, I added the comment about 0.75... 
but anyway, I will do.

Every comment to improve the quality is welcome and this one goes in that direction. :)

jabrena, 2018-04-06:
Reached.
https://travis-ci.org/ev3dev-lang-java/ev3dev-lang-java/builds/363201844

Juan Antonio

closed 2018-04-06 by jabrena