ev3dev-lang-java/ev3dev-lang-java #216: Develop documentation about

jabrena, 2017-04-03:
@krish-pillai

This is the new Index about all documentation:
https://github.com/ev3dev-lang-java/docs/blob/develop/README.md

If you like, you could start in this areas:

- Exercises
  - Exercises with Actuators
  - Exercises with Sensors
  - Exercises combining Actuators & Sensors
- Applications
  - Basic BumperCar Robot
  - BumperCar Robot with Subsumption architecture
  - Follow Line robot with EV3 Light Sensor
  - Follow Line robot using OpenCV
  - Detect a Human Face with OpenCV

jabrena, 2017-05-28:
http://www.java2s.com/Code/Java/J2ME/SimpleCanvas.htm
https://github.com/miguelbayon/bouncing-balls?files=1

jabrena, 2018-04-29:
What is your opinion @FaveAsian?

jabrena, 2018-04-30:
https://github.com/mindstorms-cop/lejos-playground/blob/master/src/main/java/org/mindstormscop/drehkreisel/Main.java

jabrena, 2018-04-30:
https://github.com/tungnx94/LegoMS/tree/master/legoMS3_vs2/src/robotComponents

FaveAsian, 2018-05-01:
Is this going to be code example for new beginners? I just need some background knowledge of what is going on with the documentation.

jabrena, 2018-05-01:
Hi @FaveAsian,

> The documentation should be for beginners in Java.

Take a look current documentation:
http://ev3dev-lang-java.github.io/docs/support/index.html

We could improve the area about fundamentals in several ways.
What ideas do you have?
How to enrich the educational experience in your classroom with a better documentation for your students.

Cheers

jabrena, 2018-05-04:
Take your time mate. :)

FaveAsian, 2018-06-05:
> Any progress?

I haven't been able to review much. What I can say from what I have seen is
besides installing java code needing a modification. You should also tell
the user not to unzip the java file and just upload the zip file as that
got me confused. Other things I would have to look at another time.

jabrena, 2018-06-13:
I will upgrade the part about zip

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena