ev3dev-lang-java/ev3dev-lang-java #220: Install Java 9 JDK in a easy way for EV3 Brick

jabrena, 2017-04-09:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/182

jabrena, 2018-03-30:
In terms of the technology, the development of the repository about OpenJDK  for EV3Dev was very disruptive for the project:
https://github.com/ev3dev-lang-java/openjdk-ev3

``` 
Linux ev3dev 4.9.58-ev3dev-1.6.0-ev3 #1 PREEMPT Tue Oct 31 22:16:58 CDT 2017 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Wed Mar 28 06:26:41 2018 from 192.168.1.198
robot@ev3dev:~$ ./jre-ev3/bin/java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46)
OpenJDK Client VM (build 10+46, mixed mode)
robot@ev3dev:~$

JakubVanek, 2018-03-30:
What about leJOS-like installer? You can look it up in the SD Card sources.

jabrena, 2018-03-30:
Updated the installer to support OpenJDK 10 JRI.

Traces:

```
robot@ev3dev:~/installer$ sudo ./installer.sh

##############################
# EV3Dev-lang-java Installer #
##############################
# Last update: 2018/03/30    #
##############################

Init installer
rm: cannot remove './module2/*': No such file or directory
--2018-03-30 14:05:08--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/utilities.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.132.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.132.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 896 [text/plain]
Saving to: 'utilities.sh'

utilities.sh                             100%[===============================================================================>]     896  --.-KB/s    in 0.001s  

Last-modified header missing -- time-stamps turned off.
2018-03-30 14:05:10 (1.21 MB/s) - 'utilities.sh' saved [896/896]

#########################################
# PLATFORM SECTION                      #
#########################################

--2018-03-30 14:05:11--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/platform.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.132.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.132.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1006 [text/plain]
Saving to: 'platform.sh'

platform.sh                              100%[===============================================================================>]    1006  --.-KB/s    in 0.001s  

Last-modified header missing -- time-stamps turned off.
2018-03-30 14:05:13 (1.11 MB/s) - 'platform.sh' saved [1006/1006]

The user has a EV3 Brick
Platform detected: EV3

#########################################
# JAVA SECTION                          #
#########################################

--2018-03-30 14:05:14--  https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules/java.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.132.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.132.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1601 (1.6K) [text/plain]
Saving to: 'java.sh'

java.sh                                  100%[===============================================================================>]   1.56K  --.-KB/s    in 0.001s  

Last-modified header missing -- time-stamps turned off.
2018-03-30 14:05:15 (1.57 MB/s) - 'java.sh' saved [1601/1601]

No java detected
--2018-03-30 14:05:16--  https://github.com/ev3dev-lang-java/openjdk-ev3/releases/download/v0.4.5/jri10-ev3.tar.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/116190587/3a598376-329c-11e8-85f1-883dcbbba34e?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180330%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180330T140518Z&X-Amz-Expires=300&X-Amz-Signature=1eddd75f8799693a50f2f73ee02660506f3aaf10cff0f3021812e36fe27d22b1&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Djri10-ev3.tar.gz&response-content-type=application%2Foctet-stream [following]
--2018-03-30 14:05:18--  https://github-production-release-asset-2e65be.s3.amazonaws.com/116190587/3a598376-329c-11e8-85f1-883dcbbba34e?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180330%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180330T140518Z&X-Amz-Expires=300&X-Amz-Signature=1eddd75f8799693a50f2f73ee02660506f3aaf10cff0f3021812e36fe27d22b1&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Djri10-ev3.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 54.231.49.226
[... 157 lines cut ...]
jri-ev3/legal/java.security.jgss/ASSEMBLY_EXCEPTION
jri-ev3/legal/java.security.jgss/ADDITIONAL_LICENSE_INFO
jri-ev3/legal/java.security.jgss/LICENSE
jri-ev3/legal/java.sql/
jri-ev3/legal/java.sql/ASSEMBLY_EXCEPTION
jri-ev3/legal/java.sql/ADDITIONAL_LICENSE_INFO
jri-ev3/legal/java.sql/LICENSE
jri-ev3/legal/java.sql.rowset/
jri-ev3/legal/java.sql.rowset/ASSEMBLY_EXCEPTION
jri-ev3/legal/java.sql.rowset/ADDITIONAL_LICENSE_INFO
jri-ev3/legal/java.sql.rowset/LICENSE
jri-ev3/legal/java.se/
jri-ev3/legal/java.se/ASSEMBLY_EXCEPTION
jri-ev3/legal/java.se/ADDITIONAL_LICENSE_INFO
jri-ev3/legal/java.se/LICENSE
jri-ev3/legal/jdk.jdwp.agent/
jri-ev3/legal/jdk.jdwp.agent/ASSEMBLY_EXCEPTION
jri-ev3/legal/jdk.jdwp.agent/ADDITIONAL_LICENSE_INFO
jri-ev3/legal/jdk.jdwp.agent/LICENSE
jri-ev3/release
update-alternatives: using /opt/jri-10-build-46/bin/java to provide /usr/bin/java (java) in auto mode
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode)
```

jabrena, 2018-03-30:
For LeJOS, it is a bit different @JakubVanek.

In LeJOS, the developer burn the SD and add the JRE/JRI in the process but in this project, the approach allow to update java several times because the JRE/JRI installation is decoupled with the process to burn the SO to run in the Brick.

closed 2018-03-30 by jabrena