ev3dev-lang-java/ev3dev-lang-java #226: Revivew BrickPi+ linux problems

jabrena, 2017-04-11:
``` 
[    4.222994] uart-pl011 3f201000.uart: no DMA platform data
[    4.345274] Registered BrickPi line discipline. (28)
[    4.405390] random: nonblocking pool is initialized
[    7.129514] systemd[1]: Unit brickman.service entered failed state.
[    7.401742] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.404971] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.521663] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    7.524576] brcmfmac: brcmf_add_if: ignore IF event
[    7.532941] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    7.535967] brcmfmac: power management disabled
[    8.228277] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   14.349549] tty ttyAMA0: Failed to init while setting sensors. (-110)
```

jabrena, 2017-04-11:
Hi @Dlech, I observed that today in some test, the system is quite slow and the system doesn´t detect the brickpi+ and the ssh connection require 2 tries to connect. Maybe problems with SD Card? how to diagnose? Any clue?

jabrena, 2017-04-11:
Does exist some way to restart the process with BrickPi+?

dlech, 2017-04-12:
Trying a different SD card sounds like a good idea.

jabrena, 2017-04-12:
But why the sd card was degraded? Does the sd card a number of cycles?

jabrena, 2017-04-12:
https://www.sdcard.org/developers/overview/speed_class/
https://superuser.com/questions/17350/whats-the-life-expectancy-of-an-sd-card

jabrena, 2017-04-12:
http://ideaheap.com/2013/07/stopping-sd-card-corruption-on-a-raspberry-pi/

jabrena, 2017-04-24:
I formatted the sdcard without any problem.
It is necessary to install `BatteryMonitor` to avoid this kind of problems.
https://github.com/ev3dev-lang-java/batteryMonitor

closed 2017-04-24 by jabrena