ev3dev-lang-java/ev3dev-lang-java #232: Compile OpenJDK 9 from Scratch

jabrena, 2017-04-15:
https://mihail.stoynov.com/2015/01/29/building-openjdk-9-on-a-osx-or-any-linux/

JakubVanek, 2017-12-26:
I am experimenting with JDK tree now and I've just found a configuration that might work:
```
bash configure --openjdk-target=arm-linux-gnueabi --enable-headless-only --with-freetype-lib=/usr/lib/arm-linux-gnueabi --with-freetype-include=/usr/include --with-jvm-variants=client --with-extra-cflags="-Os -march=armv5te -mtune=arm926ej-s"
```
While trying to satisfy all the build requirements, I've made a mess in the jessie-cross container, so I can't post the required packages to install. But otherwise, the trial-and-error works quite well.
- `--openjdk-target=arm-linux-gnueabi` → signals that this is  cross-compilation
- `--enable-headless-only` → actually, I've got no idea what the real implications are, but it makes sense
- `--with-freetype-lib=/usr/lib/arm-linux-gnueabi --with-freetype-include=/usr/include` → to find the armel variant of FreeType library.
- `--with-jvm-variants=client` → just a guess, it should make the program start faster
- `--with-extra-cflags="-Os -march=armv5te -mtune=arm926ej-s"` → optimize jvm for size and for the EV3 CPU; the latter is required, else the compiler defaults to armv4t, which is unsupported.
There are also a few interesting options worth thinking through: Minimal JVM and AOT compilation.

JakubVanek, 2017-12-26:
Apparently, `-Wno-maybe-uninitialized` is needed in CFLAGS, as otherwise the compilation stops with and error in `c1_LIRGenerator_arm.cpp`.

JakubVanek, 2017-12-26:
Problem: ASFLAGS does not contain extra CFLAGS
Solution: patch `common/autoconf/flags.m4` to:
```diff
diff -r a08cbfc0e4ec common/autoconf/flags.m4
--- a/common/autoconf/flags.m4	Thu Aug 03 18:56:56 2017 +0000
+++ b/common/autoconf/flags.m4	Tue Dec 26 21:48:34 2017 +0000
@@ -30,7 +30,7 @@
 AC_DEFUN([FLAGS_SETUP_ABI_PROFILE],
 [
   AC_ARG_WITH(abi-profile, [AS_HELP_STRING([--with-abi-profile],
-      [specify ABI profile for ARM builds (arm-vfp-sflt,arm-vfp-hflt,arm-sflt, armv5-vfp-sflt,armv6-vfp-hflt,arm64,aarch64) @<:@toolchain dependent@:>@ ])])
+      [specify ABI profile for ARM builds (arm-vfp-sflt,arm-vfp-hflt,arm-sflt,arm-ev3,armv5-vfp-sflt,armv6-vfp-hflt,arm64,aarch64) @<:@toolchain dependent@:>@ ])])
 
   if test "x$with_abi_profile" != x; then
     if test "x$OPENJDK_TARGET_CPU" != xarm && \
@@ -51,6 +51,9 @@
     elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarm-sflt; then
       ARM_FLOAT_TYPE=sflt
       ARM_ARCH_TYPE_FLAGS='-march=armv5t -marm'
+    elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarm-ev3; then
+      ARM_FLOAT_TYPE=sflt
+      ARM_ARCH_TYPE_FLAGS='-mcpu=arm926ej-s -marm'
     elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarmv5-vfp-sflt; then
       ARM_FLOAT_TYPE=vfp-sflt
       ARM_ARCH_TYPE_FLAGS='-march=armv5t -marm'
@@ -1305,6 +1308,8 @@
   if test "x$OPENJDK_$1_OS" = xlinux; then
     if test "x$OPENJDK_$1_CPU" = xx86; then
       $2JVM_ASFLAGS="[$]$2JVM_ASFLAGS -march=i586"
+    elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarm-ev3; then
+      $2JVM_ASFLAGS="-mcpu=arm926ej-s"
     fi
   elif test "x$OPENJDK_$1_OS" = xmacosx; then
     $2JVM_ASFLAGS="[$]$2JVM_ASFLAGS -x assembler-with-cpp -mno-omit-leaf-frame-pointer -mstack-alignment=16"
```
Current configure line:
```sh
bash ./configure --openjdk-target=arm-linux-gnueabi --with-abi-profile=arm-ev3 --enable-headless-only --with-freetype-lib=/usr/lib/arm-linux-gnueabi --with-freetype-include=/usr/include --with-jvm-variants=minimal --with-extra-cflags="-Os -Wno-maybe-uninitialized"
```
The Minimal JVM will be sufficient for the beginning. Also, we have to use the `--with-abi-profile` option (vanilla would use arm-sflt).

JakubVanek, 2017-12-26:
Problem: Java on softfloat fails linking on `__aeabi_dsub_glibc` and similar.
Cause/Solution: http://mail.openjdk.java.net/pipermail/aarch32-port-dev/2016-November/000616.html

jabrena, 2017-12-26:
Hi @JakubVanek, nice to read your comments.
I was exploring OpenJ9, this project has better support for ARM for EV3.
Are you interested to participate?

Cheers

JakubVanek, 2017-12-26:
Hi,
I'm not sure OpenJ9 has better support for ARM. [OpenJDK inherited the ARM port from the Oracle EJDK](https://www.voxxed.com/2016/10/oracle-open-sourcing-jdk-9-arm-ports/), which the [leJOS JRE](http://www.oracle.com/technetwork/java/embedded/downloads/javase/javaseemeddedev3-1982511.html) is using.

Regards

jabrena, 2017-12-26:
Interesting comment :)

Returning to previous point, so are you working to run OpenJDK 9 on EV3? What is the status? The idea is very promising. Run the JShell on the brick could be amazing:
https://docs.oracle.com/javase/9/jshell/introduction-jshell.htm#JSHEL-GUID-630F27C8-1195-4989-9F6B-2C51D46F52C8

JakubVanek, 2017-12-27:
Yes, I'm trying to get OpenJDK running on the EV3. It looks quite good - the build has finished successfully now. I can't test it at the moment though, because I don't have a spare microSD card.
I've uploaded my compressed images folder [here](https://drive.google.com/open?id=1js9A7BK0pi9QIzq7JYFWLOhuNWfagdSr), but beware: it might be needed to install libraries from jessie-backports (apt-mess in the container).

How to build:
```sh
## preparation:
# <setup docker container>
# <enter docker container>
# <enable backports (for JDK)>
# <install JDK8>
# <install armel libraries needed by JDK9, roughly these:
#   libcups2-dev:armel, libfreetype6-dev:armel, libasound2-dev:armel>

## build:
hg clone http://hg.openjdk.java.net/jdk9/jdk9 jdk9
cd jdk9
bash get_source.sh

# now save the build.patch and the float.patch to this directory
patch < build.patch # apply the build.patch to the *root jdk directory*
cd hotspot; patch < ../float.patch; cd .. # apply the float.patch to the *hotspot subdirectory*

# let's configure it
bash ./configure --openjdk-target=arm-linux-gnueabi \
    --with-abi-profile=arm-ev3 \
    --enable-headless-only \
    --with-freetype-lib=/usr/lib/arm-linux-gnueabi \
    --with-freetype-include=/usr/include \
    --with-jvm-variants=minimal \
    --with-extra-cflags="-Os -Wno-maybe-uninitialized -D__SOFTFP__" \
    --with-version-pre="mindstorms" \
    AR="arm-linux-gnueabi-gcc-ar" \
    NM="arm-linux-gnueabi-gcc-nm" \
    BUILD_AR="gcc-ar" \
    BUILD_NM="gcc-nm"

# let's build it
make clean images # hmm, this sounds nice
# j2sdk images should be in build/linux-arm-normal-minimal-release/images
```
build.patch:
```diff
diff -r a08cbfc0e4ec common/autoconf/flags.m4
--- a/common/autoconf/flags.m4	Thu Aug 03 18:56:56 2017 +0000
+++ b/common/autoconf/flags.m4	Wed Dec 27 00:52:44 2017 +0100
@@ -30,7 +30,7 @@
 AC_DEFUN([FLAGS_SETUP_ABI_PROFILE],
 [
   AC_ARG_WITH(abi-profile, [AS_HELP_STRING([--with-abi-profile],
-      [specify ABI profile for ARM builds (arm-vfp-sflt,arm-vfp-hflt,arm-sflt, armv5-vfp-sflt,armv6-vfp-hflt,arm64,aarch64) @<:@toolchain dependent@:>@ ])])
+      [specify ABI profile for ARM builds (arm-vfp-sflt,arm-vfp-hflt,arm-sflt,arm-ev3,armv5-vfp-sflt,armv6-vfp-hflt,arm64,aarch64) @<:@toolchain dependent@:>@ ])])
 
   if test "x$with_abi_profile" != x; then
     if test "x$OPENJDK_TARGET_CPU" != xarm && \
@@ -51,6 +51,9 @@
     elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarm-sflt; then
       ARM_FLOAT_TYPE=sflt
       ARM_ARCH_TYPE_FLAGS='-march=armv5t -marm'
+    elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarm-ev3; then
+      ARM_FLOAT_TYPE=sflt
+      ARM_ARCH_TYPE_FLAGS='-mcpu=arm926ej-s -marm'
     elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarmv5-vfp-sflt; then
       ARM_FLOAT_TYPE=vfp-sflt
       ARM_ARCH_TYPE_FLAGS='-march=armv5t -marm'
@@ -1305,6 +1308,8 @@
   if test "x$OPENJDK_$1_OS" = xlinux; then
     if test "x$OPENJDK_$1_CPU" = xx86; then
       $2JVM_ASFLAGS="[$]$2JVM_ASFLAGS -march=i586"
+    elif test "x$OPENJDK_TARGET_ABI_PROFILE" = xarm-ev3; then
+      $2JVM_ASFLAGS="-mcpu=arm926ej-s"
     fi
   elif test "x$OPENJDK_$1_OS" = xmacosx; then
     $2JVM_ASFLAGS="[$]$2JVM_ASFLAGS -x assembler-with-cpp -mno-omit-leaf-frame-pointer -mstack-alignment=16"
```
float.patch:
```diff
diff -r b756e7a2ec33 src/cpu/arm/vm/assembler_arm_32.hpp
--- a/src/cpu/arm/vm/assembler_arm_32.hpp	Thu Aug 03 18:56:57 2017 +0000
+++ b/src/cpu/arm/vm/assembler_arm_32.hpp	Wed Dec 27 01:08:37 2017 +0100
@@ -1234,10 +1234,17 @@
 
 // Imported code from glibc soft-fp bundle for
 // calculation accuracy improvement. See CR 6757269.
-extern double __aeabi_fadd_glibc(float, float);
-extern double __aeabi_fsub_glibc(float, float);
-extern double __aeabi_dadd_glibc(double, double);
-extern double __aeabi_dsub_glibc(double, double);
+
+//extern double __aeabi_fadd_glibc(float, float);
+//extern double __aeabi_fsub_glibc(float, float);
+//extern double __aeabi_dadd_glibc(double, double);
+//extern double __aeabi_dsub_glibc(double, double);
+
+// LinuxTARDIS hack
+#define __aeabi_fadd_glibc __aeabi_fadd
+#define __aeabi_fsub_glibc __aeabi_fsub
+#define __aeabi_dadd_glibc __aeabi_dadd
+#define __aeabi_dsub_glibc __aeabi_dsub
 };
 #endif // __SOFTFP__
 ```
Purpose of float.patch: `_glibc` overloads are supposed to be more precise. On EV3 this is not worth any better hack, so we redirect them to the default implementation.
Additional build flags:
- `-D__SOFTFP__` cflag - I've found this somewhere on the internet, it probably switches the JIT float implementation
- `--with-version-pre="mindstorms"` - just for fun, changes the version string from `9-internal` to `9-mindstorms`
- `AR="arm-linux-gnueabi-gcc-ar" NM="arm-linux-gnueabi-gcc-nm" BUILD_AR="gcc-ar" BUILD_NM="gcc-nm"` - removes warnings about LTO objects in archives
Java version numbers are not handled well, e.g. now I have 9-mindstorms+0-adhoc..jdk9.

JakubVanek, 2017-12-27:
Hmm, haven't heard of JShell yet, but it looks interesting. It should be included in linked build.

jabrena, 2017-12-27:
JShell for educational purposes could be nice.

jabrena, 2017-12-27:
Welcome to this Java project for Mindstorms!!!

JakubVanek, 2017-12-27:
Hi,

Thanks for the admission :)

I've tried running my OpenJDK9 in userspace QEMU, but with disappointing results.
When Java is started in the JIT mode, QEMU locks up, utilizes 100% of one core and responds only to SIGKILL.
```sh
$ qemu-arm-static -cpu arm926 /lib/arm-linux-gnueabi/ld-linux.so.3 ./java -version
```
When interpreted mode is activated, everything is OK:
```sh
$ qemu-arm-static -cpu arm926 /lib/arm-linux-gnueabi/ld-linux.so.3 ./java -Xint -version
openjdk version "9-mindstorms"
OpenJDK Runtime Environment (build 9-mindstorms+0-adhoc..jdk9)
OpenJDK Minimal VM (build 9-mindstorms+0-adhoc..jdk9, interpreted mode)
```
It *might* work on the brick, if it's only QEMU what has problems with the JIT.

I'll try to compile OpenJ9 for ARM to see, whether we have at least some alternative. :smile:

jabrena, 2017-12-28:
Try try, the idea is build a stable version for the whole community. It is a good challenge.

Cheers

JakubVanek, 2018-01-03:
YEEAH, it works!
JIT-ed version query works:
```sh
robot@ev3dev:~/images/jre/bin$ ./java -verbose -version
[0.186s][info][class,load] opened: /home/robot/images/jre/lib/modules
[0.270s][info][class,load] java.lang.Object source: jrt:/java.base
[0.282s][info][class,load] java.io.Serializable source: jrt:/java.base
[0.286s][info][class,load] java.lang.Comparable source: jrt:/java.base
[0.293s][info][class,load] java.lang.CharSequence source: jrt:/java.base
[0.305s][info][class,load] java.lang.String source: jrt:/java.base
[0.325s][info][class,load] java.lang.reflect.AnnotatedElement source: jrt:/java.base
[0.331s][info][class,load] java.lang.reflect.GenericDeclaration source: jrt:/java.base
[0.335s][info][class,load] java.lang.reflect.Type source: jrt:/java.base
[0.352s][info][class,load] java.lang.Class source: jrt:/java.base
[0.358s][info][class,load] java.lang.Cloneable source: jrt:/java.base
[0.378s][info][class,load] java.lang.ClassLoader source: jrt:/java.base
[0.391s][info][class,load] java.lang.System source: jrt:/java.base
[0.401s][info][class,load] java.lang.Throwable source: jrt:/java.base
[0.405s][info][class,load] java.lang.Error source: jrt:/java.base
[0.411s][info][class,load] java.lang.ThreadDeath source: jrt:/java.base
[0.416s][info][class,load] java.lang.Exception source: jrt:/java.base
[0.422s][info][class,load] java.lang.RuntimeException source: jrt:/java.base
[0.435s][info][class,load] java.lang.SecurityManager source: jrt:/java.base
[0.443s][info][class,load] java.security.ProtectionDomain source: jrt:/java.base
[0.452s][info][class,load] java.security.AccessControlContext source: jrt:/java.base
[0.458s][info][class,load] java.security.SecureClassLoader source: jrt:/java.base
[0.465s][info][class,load] java.lang.ReflectiveOperationException source: jrt:/java.base
[0.470s][info][class,load] java.lang.ClassNotFoundException source: jrt:/java.base
[0.475s][info][class,load] java.lang.LinkageError source: jrt:/java.base
[0.480s][info][class,load] java.lang.NoClassDefFoundError source: jrt:/java.base
[0.485s][info][class,load] java.lang.ClassCastException source: jrt:/java.base
[0.490s][info][class,load] java.lang.ArrayStoreException source: jrt:/java.base
[0.495s][info][class,load] java.lang.VirtualMachineError source: jrt:/java.base
[0.500s][info][class,load] java.lang.OutOfMemoryError source: jrt:/java.base
[0.507s][info][class,load] java.lang.StackOverflowError source: jrt:/java.base
[0.515s][info][class,load] java.lang.IllegalMonitorStateException source: jrt:/java.base
[0.523s][info][class,load] java.lang.ref.Reference source: jrt:/java.base
[0.528s][info][class,load] java.lang.ref.SoftReference source: jrt:/java.base
[0.533s][info][class,load] java.lang.ref.WeakReference source: jrt:/java.base
[0.538s][info][class,load] java.lang.ref.FinalReference source: jrt:/java.base
[0.543s][info][class,load] java.lang.ref.PhantomReference source: jrt:/java.base
[0.551s][info][class,load] java.lang.ref.Finalizer source: jrt:/java.base
[0.560s][info][class,load] java.lang.Runnable source: jrt:/java.base
[0.571s][info][class,load] java.lang.Thread source: jrt:/java.base
[0.581s][info][class,load] java.lang.Thread$UncaughtExceptionHandler source: jrt:/java.base
[0.588s][info][class,load] java.lang.ThreadGroup source: jrt:/java.base
[0.610s][info][class,load] java.util.Map source: jrt:/java.base
[0.621s][info][class,load] java.util.Dictionary source: jrt:/java.base
[0.625s][info][class,load] java.util.Hashtable source: jrt:/java.base
[0.633s][info][class,load] java.util.Properties source: jrt:/java.base
[0.648s][info][class,load] java.lang.Module source: jrt:/java.base
[0.661s][info][class,load] java.lang.reflect.AccessibleObject source: jrt:/java.base
[0.670s][info][class,load] java.lang.reflect.Member source: jrt:/java.base
[0.679s][info][class,load] java.lang.reflect.Field source: jrt:/java.base
[0.687s][info][class,load] java.lang.reflect.Parameter source: jrt:/java.base
[0.704s][info][class,load] java.lang.reflect.Executable source: jrt:/java.base
[0.710s][info][class,load] java.lang.reflect.Method source: jrt:/java.base
[0.723s][info][class,load] java.lang.reflect.Constructor source: jrt:/java.base
[0.727s][info][class,load] jdk.internal.reflect.MagicAccessorImpl source: jrt:/java.base
[0.733s][info][class,load] jdk.internal.reflect.MethodAccessor source: jrt:/java.base
[0.736s][info][class,load] jdk.internal.reflect.MethodAccessorImpl source: jrt:/java.base
[0.743s][info][class,load] jdk.internal.reflect.ConstructorAccessor source: jrt:/java.base
[0.747s][info][class,load] jdk.internal.reflect.ConstructorAccessorImpl source: jrt:/java.base
[0.752s][info][class,load] jdk.internal.reflect.DelegatingClassLoader source: jrt:/java.base
[0.761s][info][class,load] jdk.internal.reflect.ConstantPool source: jrt:/java.base
[0.774s][info][class,load] jdk.internal.reflect.FieldAccessor source: jrt:/java.base
[0.780s][info][class,load] jdk.internal.reflect.FieldAccessorImpl source: jrt:/java.base
[0.784s][info][class,load] jdk.internal.reflect.UnsafeFieldAccessorImpl source: jrt:/java.base
[0.788s][info][class,load] jdk.internal.reflect.UnsafeStaticFieldAccessorImpl source: jrt:/java.base
[0.794s][info][class,load] java.lang.annotation.Annotation source: jrt:/java.base
[... 418 lines cut ...]
	at java.management/java.lang.management.ManagementFactory.lambda$static$8(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
[... cut ...]
	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:124)
	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:412)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:357)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:383)
	at ev3dev.hardware.EV3DevFileSystem.<clinit>(EV3DevFileSystem.java:15)
	at ev3dev.misc.BumperCar.<clinit>(BumperCar.java:19)
```
Recompiling with more features enabled should help (and should make more benchmarking data available). Let's try! :smile:

JakubVanek, 2018-01-03:
Hmm, JShell refuses to work on minimal JVM as well:
```
robot@ev3dev:~/images/jdk/bin$ ./jshell 
Jan 03, 2018 6:20:01 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
Launching JShell execution engine threw: Debugging agents are not supported in this VM
Error: Could not create the Java Virtual Machine.
Error: A fatal exception has occurred. Program will exit.
```

jabrena, 2018-01-03:
Do you like if we create a new repo to store the stable version of OpenJDK for EV3 on it? We could have a folder for OpenJ9 and OpenJDK?

jabrena, 2018-01-03:
`JShell` is a nice feature to test on EV3. It is a key feature for Java 9.
In few days, I could join for the tests.

jabrena, 2018-01-03:
**Idea:**
Using JShell, it could be possible to load latest library and dependencies on JShell to interact with the robot for experiments. It could be nice in terms of education.

JakubVanek, 2018-01-03:
https://drive.google.com/open?id=1jzRwYsR_Ya4hk-eqVlp0ACNpCo68DiOA
But this build is really an overkill. The configure line:
```
bash ./configure --openjdk-target=arm-linux-gnueabi --with-abi-profile=arm-ev3 --disable-headless-only --with-freetype-lib=/usr/lib/arm-linux-gnueabi --with-freetype-include=/usr/include --with-jvm-variants=server,client,minimal --enable-ccache --with-extra-cflags="-Os -Wno-maybe-uninitialized -D__SOFTFP__" --with-version-pre="mindstorms" AR="arm-linux-gnueabi-gcc-ar" NM="arm-linux-gnueabi-gcc-nm" BUILD_AR="gcc-ar" BUILD_NM="gcc-nm"
```
- The server VM doesn't work on ARM < v7. 
- We're probably not going to need non-headless configuration.

Here are results from the JVM benchmark:
```sh
robot@ev3dev:~$ java -client -jar jvm-benchmark-all-0.1.0.jar 
GeneralBench 1.2
       6 byte[16384] manual copies:        97 ms    1013443 bytes/sec
    1220 byte[16384] arraycopies:         151 ms  132374039 bytes/sec
      24 int[4096] manual copies:         105 ms    3744914 bytes/sec
    1220 int[4096] arraycopies:           145 ms  137851586 bytes/sec
  200000 byte add:                        100 ms    2000000 ops/sec
  200000 byte sub:                          2 ms  100000000 ops/sec
  200000 byte mul:                          3 ms   66666666 ops/sec
  200000 byte div:                         34 ms    5882352 ops/sec
  200000 short add:                        97 ms    2061855 ops/sec
  200000 short sub:                         1 ms  200000000 ops/sec
  200000 short mul:                         2 ms  100000000 ops/sec
  200000 short div:                        41 ms    4878048 ops/sec
  200000 char add:                         97 ms    2061855 ops/sec
  200000 char sub:                          2 ms  100000000 ops/sec
  200000 char mul:                          2 ms  100000000 ops/sec
  200000 char div:                         36 ms    5555555 ops/sec
  200000 int add:                          92 ms    2173913 ops/sec
  200000 int sub:                           1 ms  200000000 ops/sec
  200000 int mul:                           1 ms  200000000 ops/sec
  200000 int div:                          76 ms    2631578 ops/sec
  200000 long add:                        299 ms     668896 ops/sec
  200000 long sub:                        167 ms    1197604 ops/sec
  200000 long mul:                          1 ms  200000000 ops/sec
  200000 long div:                        159 ms    1257861 ops/sec
  200000 float add:                       144 ms    1388888 ops/sec
  200000 float sub:                         1 ms  200000000 ops/sec
  200000 float mul:                         1 ms  200000000 ops/sec
  200000 float div:                         1 ms  200000000 ops/sec
  200000 double add:                      169 ms    1183431 ops/sec
  200000 double sub:                        1 ms  200000000 ops/sec
  200000 double mul:                        1 ms  200000000 ops/sec
  200000 double div:                        1 ms  200000000 ops/sec
  200000 method calls:                    103 ms    1941747 ops/sec
  200000 static method calls:             102 ms    1960784 ops/sec
    2000 string concats:                  598 ms       3344 ops/sec
   20000 string compares (easy):          122 ms     163934 ops/sec
    1000 string compares (hard):            6 ms     166666 ops/sec
   20000 object creations:                183 ms     109289 ops/sec
 6242048 Total Loop Executions:          6177 ms    1010530 loops/sec
Note: each Loop Execution includes multiple Java operations
```
```sh
robot@ev3dev:~$ java -minimal -jar jvm-benchmark-all-0.1.0.jar 
GeneralBench 1.2
       6 byte[16384] manual copies:        81 ms    1213629 bytes/sec
    1220 byte[16384] arraycopies:         129 ms  154949457 bytes/sec
      24 int[4096] manual copies:         107 ms    3674915 bytes/sec
    1220 int[4096] arraycopies:           133 ms  150289323 bytes/sec
  200000 byte add:                         96 ms    2083333 ops/sec
  200000 byte sub:                          1 ms  200000000 ops/sec
  200000 byte mul:                          3 ms   66666666 ops/sec
  200000 byte div:                         34 ms    5882352 ops/sec
  200000 short add:                        97 ms    2061855 ops/sec
  200000 short sub:                         2 ms  100000000 ops/sec
  200000 short mul:                         1 ms  200000000 ops/sec
  200000 short div:                        41 ms    4878048 ops/sec
  200000 char add:                         97 ms    2061855 ops/sec
  200000 char sub:                          2 ms  100000000 ops/sec
[... 91 lines cut ...]
  200000 int sub:                          47 ms    4255319 ops/sec
  200000 int mul:                          49 ms    4081632 ops/sec
  200000 int div:                          80 ms    2500000 ops/sec
  200000 long add:                         59 ms    3389830 ops/sec
  200000 long sub:                         59 ms    3389830 ops/sec
  200000 long mul:                         83 ms    2409638 ops/sec
  200000 long div:                        204 ms     980392 ops/sec
  200000 float add:                       102 ms    1960784 ops/sec
  200000 float sub:                       103 ms    1941747 ops/sec
  200000 float mul:                        83 ms    2409638 ops/sec
  200000 float div:                       159 ms    1257861 ops/sec
  200000 double add:                      141 ms    1418439 ops/sec
  200000 double sub:                      147 ms    1360544 ops/sec
[... cut ...]
    2000 string concats:                  592 ms       3378 ops/sec
   20000 string compares (easy):          405 ms      49382 ops/sec
    1000 string compares (hard):            3 ms     333333 ops/sec
   20000 object creations:                100 ms     200000 ops/sec
 6242048 Total Loop Executions:         10020 ms     622958 loops/sec
Note: each Loop Execution includes multiple Java operations
```
And the ev3dev samples might work now, it crashes on that it can't find a lego-* class directory in sysfs (I don't have any sensors plugged in, so it's OK).

JakubVanek, 2018-01-03:
I agree with the idea of creating a repository for it. I think creating some Dockerfiles and storing builds through Releases might work well.

jabrena, 2018-01-03:
any suggestion for the name?, Tomorrow, I will create a repo and I will give grants to upload the distro. Did you run `JShell` finally?

The idea to use a `Dockerfile` is perfect to automate the process. We could build in automatic way with Travis and create snapshots in the same way that EV3Dev:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/

But little by little.

Juan Antonio

jabrena, 2018-01-03:
Questions:

* The server VM doesn't work on ARM < v7. Why?
* What is the difference between client & minimal? I didn´t use that flag in the past.
* We're probably not going to need non-headless configuration. Any Java process running on EV3 will only use non visual elements.

jabrena, 2018-01-03:
Fetching the results from Oracle JRE 8:

```
GeneralBench 1.2
       6 byte[16384] manual copies:        87 ms    1129931 bytes/sec
    1220 byte[16384] arraycopies:         162 ms  123385679 bytes/sec
      24 int[4096] manual copies:         111 ms    3542486 bytes/sec
    1220 int[4096] arraycopies:           144 ms  138808888 bytes/sec
  200000 byte add:                         96 ms    2083333 ops/sec
  200000 byte sub:                          9 ms   22222222 ops/sec
  200000 byte mul:                          3 ms   66666666 ops/sec
  200000 byte div:                         35 ms    5714285 ops/sec
  200000 short add:                       116 ms    1724137 ops/sec
  200000 short sub:                         2 ms  100000000 ops/sec
  200000 short mul:                         1 ms  200000000 ops/sec
  200000 short div:                        47 ms    4255319 ops/sec
  200000 char add:                        119 ms    1680672 ops/sec
  200000 char sub:                          2 ms  100000000 ops/sec
  200000 char mul:                          2 ms  100000000 ops/sec
  200000 char div:                         41 ms    4878048 ops/sec
  200000 int add:                         105 ms    1904761 ops/sec
  200000 int sub:                           2 ms  100000000 ops/sec
  200000 int mul:                           1 ms  200000000 ops/sec
  200000 int div:                          55 ms    3636363 ops/sec
  200000 long add:                        108 ms    1851851 ops/sec
  200000 long sub:                         -1 ms -200000000 ops/sec
  200000 long mul:                         -2 ms -100000000 ops/sec
  200000 long div:                        357 ms     560224 ops/sec
  200000 float add:                       238 ms     840336 ops/sec
  200000 float sub:                         1 ms  200000000 ops/sec
  200000 float mul:                         1 ms  200000000 ops/sec
  200000 float div:                         1 ms  200000000 ops/sec
  200000 double add:                      315 ms     634920 ops/sec
  200000 double sub:                        1 ms  200000000 ops/sec
  200000 double mul:                        1 ms  200000000 ops/sec
  200000 double div:                        1 ms  200000000 ops/sec
  200000 method calls:                     85 ms    2352941 ops/sec
  200000 static method calls:              75 ms    2666666 ops/sec
    2000 string concats:                  491 ms       4073 ops/sec
   20000 string compares (easy):           62 ms     322580 ops/sec
    1000 string compares (hard):            3 ms     333333 ops/sec
   20000 object creations:                135 ms     148148 ops/sec
 6242048 Total Loop Executions:          5200 ms    100393 loops/sec
```

JakubVanek, 2018-01-03:
Yay! JShell is working! But there's an issue with it: it takes  _forever_ to load. And it takes a lot of memory.
[image]
The kills are me losing the patience with slow startup.
And... oh... it takes several seconds before the typed letters start to appear on the prompt.

jabrena, 2018-01-03:
Do you have Twitter account?

jabrena, 2018-01-03:
Can you paste the result of this command when JShell is executed?

```
ps aux | sort -n -k 4
```

jabrena, 2018-01-03:
A proposal for the future repository to store the builds about OpenJDK 9 builds: `openjdk-9-ev3`
What is your opinion @JakubVanek ?

Juan Antonio

jabrena, 2018-01-03:
What is the content of the folder /bin after the compilation?
How many commands is possible to execute?

jabrena, 2018-01-03:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/220

JakubVanek, 2018-01-03:
> any suggestion for the name?

I don't know, but `jre` sound OK.

EDIT: I'm too slow. :smile: Yeah, `openjdk-9-ev3` is more informational.

> The server VM doesn't work on ARM < v7. Why?

The VM refuses to start, printing this error (there should be `-J-client -R-client` or minimal):
```
robot@ev3dev:~/images/jdk/bin$ ./jshell
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
```
The C2 JIT was probably designed with newer server-grade hardware in mind.

> What is the difference between client & minimal? I didn´t use that flag in the past.

It uses different JVM variants. I actually don't know the actual difference, but [this article](https://medium.com/@metebalci/demystifying-the-jvm-jvm-variants-cppinterpreter-and-templateinterpreter-9f7d5b781850) gives me some idea. At least one of the flags is needed on this build, because the JVM default to the broken server JIT.

> We're probably not going to need non-headless configuration. Any Java process running on EV3 will only use non visual elements.

Well, the only option probably would be to tunnel X11 through SSH. 

> Do you have Twitter account?

Yeah, https://twitter.com/sileny_linuxak.

> jshell

I have started it as `./jshell -J-minimal -R-minimal --execution local`
```sh
robot@ev3dev:~$ ps aux | sort -n -k 4
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
avahi      251  0.0  0.0   3840    44 ?        Ss   22:36   0:00 avahi-daemon: running [ev3dev.local]
avahi      269  0.0  0.0   3840    44 ?        S    22:36   0:00 avahi-daemon: chroot helper
message+   259  0.2  0.0   5688    44 ?        Ss   22:36   0:01 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
robot      490  0.0  0.0   4952    44 ?        Ss   22:39   0:00 /lib/systemd/systemd --user
robot      493  0.0  0.0  25124    40 ?        S    22:39   0:00 (sd-pam)         
robot      496  0.1  0.0  12168    44 ?        S    22:39   0:00 sshd: robot@pts/0   
robot      498  0.0  0.0   5160    20 pts/0    Ss   22:39   0:00 -bash
root         2  0.0  0.0      0     0 ?        S    22:35   0:00 [kthreadd]
root         3  0.5  0.0      0     0 ?        S    22:35   0:03 [ksoftirqd/0]
root         5  0.0  0.0      0     0 ?        S<   22:35   0:00 [kworker/0:0H]
root         6  0.0  0.0      0     0 ?        S    22:35   0:00 [kworker/u2:0]
root         7  0.4  0.0      0     0 ?        S    22:35   0:02 [rcu_preempt]
root         8  0.0  0.0      0     0 ?        S    22:35   0:00 [rcu_sched]
root         9  0.0  0.0      0     0 ?        S    22:35   0:00 [rcu_bh]
root        10  0.0  0.0      0     0 ?        S    22:35   0:00 [kdevtmpfs]
root        11  0.0  0.0      0     0 ?        S<   22:35   0:00 [netns]
root        12  0.0  0.0      0     0 ?        S<   22:35   0:00 [writeback]
root        13  0.0  0.0      0     0 ?        S<   22:35   0:00 [crypto]
root        14  0.0  0.0      0     0 ?        S<   22:35   0:00 [bioset]
root        15  0.0  0.0      0     0 ?        S<   22:35   0:00 [kblockd]
root        16  0.3  0.0      0     0 ?        S    22:35   0:02 [kworker/0:1]
root        17  0.0  0.0      0     0 ?        S<   22:35   0:00 [rpciod]
root        18  4.3  0.0      0     0 ?        S    22:35   0:29 [kswapd0]
root        19  0.0  0.0      0     0 ?        S    22:35   0:00 [fsnotify_mark]
root        20  0.0  0.0      0     0 ?        S<   22:35   0:00 [nfsiod]
root        43  0.0  0.0      0     0 ?        S<   22:35   0:00 [bioset]
root        44  0.0  0.0      0     0 ?        S<   22:35   0:00 [bioset]
root        45  0.0  0.0      0     0 ?        S<   22:35   0:00 [bioset]
root        46  0.0  0.0      0     0 ?        S<   22:35   0:00 [bioset]
root        47  0.0  0.0      0     0 ?        S<   22:35   0:00 [bioset]
root        48  0.0  0.0      0     0 ?        S    22:35   0:00 [irq/56-spi_davi]
root        49  0.0  0.0      0     0 ?        S    22:35   0:00 [spi1]
root        50  0.0  0.0      0     0 ?        S    22:35   0:00 [irq/20-spi_davi]
root        51  5.5  0.0      0     0 ?        S    22:35   0:37 [spi0]
root        52  0.0  0.0      0     0 ?        S    22:35   0:00 [kworker/0:2]
root        54  0.0  0.0      0     0 ?        S<   22:35   0:00 [deferwq]
[... 7 lines cut ...]
root       321  0.0  0.0      0     0 ?        S<   22:36   0:00 [kworker/u3:0]
root       322  0.0  0.0      0     0 ?        S<   22:36   0:00 [hci0]
root       323  0.0  0.0      0     0 ?        S<   22:36   0:00 [hci0]
root       326  0.0  0.0      0     0 ?        S<   22:36   0:00 [kworker/u3:2]
root       328  0.0  0.0   2004    40 ?        S<   22:36   0:00 /usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep
root       330  0.1  0.0   7912    48 ?        Ss   22:36   0:01 /usr/sbin/connmand -n
root       331  0.0  0.0   4996    48 ?        Ss   22:36   0:00 /usr/lib/bluetooth/bluetoothd
root       347  0.0  0.0   7912    44 ?        Ss   22:36   0:00 /usr/sbin/sshd -D
root       355  0.0  0.0   6908    44 ?        Ss   22:36   0:00 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
root       485  0.1  0.0  12168    44 ?        Ss   22:39   0:00 sshd: robot [priv]  
root       503  0.1  0.0  12168    44 ?        Ss   22:39   0:00 sshd: robot [priv]  
root       596  0.0  0.0      0     0 ?        S    22:45   0:00 [kworker/0:0]
root       613  0.2  0.0      0     0 ?        S    22:46   0:00 [kworker/u2:1]
[... cut ...]
root       343  1.8  1.5  53376   916 tty1     Ssl+ 22:36   0:12 /usr/sbin/brickman
root       457  0.1  1.7  21020  1008 ?        Ss   22:37   0:01 /usr/sbin/nmbd -D
robot      617 12.0  2.3   6772  1404 pts/1    S+   22:47   0:00 sort -n -k 4
root         1  1.4  2.3  23760  1400 ?        Ds   22:35   0:09 /sbin/init splash
robot      616 45.0  2.9   4652  1732 pts/1    R+   22:47   0:00 ps aux
robot      576 73.1 31.4 100664 18420 pts/0    Sl+  22:42   3:27 ./jshell -J-minimal -R-minimal --execution local
```

jabrena, 2018-01-03:
And the result of:

``` 
free -m
```

JakubVanek, 2018-01-03:
It is now trying to do something with `int x = 0`. It has already taken a few minutes.
```
robot@ev3dev:~$ free -m
             total       used       free     shared    buffers     cached
Mem:            57         55          1          0          1          2
-/+ buffers/cache:         51          5
Swap:           95         43         52
```
The system is swapping to death, as far as I can tell.

JakubVanek, 2018-01-03:
JShell may provide some "split-mode", when the executing JVM would be on the brick, whether the compiling JVM would be on a computer.
http://hg.openjdk.java.net/jdk9/jdk9/langtools/file/65bfdabaab9c/src/jdk.jshell/share/classes/jdk/jshell/spi/package-info.java

jabrena, 2018-01-03:
I have just created the new repo:
https://github.com/ev3dev-lang-java/openjdk-9-ev3

I added permissions to work on in directly.

How many seconds to run a simple example:
https://github.com/ev3dev-lang-java/template_project_gradle

JakubVanek, 2018-01-03:
I'll take a look tomorrow, I should give myself some sleep.

jabrena, 2018-01-03:
Yes, it could be interesting that Split mode, but if you like we return to the beginning, execution of Java 9 bytecodes. With Oracle JRE8, the time to run a Hello World is 8 seconds, with the OpenJDK 9, what is the time?

Good night mate.

You are the man!!!

Cheers

Juan Antonio

JakubVanek, 2018-01-04:
```
robot@ev3dev:~$ echo 3 | sudo tee /proc/sys/vm/drop_caches 
3
robot@ev3dev:~$ time images/jre/bin/java -client -Xshare:on -jar template_project_gradle-all-0.2.0.jar; time images/jre/bin/java -client -Xshare:on -jar template_project_gradle-all-0.2.0.jar; time images/jre/bin/java -client -Xshare:on -jar template_project_gradle-all-0.2.0.jar
Creating Motor A & B
2018-01-04 06:09:55 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 06:09:55 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outA
2018-01-04 06:09:55 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 06:09:56 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port4/address
2018-01-04 06:09:56 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-01-04 06:09:56 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 06:09:57 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor4/address
2018-01-04 06:09:57 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 06:09:57 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outB
2018-01-04 06:09:57 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 06:09:57 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port5/address
2018-01-04 06:09:57 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-01-04 06:09:57 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 06:09:57 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor5/address
Defining the Stop mode
Defining motor speed
Go Forward with the motors
Stop motors
Go Backward with the motors
Stop motors
Checking Battery
2018-01-04 06:10:02 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
Votage: 8.4038
Emergency Stop

real	1m11.152s
user	0m46.600s
sys	0m2.360s
Creating Motor A & B
2018-01-04 06:11:00 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 06:11:01 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outA
2018-01-04 06:11:01 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 06:11:01 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port4/address
2018-01-04 06:11:01 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-01-04 06:11:02 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 06:11:02 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor6/address
2018-01-04 06:11:02 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 06:11:02 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outB
2018-01-04 06:11:02 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 06:11:02 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port5/address
2018-01-04 06:11:02 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-01-04 06:11:03 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 06:11:03 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor7/address
Defining the Stop mode
Defining motor speed
Go Forward with the motors
Stop motors
Go Backward with the motors
Stop motors
Checking Battery
2018-01-04 06:11:07 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
Votage: 8.4038
Emergency Stop

real	1m5.363s
user	0m45.070s
sys	0m1.990s
Creating Motor A & B
2018-01-04 06:12:04 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 06:12:05 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outA
2018-01-04 06:12:05 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 06:12:05 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port4/address
2018-01-04 06:12:05 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-01-04 06:12:06 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 06:12:06 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor8/address
2018-01-04 06:12:06 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 06:12:06 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outB
2018-01-04 06:12:06 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 06:12:06 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port5/address
2018-01-04 06:12:06 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
2018-01-04 06:12:06 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 06:12:07 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor9/address
Defining the Stop mode
Defining motor speed
Go Forward with the motors
Stop motors
Go Backward with the motors
Stop motors
Checking Battery
2018-01-04 06:12:11 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
Votage: 8.4038
Emergency Stop

real	1m3.652s
user	0m44.800s
sys	0m1.830s
robot@ev3dev:~$ 
```
It takes quite a long time. There's a noticeable delay when loading the motors -- additional classes are loaded.
I have to run to school :smile:

jabrena, 2018-01-04:
Can you test a Hello World?
https://introcs.cs.princeton.edu/java/11hello/HelloWorld.java.html

jabrena, 2018-01-04:
Yes, that behaviour is normal, when the jvm begin, it has to load classes. My question is the number of seconds from the execution of the command until the first log message. It should be in 8 seconds or less. This is my question.

jabrena, 2018-01-04:
Add this flag to show classes loading:
https://stackoverflow.com/questions/10230279/java-verbose-class-loading

JakubVanek, 2018-01-04:
Hmm, Minimal JVM is faster: 
```
robot@ev3dev:~$ echo 3 | sudo tee /proc/sys/vm/drop_caches 
3
robot@ev3dev:~$ time images/jre/bin/java -minimal -jar template_project_gradle-all-0.2.0.jar; time images/jre/bin/java -minimal -jar template_project_gradle-all-0.2.0.jar; time images/jre/bin/java -minimal -jar template_project_gradle-all-0.2.0.jar
Creating Motor A & B
Exception in thread "main" java.lang.InternalError: Unsupported Management version
	at java.base/java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.base/java.lang.ClassLoader.loadLibrary0(Unknown Source)
	at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.base/java.lang.Runtime.loadLibrary0(Unknown Source)
	at java.base/java.lang.System.loadLibrary(Unknown Source)
	at java.management/java.lang.management.ManagementFactory.lambda$static$8(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.management/java.lang.management.ManagementFactory.<clinit>(Unknown Source)
	at ch.qos.logback.classic.joran.action.JMXConfiguratorAction.begin(JMXConfiguratorAction.java:61)
	at ch.qos.logback.core.joran.spi.Interpreter.callBeginAction(Interpreter.java:269)
	at ch.qos.logback.core.joran.spi.Interpreter.startElement(Interpreter.java:145)
	at ch.qos.logback.core.joran.spi.Interpreter.startElement(Interpreter.java:128)
	at ch.qos.logback.core.joran.spi.EventPlayer.play(EventPlayer.java:50)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:165)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:152)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:110)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:53)
	at ch.qos.logback.classic.util.ContextInitializer.configureByResource(ContextInitializer.java:75)
	at ch.qos.logback.classic.util.ContextInitializer.autoConfig(ContextInitializer.java:150)
	at org.slf4j.impl.StaticLoggerBinder.init(StaticLoggerBinder.java:84)
	at org.slf4j.impl.StaticLoggerBinder.<clinit>(StaticLoggerBinder.java:55)
	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:150)
	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:124)
	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:412)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:357)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:383)
	at ev3dev.hardware.EV3DevFileSystem.<clinit>(EV3DevFileSystem.java:15)
	at examples.MyFirstRobot.main(MyFirstRobot.java:13)

real	0m35.872s
user	0m25.450s
sys	0m1.600s
Creating Motor A & B
Exception in thread "main" java.lang.InternalError: Unsupported Management version
	at java.base/java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.base/java.lang.ClassLoader.loadLibrary0(Unknown Source)
	at java.base/java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.base/java.lang.Runtime.loadLibrary0(Unknown Source)
	at java.base/java.lang.System.loadLibrary(Unknown Source)
	at java.management/java.lang.management.ManagementFactory.lambda$static$8(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.management/java.lang.management.ManagementFactory.<clinit>(Unknown Source)
	at ch.qos.logback.classic.joran.action.JMXConfiguratorAction.begin(JMXConfiguratorAction.java:61)
	at ch.qos.logback.core.joran.spi.Interpreter.callBeginAction(Interpreter.java:269)
	at ch.qos.logback.core.joran.spi.Interpreter.startElement(Interpreter.java:145)
	at ch.qos.logback.core.joran.spi.Interpreter.startElement(Interpreter.java:128)
	at ch.qos.logback.core.joran.spi.EventPlayer.play(EventPlayer.java:50)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:165)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:152)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:110)
	at ch.qos.logback.core.joran.GenericConfigurator.doConfigure(GenericConfigurator.java:53)
	at ch.qos.logback.classic.util.ContextInitializer.configureByResource(ContextInitializer.java:75)
	at ch.qos.logback.classic.util.ContextInitializer.autoConfig(ContextInitializer.java:150)
	at org.slf4j.impl.StaticLoggerBinder.init(StaticLoggerBinder.java:84)
	at org.slf4j.impl.StaticLoggerBinder.<clinit>(StaticLoggerBinder.java:55)
	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:150)
	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:124)
	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:412)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:357)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:383)
	at ev3dev.hardware.EV3DevFileSystem.<clinit>(EV3DevFileSystem.java:15)
	at examples.MyFirstRobot.main(MyFirstRobot.java:13)

[... 108 lines cut ...]
2018-01-04 14:43:49 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outA
2018-01-04 14:43:49 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor14/address
2018-01-04 14:43:49 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
2018-01-04 14:43:49 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: outB
2018-01-04 14:43:49 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
[... cut ...]
2018-01-04 14:43:50 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: outB
2018-01-04 14:43:50 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor15/address
Defining the Stop mode
Defining motor speed
Go Forward with the motors
Stop motors
Go Backward with the motors
Stop motors
Checking Battery
2018-01-04 14:43:54 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
Votage: 8.3938
Emergency Stop

real	1m11.123s
user	0m49.770s
sys	0m2.260s
robot@ev3dev:~$ 
```

JakubVanek, 2018-01-04:
```
robot@ev3dev:~$ time images/jre/bin/java -client -Xshare:on -verbose:class -jar template_project_gradle-all-0.2.0.jar; time images/jre/bin/java -client -Xshare:on -verbose:class -jar template_project_gradle-all-0.2.0.jar; time images/jre/bin/java -client -Xshare:on -verbose:class -jar template_project_gradle-all-0.2.0.jar
```
Run 1:
```
[...]
[11.762s][info][class,load] java.lang.PublicMethods$Key source: shared objects file
Creating Motor A & B
[...]
[73.049s][info][class,load] examples.MyFirstRobot$1 source: file:/home/robot/template_project_gradle-all-0.2.0.jar
Defining the Stop mode
[...]
real	1m18.496s
user	0m47.930s
sys	0m3.020s
```
Run 2:
```
[...]
[10.517s][info][class,load] java.lang.PublicMethods$Key source: shared objects file
Creating Motor A & B
[...]
[68.642s][info][class,load] examples.MyFirstRobot$1 source: file:/home/robot/template_project_gradle-all-0.2.0.jar
Defining the Stop mode
[...]
real	1m13.617s
user	0m47.500s
sys	0m2.710s
```
Run 3:
```
[...]
[10.613s][info][class,load] java.lang.PublicMethods$Key source: shared objects file
Creating Motor A & B
[...]
[69.769s][info][class,load] examples.MyFirstRobot$1 source: file:/home/robot/template_project_gradle-all-0.2.0.jar
Defining the Stop mode
[...]
real	1m14.663s
user	0m47.900s
sys	0m2.500s
```
I hope that it's only the `-verbose:class` switch that made it to slow down.

JakubVanek, 2018-01-04:
```
robot@ev3dev:~$ time ./images/jdk/bin/java -client HelloWorld; time ./images/jdk/bin/java -client HelloWorld; time ./images/jdk/bin/java -client HelloWorld
Hello, World

real	0m9.323s
user	0m5.240s
sys	0m0.570s
Hello, World

real	0m7.060s
user	0m5.120s
sys	0m0.510s
Hello, World

real	0m6.894s
user	0m5.040s
sys	0m0.450s
```
Cached third run:
```
[0.301s][info][class,load] opened: /home/robot/images/jdk/lib/modules
[0.408s][info][class,load] java.lang.Object source: shared objects file
[0.414s][info][class,load] java.io.Serializable source: shared objects file
[0.419s][info][class,load] java.lang.Comparable source: shared objects file
[0.424s][info][class,load] java.lang.CharSequence source: shared objects file
[0.435s][info][class,load] java.lang.String source: shared objects file
[0.441s][info][class,load] java.lang.reflect.AnnotatedElement source: shared objects file
[0.446s][info][class,load] java.lang.reflect.GenericDeclaration source: shared objects file
[0.454s][info][class,load] java.lang.reflect.Type source: shared objects file
[0.459s][info][class,load] java.lang.Class source: shared objects file
[0.469s][info][class,load] java.lang.Cloneable source: shared objects file
[0.477s][info][class,load] java.lang.ClassLoader source: shared objects file
[0.482s][info][class,load] java.lang.System source: shared objects file
[0.489s][info][class,load] java.lang.Throwable source: shared objects file
[0.494s][info][class,load] java.lang.Error source: shared objects file
[0.498s][info][class,load] java.lang.ThreadDeath source: shared objects file
[0.503s][info][class,load] java.lang.Exception source: shared objects file
[0.508s][info][class,load] java.lang.RuntimeException source: shared objects file
[0.512s][info][class,load] java.lang.SecurityManager source: shared objects file
[0.518s][info][class,load] java.security.ProtectionDomain source: shared objects file
[0.525s][info][class,load] java.security.AccessControlContext source: shared objects file
[0.530s][info][class,load] java.security.SecureClassLoader source: shared objects file
[0.536s][info][class,load] java.lang.ReflectiveOperationException source: shared objects file
[0.541s][info][class,load] java.lang.ClassNotFoundException source: shared objects file
[0.546s][info][class,load] java.lang.LinkageError source: shared objects file
[0.550s][info][class,load] java.lang.NoClassDefFoundError source: shared objects file
[0.556s][info][class,load] java.lang.ClassCastException source: shared objects file
[0.564s][info][class,load] java.lang.ArrayStoreException source: shared objects file
[0.568s][info][class,load] java.lang.VirtualMachineError source: shared objects file
[0.576s][info][class,load] java.lang.OutOfMemoryError source: shared objects file
[0.581s][info][class,load] java.lang.StackOverflowError source: shared objects file
[0.586s][info][class,load] java.lang.IllegalMonitorStateException source: shared objects file
[0.592s][info][class,load] java.lang.ref.Reference source: shared objects file
[0.597s][info][class,load] java.lang.ref.SoftReference source: shared objects file
[0.602s][info][class,load] java.lang.ref.WeakReference source: shared objects file
[0.607s][info][class,load] java.lang.ref.FinalReference source: shared objects file
[0.611s][info][class,load] java.lang.ref.PhantomReference source: shared objects file
[0.617s][info][class,load] java.lang.ref.Finalizer source: shared objects file
[0.622s][info][class,load] java.lang.Runnable source: shared objects file
[0.634s][info][class,load] java.lang.Thread source: shared objects file
[0.639s][info][class,load] java.lang.Thread$UncaughtExceptionHandler source: shared objects file
[0.644s][info][class,load] java.lang.ThreadGroup source: shared objects file
[0.650s][info][class,load] java.util.Dictionary source: shared objects file
[0.661s][info][class,load] java.util.Map source: shared objects file
[0.668s][info][class,load] java.util.Hashtable source: shared objects file
[0.673s][info][class,load] java.util.Properties source: shared objects file
[0.678s][info][class,load] java.lang.Module source: shared objects file
[0.684s][info][class,load] java.lang.reflect.AccessibleObject source: shared objects file
[0.689s][info][class,load] java.lang.reflect.Member source: shared objects file
[0.696s][info][class,load] java.lang.reflect.Field source: shared objects file
[... 459 lines cut ...]
[8.823s][info][class,load] java.lang.Shutdown$Lock source: shared objects file

real	0m8.999s
user	0m5.440s
sys	0m0.530s
```
Minimal VM:
```
robot@ev3dev:~$ time ./images/jdk/bin/java -minimal HelloWorld; time ./images/jdk/bin/java -minimal HelloWorld; time ./images/jdk/bin/java -minimal HelloWorld
Hello, World

real	0m11.680s
user	0m7.950s
sys	0m0.750s
Hello, World

real	0m10.919s
user	0m8.160s
sys	0m0.580s
Hello, World

real	0m10.843s
user	0m8.090s
sys	0m0.470s
```

jabrena, 2018-01-04:
The times is similar to Oracle JRE8. 
I have to test.

The build allow compiling?
What is the content of the folder where is the command java?
jps? javac?

JakubVanek, 2018-01-04:
The zipped build is a complete built image, including a JRE as a subset. Compiling on the brick works, HelloWorld was compiled on the brick.
JDK directory:
```
appletviewer
idlj
jar
jarsigner
java
javac
javadoc
javah
javap
jcmd
jconsole
jdb
jdeprscan
jdeps
jhsdb
jimage
jinfo
jjs
jlink
jmap
jmod
jps
jrunscript
jshell
jstack
jstat
jstatd
keytool
orbd
pack200
policytool
rmic
rmid
rmiregistry
serialver
servertool
schemagen
tnameserv
unpack200
wsgen
wsimport
xjc
```
JRE:
```
appletviewer
idlj
java
jjs
jrunscript
keytool
orbd
pack200
rmid
rmiregistry
servertool
tnameserv
unpack200
```

jabrena, 2018-01-04:
How many seconds to compile a Hello World?

jabrena, 2018-01-05:
Hi, is it possible if you commit the build in the new repo? I could upgrade the installer to start using the new openjdk build?

JakubVanek, 2018-01-05:
Hi,
yes, I'll do it during the weekend (maybe even today).
Regarding javac on the brick, these are the measurements:
Run 1:
```
real	1m5.669s
user	0m53.230s
sys	0m2.730s
```
Run 2:
```
real	1m4.061s
user	0m53.890s
sys	0m2.510s
```
Run 3:
```
real	1m2.507s
user	0m54.290s
sys	0m2.300s
```
Regards

jabrena, 2018-01-05:
In my case, I will upgrade this project.
https://github.com/ev3dev-lang-java/installer

With your development, now the process could be full automated.

I will explain:

1. Install ev3dev
2. Install installer in the brick.

The installer will manage: jvm, native libraries, battery-monitor...

XD

JakubVanek, 2018-01-05:
Hmmm, it might be even better to prepare a debian package for it.

jabrena, 2018-01-05:
Fantastic idea!!!

jabrena, 2018-01-05:
But if you like we could start with the zip in the repo, we have to review licence matters, etc...

jabrena, 2018-01-05:
In few days, I will have my brick to review everything. But as you say, the debian package is nice but at the end is a process to download a zip, unzip and locate in /opt/ 

Cheers

Juan Antonio

jabrena, 2018-01-05:
In the repo we could execute a docker on travis to review periodically the openjdk build process

JakubVanek, 2018-01-06:
I've finished the basic building: https://github.com/ev3dev-lang-java/openjdk-9-ev3/tree/initial

jabrena, 2018-01-06:
I will add the travis support.

Can you commit the build result as a zip in the repo?

jabrena, 2018-01-06:
Proposal of zip format: images/openjdk-9-ev3-release-20180106.tar.gz

jabrena, 2018-01-06:
Every 2 months, we could generate a new build with latest sources

jabrena, 2018-01-06:
What is the purpose of ev3.patch and float.patch?

jabrena, 2018-01-06:
For travis, the scripts should run without any human interaction.

[16 more replies]

closed 2018-03-13 by jabrena