ev3dev-lang-java/ev3dev-lang-java #254: Test ROS2 for java

jabrena, 2017-04-29:
https://github.com/esteve/ros2_java/blob/master/README.md

jabrena, 2017-04-29:
Following the documentation:

``` 
mkdir -p ~/ament_ws/src
cd ~/ament_ws
wget https://raw.githubusercontent.com/esteve/ament_java/master/ament_java.repos
vcs import ~/ament_ws/src < ament_java.repos
src/ament/ament_tools/scripts/ament.py build --symlink-install --isolated
```

esteve, 2017-04-30:
Just for consolidate the instructions to install the dependencies and build instructions for OSX, here's what I did on Travis:

```
brew tap osrf/ros2
brew tap ros/deps
brew install python3 wget cppcheck gtest tinyxml eigen pcre cmake # cmake already installed
brew install asio tinyxml2
brew install Caskroom/cask/java
brew install gradle
brew tap homebrew/science
sudo -H python3 -m pip install empy setuptools nose vcstool pep8 pydocstyle pyflakes flake8 mock coverage
mkdir -p ~/ament_ws/src
cd ~/ament_ws
wget https://raw.githubusercontent.com/esteve/ament_java/master/ament_java.repos
vcs import ~/ament_ws/src < ament_java.repos
src/ament/ament_tools/scripts/ament.py build --parallel --symlink-install --isolated
mkdir -p ~/ros2_java_ws/src
cd ~/ros2_java_ws
wget https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop_travis.repos
vcs import ~/ros2_java_ws/src < ros2_java_desktop_travis.repos
cd ~/ros2_java_ws/src/ros2/rosidl_typesupport
patch -p1 < ../../ros2_java/ros2_java/rosidl_typesupport_ros2_java_travis.patch
cd ~/ros2_java_ws
. ~/ament_ws/install_isolated/local_setup.sh
ament build --parallel --symlink-install --isolated
```

this uses a smaller repos file that only builds messages for C and Java, but not C++ and Python, to speed up the building. If you want to build the whole of ROS2 and `ros2_java`, you can use `https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop.repos` instead, but it's not really necessary.

In any case, make sure to read the instructions from the ROS2 wiki too (https://github.com/ros2/ros2/wiki/OSX-Development-Setup), so you have a understanding idea of the whole system if you want to interact with the rest of the ROS2 ecosystem (e.g. if you want to write a Python script that talks to a `ros2_java` application).

Let me know how it goes and if there's anything I can help with.

esteve, 2017-04-30:
I forgot, once you've built all this, you can just run the following exaples to check that it all works:

On one terminal:

```
. ~/ros2_java_ws/install_isolated/local_setup.sh
java org.ros2.rcljava.examples.Listener
```

and on another one:

```
. ~/ros2_java_ws/install_isolated/local_setup.sh
java org.ros2.rcljava.examples.Talker
```

`ros2_java` also supports services:

On one terminal:

```
. ~/ros2_java_ws/install_isolated/local_setup.sh
java org.ros2.rcljava.examples.AddTwoIntsService
```

and on another one:
```
. ~/ros2_java_ws/install_isolated/local_setup.sh
java org.ros2.rcljava.examples.AddTwoIntsClient
```

jabrena, 2017-04-30:
Hi @esteve, 

after more or less 30m. compiling I saw in the screen:

`BUILD SUCCESSFUL`

Testing in a different terminal the first example y receive this error:

``` 
➜  ros2_java_ws . ~/ros2_java_ws/install_isolated/local_setup.zsh
➜  ros2_java_ws java org.ros2.rcljava.examples.Listener
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.NoClassDefFoundError: org/ros2/rcljava/interfaces/MessageDefinition
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.getDeclaredMethods0(Native Method)
	at java.lang.Class.privateGetDeclaredMethods(Class.java:2701)
	at java.lang.Class.privateGetMethodRecursive(Class.java:3048)
	at java.lang.Class.getMethod0(Class.java:3018)
	at java.lang.Class.getMethod(Class.java:1784)
	at sun.launcher.LauncherHelper.validateMainClass(LauncherHelper.java:544)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:526)
Caused by: java.lang.ClassNotFoundException: org.ros2.rcljava.interfaces.MessageDefinition
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 19 more
➜  ros2_java_ws 
```

jabrena, 2017-04-30:
Is it possible that I am not loading something for zsh?

```
➜  ros2_java_ws cat ~/ros2_java_ws/install_isolated/local_setup.zsh
# copied from ament_package/template/isolated_prefix_level/local_setup.zsh

AMENT_SHELL=zsh

# source local_setup.sh from same directory as this file
AMENT_CURRENT_PREFIX=$(builtin cd -q "`dirname "${(%):-%N}"`" > /dev/null && pwd)

# function to convert array-like strings into arrays
# to wordaround SH_WORD_SPLIT not being set
ament_zsh_to_array() {
  local _listname=$1
  local _dollar="$"
  local _split="{="
  local _to_array="(\"$_dollar$_split$_listname}\")"
  eval $_listname=$_to_array
}

# trace output
if [ -n "$AMENT_TRACE_SETUP_FILES" ]; then
  echo ". \"$AMENT_CURRENT_PREFIX/local_setup.sh\""
fi
. "$AMENT_CURRENT_PREFIX/local_setup.sh"
```

jabrena, 2017-04-30:
If I load the .sh file in another terminal, I receive the following message:

``` 
➜  ros2_java_ws . ~/ros2_java_ws/install_isolated/local_setup.sh
➜  ros2_java_ws java org.ros2.rcljava.examples.Listener
Error: no se ha encontrado o cargado la clase principal org.ros2.rcljava.examples.Listener
```

esteve, 2017-05-02:
This change (https://github.com/esteve/ament_java/commit/8d2efbca2fc4dc1a9a8d1d5d1cac8fc147ab3615) should fix the issues with ZSH. You can update both workspaces typing the following:

`
cd ~/ament_ws
vcs pull
src/ament/ament_tools/scripts/ament.py build --parallel --symlink-install --isolated
cd ~/ros2_java_ws
vcs pull
. ~/ament_ws/install_isolated/local_setup.sh
ament build --parallel --symlink-install --isolated
`

let me know if it fixes the issues.

jabrena, 2017-05-02:
Hi @esteve, 

today/tomorrow, I will compile again with this change. 
I wish I can run the examples.

Cheers

Juan Antonio

jabrena, 2017-05-02:
Hi Steve, I followed your guide, but when I execute the talker, I receive the following error:

```
➜  ros2_java_ws . ~/ros2_java_ws/install_isolated/local_setup.zsh
➜  ros2_java_ws java org.ros2.rcljava.examples.Listener            
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/Users/jabrena/ros2_java_ws/build_isolated/rcljava_common/jars/slf4j-jdk14-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/Users/jabrena/ros2_java_ws/build_isolated/rcljava_common/jars/slf4j-log4j12-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.JDK14LoggerFactory]
may 02, 2017 10:59:03 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 02, 2017 10:59:03 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 02, 2017 10:59:03 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 02, 2017 10:59:03 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 02, 2017 10:59:03 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 02, 2017 10:59:03 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 02, 2017 10:59:03 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 02, 2017 10:59:03 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 02, 2017 10:59:03 PM org.ros2.rcljava.RCLJava rclJavaInit
GRAVE: No RMW implementation found
➜  ros2_java_ws 
```

esteve, 2017-05-03:
@jabrena that is weird. The classpath is now properly set, check with `set` and there should be a `CLASSPATH` environment variable after sourcing the shell scripts.

But it seems the `DYLD_LIBRARY_PATH` environment is not properly set, so even if `CLASSPATH` is correct (it seems so, since `java` can find the JARs), `ros2_java` needs to load the JNI libraries.

Could you try deleting both workspaces and starting over? I know it's a bit tedious, but I can't seem to reproduce it locally. And if it still does not work, could you try running it all from a Bash shell?

Thanks!

esteve, 2017-05-03:
@jabrena I just built it all from scratch and it worked. Could you try the following?

```
mkdir -p ~/ros2_java/ament_ws/src
cd ~/ros2_java/ament_ws
wget https://raw.githubusercontent.com/esteve/ament_java/master/ament_java.repos
vcs import ~/ros2_java/ament_ws/src < ament_java.repos
src/ament/ament_tools/scripts/ament.py build --parallel --symlink-install --isolated
mkdir -p ~/ros2_java/ros2_java_ws/src
cd ~/ros2_java/ros2_java_ws
wget https://raw.githubusercontent.com/esteve/ros2_java/master/ros2_java_desktop_travis.repos
vcs import ~/ros2_java/ros2_java_ws/src < ros2_java_desktop_travis.repos
cd ~/ros2_java/ros2_java_ws/src/ros2/rosidl_typesupport
patch -p1 < ../../ros2_java/ros2_java/rosidl_typesupport_ros2_java_travis.patch
cd ~/ros2_java/ros2_java_ws
. ~/ros2_java/ament_ws/install_isolated/local_setup.zsh
ament build --parallel --symlink-install --isolated
. ~/ros2_java/ros2_java_ws/install_isolated/local_setup.zsh
java org.ros2.rcljava.examples.Listener
```

and these are the contents of `CLASSPATH` and `DYLD_LIBRARY_PATH` after sourcing the `local_setup.zsh` file from the `ros2_java` workspace:

```
% echo $CLASSPATH 
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/visualization_msgs/share/visualization_msgs/java/visualization_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/trajectory_msgs/share/trajectory_msgs/java/trajectory_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/stereo_msgs/share/stereo_msgs/java/stereo_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/std_srvs/share/std_srvs/java/std_srvs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/shape_msgs/share/shape_msgs/java/shape_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/sensor_msgs/share/sensor_msgs/java/sensor_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava/share/rcljava/java/rcljava.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/nav_msgs/share/nav_msgs/java/nav_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/diagnostic_msgs/share/diagnostic_msgs/java/diagnostic_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/geometry_msgs/share/geometry_msgs/java/geometry_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/actionlib_msgs/share/actionlib_msgs/java/actionlib_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/std_msgs/share/std_msgs/java/std_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcl_interfaces/share/rcl_interfaces/java/rcl_interfaces_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/lifecycle_msgs/share/lifecycle_msgs/java/lifecycle_msgs_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/example_interfaces/share/example_interfaces/java/example_interfaces_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/builtin_interfaces/share/builtin_interfaces/java/builtin_interfaces_messages.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/rcljava_common.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-jdk14-1.7.21.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-log4j12-1.7.21.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/log4j-1.2.17.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-api-1.7.21.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_examples/share/rcljava_examples/java/rcljava_examples.jar:.
% echo $DYLD_LIBRARY_PATH 
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/visualization_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/trajectory_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/tinyxml_vendor/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/stereo_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/std_srvs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/shape_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/sensor_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcl_lifecycle/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcl/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/nav_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/diagnostic_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/geometry_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/actionlib_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/std_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcl_interfaces/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/lifecycle_msgs/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/example_interfaces/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/builtin_interfaces/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_cpp/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_c/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rmw_fastrtps_cpp/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolat [... cut ...] ated/rcljava_common/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcutils/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/poco_vendor/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/fastrtps/lib:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/fastcdr/lib:/Users/esteve/ros2_java/ament_ws/install_isolated/ament_index_cpp/lib:/Users/esteve/ros2_java/ament_ws/install_isolated/gtest_vendor/lib:/Users/esteve/ros2_java/ament_ws/install_isolated/gmock_vendor/lib
```

jabrena, 2017-05-03:
Hi @esteve,

Today, I will try again. :)

jabrena, 2017-05-06:
Starting the whole process now.

jabrena, 2017-05-06:
Hi @esteve, I compared the classpath & the native libraries that you have but the list is different:

**CLASSPATH**
``` 
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/visualization_msgs/share/visualization_msgs/java/visualization_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/trajectory_msgs/share/trajectory_msgs/java/trajectory_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/stereo_msgs/share/stereo_msgs/java/stereo_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/std_srvs/share/std_srvs/java/std_srvs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/shape_msgs/share/shape_msgs/java/shape_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/sensor_msgs/share/sensor_msgs/java/sensor_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava/share/rcljava/java/rcljava.jar:/Users/esteve/ros2_java/ros2_java_ws/install_isolated/nav_msgs/share/nav_msgs/java/nav_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/diagnostic_msgs/share/diagnostic_msgs/java/diagnostic_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/geometry_msgs/share/geometry_msgs/java/geometry_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/actionlib_msgs/share/actionlib_msgs/java/actionlib_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/std_msgs/share/std_msgs/java/std_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcl_interfaces/share/rcl_interfaces/java/rcl_interfaces_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/lifecycle_msgs/share/lifecycle_msgs/java/lifecycle_msgs_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/example_interfaces/share/example_interfaces/java/example_interfaces_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/builtin_interfaces/share/builtin_interfaces/java/builtin_interfaces_messages.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/rcljava_common.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-jdk14-1.7.21.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-log4j12-1.7.21.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/log4j-1.2.17.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-api-1.7.21.jar:
/Users/esteve/ros2_java/ros2_java_ws/install_isolated/rcljava_examples/share/rcljava_examples/java/rcljava_examples.jar:.
```

jabrena, 2017-05-06:
if I execute the bash script to run the examples, the information to debug is mininum:

```
. ~/ros2_java/ros2_java_ws/install_isolated/local_setup.sh
java org.ros2.rcljava.examples.Listener
``` 

**Output:**

``` 
bash-3.2$ java org.ros2.rcljava.examples.Listener
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/Users/jabrena/ros2_java/ros2_java_ws/build_isolated/rcljava_common/jars/slf4j-jdk14-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/Users/jabrena/ros2_java/ros2_java_ws/build_isolated/rcljava_common/jars/slf4j-log4j12-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [org.slf4j.impl.JDK14LoggerFactory]
may 06, 2017 1:04:30 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 06, 2017 1:04:30 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 06, 2017 1:04:30 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 06, 2017 1:04:30 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 06, 2017 1:04:30 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 06, 2017 1:04:30 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 06, 2017 1:04:30 PM org.ros2.rcljava.common.JNIUtils loadLibrary
INFORMACIÓN: Loading library: org_ros2_rcljava_rcl_java
may 06, 2017 1:04:30 PM org.ros2.rcljava.RCLJava rclJavaInit
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: no org_ros2_rcljava_rcl_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 06, 2017 1:04:30 PM org.ros2.rcljava.RCLJava rclJavaInit
GRAVE: No RMW implementation found
```

jabrena, 2017-05-06:
But if I execute the example in a manual way:

``` 
java -Djava.library.path=/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/visualization_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/trajectory_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/tinyxml_vendor/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/stereo_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/std_srvs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/shape_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/sensor_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcl_lifecycle/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcl/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/nav_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/diagnostic_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/geometry_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/actionlib_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/std_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcl_interfaces/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/lifecycle_msgs/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/example_interfaces/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/builtin_interfaces/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_cpp/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_c/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rmw_fastrtps_cpp/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_introspection_cpp/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_introspection_c/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rmw/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rosidl_generator_c/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_common/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcutils/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/poco_vendor/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/fastrtps/lib:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/fastcdr/lib -cp /Users/jabrena/ros2_java/ros2_java_ws/install_isolated/visualization_msgs/share/visualization_msgs/java/visualization_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/trajectory_msgs/share/trajectory_msgs/java/trajectory_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/stereo_msgs/share/stereo_msgs/java/stereo_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/std_srvs/share/std_srvs/java/std_srvs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/shape_msgs/share/shape_msgs/java/shape_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/sensor_msgs/share/sensor_msgs/java/sensor_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava/share/rcljava/java/rcljava.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/nav_msgs/share/nav_msgs/java/nav_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/diagnostic_msgs/share/diagnostic_msgs/java/diagnostic_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/geometry_msgs/share/geometry_msgs/java/geometry_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/actionlib_msgs/share/actionlib_msgs/java/actionlib_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/std_msgs/share/std_msgs/java/std_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcl_interfaces/share/rcl_interfaces/java/rcl_interfaces_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/lifecycle_msgs/share/lifecycle_msgs/java/lifecycle_msgs_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/example_interfaces/share/example_interfaces/java/example_interfaces_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/builtin_interfaces/share/builtin_interfaces/java/builtin_interfaces_messages.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/rcljava_common.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-jdk14-1.7.21.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-log4j12-1.7.21.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/log4j-1.2.17.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_common/share/rcljava_common/java/slf4j-api-1.7.21.jar:/Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rcljava_examples/share/rcljava_examples/java/rcljava_examples.jar:. org.ros2.rcljava.examples.Listener
``` 

I receive more information:

``` 
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/Users/jabrena/ros2_java/ros2_java_ws/build_isolated/rcljava_common/jars/slf4j-jdk14-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/Users/jabrena/ros2_java/ros2_java_ws/build_isolated/rcljava_common/jars/slf4j-log4j12-1.7.21.jar!/org/slf4j/impl/StaticLoggerBinder.class]
[... cut ...]
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:49)
	at org.ros2.rcljava.common.JNIUtils.loadLibrary(JNIUtils.java:30)
	at org.ros2.rcljava.RCLJava.setRMWImplementation(RCLJava.java:171)
	at org.ros2.rcljava.RCLJava.rclJavaInit(RCLJava.java:130)
	at org.ros2.rcljava.examples.Listener.main(Listener.java:34)

may 06, 2017 1:03:23 PM org.ros2.rcljava.RCLJava rclJavaInit
GRAVE: No RMW implementation found
```

jabrena, 2017-05-06:
Maybe the problem is here:

``` 
INFORMACIÓN: UnsatisfiedLinkError
java.lang.UnsatisfiedLinkError: /Users/jabrena/ros2_java/ros2_java_ws/build_isolated/rcljava/liborg_ros2_rcljava_rcl_java.dylib: dlopen(/Users/jabrena/ros2_java/ros2_java_ws/build_isolated/rcljava/liborg_ros2_rcljava_rcl_java.dylib, 1): Library not loaded: libPocoFoundation.47.dylib
  Referenced from: /Users/jabrena/ros2_java/ros2_java_ws/install_isolated/rosidl_typesupport_c/lib/librosidl_typesupport_c.dylib
  Reason: image not found
```

jabrena, 2017-05-06:
In order to manage better the execution I use the variables that you defined in the Script:

```
. ~/ros2_java/ros2_java_ws/install_isolated/local_setup.sh
java -Djava.library.path=$DYLD_LIBRARY_PATH -cp $CLASSPATH org.ros2.rcljava.examples.Listener
```

The problem is the same

jabrena, 2017-05-07:
I could run the examples on Ubuntu. I will delay this test for next release. Now I will put the focus on Brickpi+/PiStorms.

jabrena, 2017-05-07:
I am going to install ROS2 for OSX to discard some problem with the installation of ros2_java
https://github.com/ros2/ros2/wiki/OSX-Development-Setup

jabrena, 2017-12-31:
It is not a priority

closed 2017-12-31 by jabrena