ev3dev-lang-java/ev3dev-lang-java #256: [ROS2] Install ROS2_java in EV3 Brick

jabrena, 2017-04-29:


jabrena, 2017-04-29:
It is necessary to install this tool:
https://github.com/dirk-thomas/vcstool
https://packages.debian.org/sid/python-vcstools

jabrena, 2017-04-29:
vi /etc/apt/sources.list

deb http://http.debian.net/debian sid main

jabrena, 2017-04-29:
```
robot@ev3dev:/etc/apt$ sudo apt-get install python-vcstool
Reading package lists... Done
Building dependency tree        
Reading state information... Done
E: Unable to locate package python-vcstool
robot@ev3dev:/etc/apt$ 

```

esteve, 2017-04-29:
@jabrena `vcstool` can be installed directly from PyPI:

`$ sudo pip3 install vcstool`

However, since the Raspberry Pi is fairly slow, you can use `CMake` toolchain (for example https://github.com/ruslo/polly) and crosscompile `ros2_java` instead of compiling on the RPI itself. I can give you more details tomorrow about the whole process, but basically you pass `--cmake-args -DCMAKE_TOOLCHAIN_FILE=PATH_TO_THE_POLLY_TOOLCHAIN --` to ament and follow https://github.com/ruslo/polly/wiki/Toolchain-list#raspberrypi2-cxx11 (similar to the ROS2 bindings for the iPhone https://github.com/esteve/ros2_objc#ios)

I've used this to crosscompile ROS2 to the RaspberryPi and the ODroid and it works pretty well.

jabrena, 2017-04-29:
Yes, of course, I will do it.

If you like, I will test in this order:

```
OSX > Raspi 3 > EV3 Brick.
```

jabrena, 2017-12-31:
It is not a priority

closed 2017-12-31 by jabrena