ev3dev-lang-java/ev3dev-lang-java #273: Add Button Support

jabrena, 2017-05-23:
https://github.com/mob41/ev3dev-lang-java/blob/develop/src/main/java/org/ev3dev/hardware/Button.java

jabrena, 2017-05-24:
First test is running.
Integrating initial implementation with LeJOS interfaces

jabrena, 2017-06-03:
Detected a bug in Button support. It is necessary to review

jabrena, 2017-06-04:
The issue was solved partially.

In this release we have covered the usage of the button with this Scenario:

``` java
package examples.sensors;

import ev3dev.sensors.Button;

public class ButtonExample {

    public static void main(final String[] args){

        System.out.println("Button example");

        Button.waitForAnyPress();

        System.out.println("Button pressed");
    }
}
``` 

But, it is necessary to research how to increase the granularity in next releases.

jabrena, 2017-06-04:
Finally, I am using this implementation:
https://github.com/mob41/ev3dev-lang-java/blob/develop/src/main/java/org/ev3dev/hardware/Button.java

@mob41, many thanks for the implementation.

Juan Antonio

closed 2017-06-04 by jabrena

mob41, 2017-06-08:
@jabrena The code idea and implementation can be used as long as you do not claim the code as your own, and follows the terms brought to the MIT License by applying a more restrictive license on the code.

This applies also to the code [LCDGraphics.java, LCD.java, Button.java, and other code licensed](https://github.com/mob41/ev3dev-lang-java/). You should better think about how you will feel if your code that you have paid a lot of time to think a concept to achieve the goal is used by others without proper notice and permission.