ev3dev-lang-java/ev3dev-lang-java #276: Improve Java Installer

jabrena, 2017-05-25:
https://github.com/serverfarmer/sf-java8/blob/master/setup.sh

jabrena, 2017-06-25:
With this idea, it is possible to download the JRE in an automatic way:

```
wget -O jdk-8u131-linux-arm32-vfp-hflt.tar.gz --header="Cookie: oraclelicense=accept-securebackup-cookie" http://download.oracle.com/otn-pub/java/jdk/8u131-b11/d54c1d3a095b4ff2b6607d096fa80163/jdk-8u131-linux-arm32-vfp-hflt.tar.gz
```

JakubVanek, 2018-03-28:
Before adding an automatic license agree, I would recommend putting a big warning before running the script.
With OpenJDK this shouldn't a problem.

jabrena, 2018-03-31:
Related issue:
https://github.com/ev3dev-lang-java/installer/issues/15

jabrena, 2018-04-11:
Hi @JakubVanek, in the past, I tried to avoid the logon authentication, but i failed. Now we have something better:
https://github.com/ev3dev-lang-java/openjdk-ev3/releases/tag/v0.5.0

jabrena, 2018-04-14:
Starting the development

jabrena, 2018-04-14:
Linking JRI installation to latest Release:
https://github.com/ev3dev-lang-java/openjdk-ev3/releases
https://github.com/ev3dev-lang-java/openjdk-ev3/releases/download/v0.5.0/jri10-ev3.tar.gz

jabrena, 2018-04-14:
For a first installation, the statement:

```
update-alternatives --install /usr/bin/java java /opt/jri-10-build-050/bin/java 1
```

But when you have to upgrade the installation from OpenJDK 9 to 10, it is necessary to have another syntax.

``` 
sudo update-alternatives --config java
```
https://askubuntu.com/questions/121654/how-to-set-default-java-version

jabrena, 2018-04-14:
When we have OpenJDK11, I will add the new option to reinstall Java.

jabrena, 2018-04-14:
Fixed.
https://github.com/ev3dev-lang-java/installer/pull/17

jabrena, 2018-04-14:
Pending testing on EV3 Brick

jabrena, 2018-04-20:
Testing on Brick with Java.

```
> Task :installJava
Strict host key checking is off. It may be vulnerable to man-in-the-middle attacks.
echo -e "maker" | sudo -S /home/robot/installer/installer.sh

ev3dev#0|
ev3dev#0|##############################
ev3dev#0|# EV3Dev-lang-java Installer #
ev3dev#0|##############################
ev3dev#0|# Last update: 2018/04/14    #
ev3dev#0|##############################
ev3dev#0|
ev3dev#0|Init installer
ev3dev#0|
ev3dev#0|#########################################
ev3dev#0|# PLATFORM SECTION                      #
ev3dev#0|#########################################
ev3dev#0|
ev3dev#0|Platform detected: ev3
ev3dev#0|
ev3dev#0|
ev3dev#0|#########################################
ev3dev#0|# JAVA SECTION                          #
ev3dev#0|#########################################
ev3dev#0|
ev3dev#0|/usr/bin/java
ev3dev#0|Found java executable in PATH
ev3dev#0|Installation complete. Java version:
ev3dev#0|[sudo] password for robot: openjdk version "10" 2018-03-20
ev3dev#0|OpenJDK Runtime Environment (build 10+46-ev3)
ev3dev#0|OpenJDK Client VM (build 10+46-ev3, mixed mode, sharing)
ev3dev#0|Dumping class cache...
ev3dev#0|Allocated shared space: 268435456 bytes at 0x80000000
ev3dev#0|Loading classes to share ...
ev3dev#0|Loading classes to share: done.
ev3dev#0|Rewriting and linking classes ...
ev3dev#0|Rewriting and linking classes: done
ev3dev#0|Number of classes 1231
ev3dev#0|    instance classes   =  1170
ev3dev#0|    obj array classes  =    53
ev3dev#0|    type array classes =     8
ev3dev#0|Updating ConstMethods ... done. 
ev3dev#0|Removing unshareable information ... done. 
ev3dev#0|Scanning all metaspace objects ... 
ev3dev#0|Allocating RW objects ... 
ev3dev#0|Allocating RO objects ... 
ev3dev#0|Relocating embedded pointers ... 
ev3dev#0|Relocating external roots ... 
ev3dev#0|Dumping symbol table ...
ev3dev#0|Relocating SystemDictionary::_well_known_klasses[] ... 
ev3dev#0|Removing java_mirror ... done. 
ev3dev#0|mc  space:      2924 [  0.0% of total] out of      4096 bytes [ 71.4% used] at 0x80000000
ev3dev#0|rw  space:   2069528 [ 18.2% of total] out of   2072576 bytes [ 99.9% used] at 0x80001000
ev3dev#0|ro  space:   4617204 [ 40.7% of total] out of   4620288 bytes [ 99.9% used] at 0x801fb000
ev3dev#0|md  space:      3080 [  0.0% of total] out of      4096 bytes [ 75.2% used] at 0x80663000
ev3dev#0|od  space:   4655900 [ 41.0% of total] out of   4657152 bytes [100.0% used] at 0x80664000
ev3dev#0|total    :  11348636 [100.0% of total] out of  11358208 bytes [ 99.9% used]
```

jabrena, 2018-04-20:
Fixed mini problem with the download of modules on wrong path.

jabrena, 2018-04-20:
The installer went fine. 

Besides, exist a option for incremental version of Java 10.

```
robot@ev3dev:~$ sudo update-alternatives --config java
[sudo] password for robot: 
There are 8 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                        Priority   Status
------------------------------------------------------------
  0            /opt/jri-10-build-050/bin/java               10        auto mode
  1            /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java   1         manual mode
  2            /opt/jdk-10-build-46-v2/bin/java             1         manual mode
  3            /opt/jdk-10-build-46/bin/java                1         manual mode
  4            /opt/jdk-9-build-46/bin/java                 1         manual mode
  5            /opt/jri-10-build-050/bin/java               10        manual mode
* 6            /opt/jri-10-build-46-v2/bin/java             1         manual mode
  7            /opt/jri-10-build-46-zero/bin/java           1         manual mode
  8            /opt/jri-10-build-46/bin/java                1         manual mode
```

jabrena, 2018-04-20:
Complete logs:

```
ev3dev#85|
ev3dev#85|##############################
ev3dev#85|# EV3Dev-lang-java Installer #
ev3dev#85|##############################
ev3dev#85|# Last update: 2018/04/14    #
ev3dev#85|##############################
ev3dev#85|
ev3dev#85|Init installer
ev3dev#85|
ev3dev#85|#########################################
ev3dev#85|# PLATFORM SECTION                      #
ev3dev#85|#########################################
ev3dev#85|
ev3dev#85|Platform detected: ev3
ev3dev#85|
ev3dev#85|
ev3dev#85|#########################################
ev3dev#85|# JAVA SECTION                          #
ev3dev#85|#########################################
ev3dev#85|
ev3dev#85|/usr/bin/java
ev3dev#85|Found java executable in PATH
ev3dev#85|But we will upgrade the Java version
ev3dev#85|/opt/jri-10-build-050
ev3dev#85|
ev3dev#85|Downloading new Java...
ev3dev#85|[sudo] password for robot: --2018-04-20 21:49:56--  https://github.com/ev3dev-lang-java/openjdk-ev3/releases/download/v0.5.0/jri10-ev3.tar.gz
ev3dev#85|Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
ev3dev#85|Connecting to github.com (github.com)|192.30.253.112|:443... connected.
ev3dev#85|HTTP request sent, awaiting response... 302 Found
ev3dev#85|Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/116190587/02cfaaf6-3a6d-11e8-906b-b7a02e1243ff?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180420%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180420T214958Z&X-Amz-Expires=300&X-Amz-Signature=7e28efcc28e9350e77791e46438ce9394f8e9cbf63d6212c688dd56c663f98ea&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Djri10-ev3.tar.gz&response-content-type=application%2Foctet-stream [following]
ev3dev#85|--2018-04-20 21:50:00--  https://github-production-release-asset-2e65be.s3.amazonaws.com/116190587/02cfaaf6-3a6d-11e8-906b-b7a02e1243ff?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180420%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180420T214958Z&X-Amz-Expires=300&X-Amz-Signature=7e28efcc28e9350e77791e46438ce9394f8e9cbf63d6212c688dd56c663f98ea&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Djri10-ev3.tar.gz&response-content-type=application%2Foctet-stream
ev3dev#85|Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.64.40
ev3dev#85|Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.64.40|:443... connected.
ev3dev#85|HTTP request sent, awaiting response... 200 OK
ev3dev#85|Length: 21960065 (21M) [application/octet-stream]
ev3dev#85|Saving to: '/home/robot/installer/jri10-ev3.tar.gz'
ev3dev#85|
ev3dev#85|     0K .......... .......... .......... .......... ..........  0% 30.4K 11m44s
ev3dev#85|    50K .......... .......... .......... .......... ..........  0% 40.6K 10m14s
ev3dev#85|   100K .......... .......... .......... .......... ..........  0% 39.4K 9m49s
ev3dev#85|   150K .......... .......... .......... .......... ..........  0% 88.7K 8m20s
ev3dev#85|   200K .......... .......... .......... .......... ..........  1% 55.5K 7m56s
ev3dev#85|   250K .......... .......... .......... .......... ..........  1% 79.5K 7m20s
ev3dev#85|   300K .......... .......... .......... .......... ..........  1%  254K 6m28s
ev3dev#85|   350K .......... .......... .......... .......... ..........  1% 94.1K 6m7s
ev3dev#85|   400K .......... .......... .......... .......... ..........  2%  208K 5m36s
ev3dev#85|   450K .......... .......... .......... .......... ..........  2%  138K 5m17s
ev3dev#85|   500K .......... .......... .......... .......... ..........  2%  195K 4m57s
ev3dev#85|   550K .......... .......... .......... .......... ..........  2%  113K 4m47s
ev3dev#85|   600K .......... .......... .......... .......... ..........  3%  117K 4m38s
ev3dev#85|   650K .......... .......... .......... .......... ..........  3% 45.8K 4m50s
ev3dev#85|   700K .......... .......... .......... .......... ..........  3%  253K 4m36s
ev3dev#85|   750K .......... .......... .......... .......... ..........  3%  342K 4m22s
ev3dev#85|   800K .......... .......... .......... .......... ..........  3%  335K 4m9s
ev3dev#85|   850K .......... .......... .......... .......... ..........  4%  401K 3m58s
ev3dev#85|   900K .......... .......... .......... .......... ..........  4%  431K 3m47s
ev3dev#85|   950K .......... .......... .......... .......... ..........  4%  183K 3m41s
ev3dev#85|  1000K .......... .......... .......... .......... ..........  4%  515K 3m32s
ev3dev#85|  1050K .......... .......... .......... .......... ..........  5%  268K 3m25s
ev3dev#85|  1100K .......... .......... .......... .......... ..........  5%  130K 3m22s
ev3dev#85|  1150K .......... .......... .......... .......... ..........  5% 59.9K 3m28s
ev3dev#85|  1200K .......... .......... .......... .......... ..........  5%  123K 3m25s
ev3dev#85|  1250K .......... .......... .......... .......... ..........  6% 35.2K 3m39s
ev3dev#85|  1300K .......... .......... .......... .......... ..........  6%  583K 3m32s
ev3dev#85|  1350K .......... .......... .......... .......... ..........  6% 41.7K 3m41s
ev3dev#85|  1400K .......... .......... .......... .......... ..........  6% 33.4K 3m53s
[... 385 lines cut ...]
[... cut ...]
ev3dev#85| 21400K .......... .......... .......... .......... .....     100%  303K=5m27s
ev3dev#85|
ev3dev#85|2018-04-20 21:55:32 (65.6 KB/s) - '/home/robot/installer/jri10-ev3.tar.gz' saved [21960065/21960065]
ev3dev#85|
ev3dev#85|Java package acquired, installing...
ev3dev#85|Installation complete. Java version:
ev3dev#85|openjdk version "10" 2018-03-20
ev3dev#85|OpenJDK Runtime Environment (build 10+46-ev3)
ev3dev#85|OpenJDK Client VM (build 10+46-ev3, mixed mode)
ev3dev#85|Dumping class cache...
```

jabrena, 2018-04-20:
Nice refactoring @JakubVanek

closed 2018-04-20 by jabrena