ev3dev-lang-java/ev3dev-lang-java #297: Disable Brickman on Startup

jabrena, 2017-06-04:


juan-medina, 2017-06-04:
http://pubs.opengroup.org/onlinepubs/009695399/functions/kill.html

juan-medina, 2017-06-04:
https://github.com/jnr/jnr-posix

jabrena, 2017-06-17:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/270

jabrena, 2017-06-17:
Some Bricks running with `EV3Dev` has a LCD out of the box. 
EV3Dev project includes a visual tool to manage some aspects of the brick, the name Brickman.

If the applications requires to use the LCD, it is necessary to disable brickman.
Besides, it is necessary to enable at the end of the execution in a quiet way.

In java, it is possible execute a logic at the end of any java process, the feature is the following one:

https://docs.oracle.com/javase/7/docs/api/java/lang/Runtime.html#addShutdownHook(java.lang.Thread)
https://www.tutorialspoint.com/java/lang/runtime_addshutdownhook.htm

Other examples:
https://gist.github.com/jabrena/164f9fee32f554895c1f5bb3d2e6ea63

closed 2017-06-17 by jabrena

jabrena, 2017-06-17:
Video example:
https://www.youtube.com/watch?v=E2MV6_4um0A