ev3dev-lang-java/ev3dev-lang-java #298: Improve Button support

jabrena, 2017-06-04:
It is necessary to run the following example with accuracy:

``` java
package examples.sensors;

import ev3dev.sensors.Button;
import lombok.extern.slf4j.Slf4j;

public @Slf4j class ButtonTest2 {

    public static void main(final String[] args){
        log.info("Demo");

        Button.UP.waitForPress();
        log.info("Pressed UP");

        Button.DOWN.waitForPress();
        log.info("Pressed DOWN");

        Button.LEFT.waitForPress();
        log.info("Pressed LEFT");

        Button.RIGHT.waitForPress();
        log.info("Pressed RIGHT");

        Button.ESCAPE.waitForPress();
        log.info("Pressed ESCAPE");

        Button.ENTER.waitForPress();
        log.info("Pressed ENTER");

    }
}
```

jabrena, 2017-07-08:
@jartu, Many thanks for the contribution. This issues was fixed.

closed 2017-07-08 by jabrena

jartu, 2017-07-08:
No problem, happy to contribute 😁

jabrena, 2017-07-12:
Linked with this example:
https://github.com/ev3dev-lang-java/examples/pull/1

To use this new feature, it is necessary this SNAPSHOT:
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/0.7.1-SNAPSHOT

commit 29ce2c9f (2017-07-08): Implementation of the EV3Key (including unit test). Fixes #298 .
commit 8dff1951 (2017-07-08): Merge pull request #357 from jartu/develop

Implementation of EV3Key - fixes #298.