ev3dev-lang-java/ev3dev-lang-java #351: Functionality of the lejos.hardware.Key#simulateEvent(int)

jartu, 2017-07-04:
Hi, 

I am working on an implementation of the `ev3dev.sensors.EV3Key` class which inherits from implements the mentioned `lejos.hardware.Key`. I am not quite sure what its `simulateEvent(int)` method should do and the LeJOS [javadoc](http://www.lejos.org/ev3/docs/lejos/hardware/Key.html#simulateEvent-int-) doesn't contain any description.

I am not even sure what the integer "event" parameter refers to... I guess it may be one of the LEFT, RIGHT, etc. constants but then it's not clear if it should simulate key "press" or "release".

My guess is that it should effectively:
1) Invoke the registered listeners.
2) Notify any threads waiting in the `#waitForPress()` and `#waitForPressAndRelease()` methods.

Can anyone please shed some light on this for me?

J.

jabrena, 2017-07-05:
Hi @jartu, it is a good question.

I am not sure too. I reviewed Javadocs and the NXT implementation and this method is new.
http://www.lejos.org/nxt/nxj/api/lejos/nxt/Button.html
http://www.lejos.org/ev3/docs/lejos/hardware/Button.html
http://www.lejos.org/ev3/docs/lejos/hardware/Key.html

In `EV3`, the implementation is better. Every object is returned an object based on an Interface:

https://github.com/ev3dev-lang-java/lejos-ev3-code-mirror/blob/master/ev3classes/src/lejos/hardware/Button.java

``` 
public static final Key ENTER = BrickFinder.getDefault().getKey("Enter");
```

I found some usages on Github:
https://github.com/lego04/robot/blob/master/robot/src/robot/Main.java
https://github.com/chbe89/mindstormer/blob/master/src/edu/kit/mindstormer/program/implementation/test/MoveFixedDistance.java
https://github.com/chbe89/mindstormer/blob/master/src/edu/kit/mindstormer/program/implementation/test/MoveFixedDistance.java

I suspect that the usage is more for Testing. If the Button in some object is waiting for some event, it is possible to set values to simulate the event. I think that it is interesting the feature but it is better if we fix before this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/298

Currently, we can`t handle the specific event for the any button. In current stable release, we only manage the case of: Button.waitForAnyPress();
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/273

``` 
package examples.sensors;

import ev3dev.sensors.Button;

public class ButtonExample {

    public static void main(final String[] args){

        System.out.println("Button example");

        Button.waitForAnyPress();

        System.out.println("Button pressed");
    }
}
``` 

If could be a good contribution if you are able to run this example:

``` 
package examples.sensors;

import ev3dev.sensors.Button;
import lombok.extern.slf4j.Slf4j;

public @Slf4j class ButtonTest2 {

    public static void main(final String[] args){
        log.info("Demo");

        Button.UP.waitForPress();
        log.info("Pressed UP");

        Button.DOWN.waitForPress();
        log.info("Pressed DOWN");

        Button.LEFT.waitForPress();
        log.info("Pressed LEFT");

        Button.RIGHT.waitForPress();
        log.info("Pressed RIGHT");

        Button.ESCAPE.waitForPress();
        log.info("Pressed ESCAPE");

        Button.ENTER.waitForPress();
        log.info("Pressed ENTER");

    }
}
```

Normally, I search on Github the usages of the community and it is more common the usage of a specific button event than the usage of the new method simulateEvent.

Many thanks @jartu 

Juan Antonio

jabrena, 2017-07-05:
In my case, I execute the programs from an `ssh`  connection from Intellij, when I have to finish a java process, in my terminal I execute: 

``` 
pkill java
```

For EV3, it is perfect. I think that the CPU for an EV3 only is able to run one program at time for BrickPi/PiStorms Bricks, the CPU resources are better so, in general I don´t use the Button to exit manually the programs, I interact the OS to kill the process but you are using some classes from `local navigation stack from LeJOS`, you know that some methods are `non-blocking` so the usage of the button is some way to not allow finish the execution in the main thread.

https://github.com/ev3dev-lang-java/lejos-navigation

``` java
import ev3dev.sensors.Button;
import ev3dev.sensors.ev3.EV3UltrasonicSensor;
import lejos.hardware.port.SensorPort;
import lejos.robotics.RangeFinderAdapter;
import lejos.robotics.navigation.DifferentialPilot;
import lejos.robotics.objectdetection.Feature;
import lejos.robotics.objectdetection.FeatureDetector;
import lejos.robotics.objectdetection.FeatureListener;
import lejos.robotics.objectdetection.RangeFeatureDetector;

import java.io.IOException;

public class FeatureAvoider2 {
     
    static final float MAX_DISTANCE = 50f;
    static final int DETECTOR_DELAY = 1000;
 
    public static void main(String[] args) throws IOException {

        final PilotConfig pilotConf = new PilotConfig();
        final DifferentialPilot pilot = pilotConf.getPilot();

        final EV3UltrasonicSensor ir = new EV3UltrasonicSensor(SensorPort.S1);
        final RangeFeatureDetector detector = new RangeFeatureDetector(
                new RangeFinderAdapter(ir.getDistanceMode()), MAX_DISTANCE, DETECTOR_DELAY);

        detector.enableDetection(true);
        pilot.forward();
         
        detector.addListener(new FeatureListener() {
            public void featureDetected(Feature feature, FeatureDetector detector) {
                detector.enableDetection(false);
                pilot.travel(-30);
                pilot.rotate(30);
                detector.enableDetection(true);
                pilot.forward();
            }       
        });

        Button.waitForAnyPress();
    }
}
```

jartu, 2017-07-06:
OK, I am leaving the #simulateEvent(int) unimplemented because I am still not sure what it should do. 

I have submitted a pull request (https://github.com/ev3dev-lang-java/ev3dev-lang-java/pull/357/files) with the EV3Key implementation that solves issue #298. Once this change gets approved I can also change the Button class to use the EV3Key better and I can also write a better version of the https://github.com/ev3dev-lang-java/examples/blob/develop/ev3dev-lang-java/src/main/java/ev3dev/sensors/ButtonExample.java .

Also note: With my implementation, it should not be a problem to call waitFor* from different threads - i.e. the limitation mentioned in the Button class no longer applies.

jabrena, 2017-07-06:
Hi @jartu, 

Tomorrow, I will do a Code Review about the Pull Request and I will test the new features.

Cheers

jabrena, 2017-07-08:
Test on Brick 1: Passed:

``` java
package examples.sensors;

import ev3dev.sensors.Button;

public class ButtonExample {

    public static void main(final String[] args){

        System.out.println("Button example");

        Button.waitForAnyPress();

        System.out.println("Button pressed");
    }
}
```

jabrena, 2017-07-08:
Test on Brick 2: **Passed**

``` java
package ev3dev.sensors;

public class ButtonExample2 {

    public static void main(final String[] args){
        System.out.println("Extended Button Example");

        System.out.println("Press UP");
        Button.UP.waitForPress();
        System.out.println("Pressed UP");

        System.out.println("Press DOWN");
        Button.DOWN.waitForPress();
        System.out.println("Pressed DOWN");

        System.out.println("Press LEFT");
        Button.LEFT.waitForPress();
        System.out.println("Pressed LEFT");

        System.out.println("Press RIGHT");
        Button.RIGHT.waitForPress();
        System.out.println("Pressed RIGHT");

        System.out.println("Press ESCAPE");
        Button.ESCAPE.waitForPress();
        System.out.println("Pressed ESCAPE");

        System.out.println("Press ENTER");
        Button.ENTER.waitForPress();
        System.out.println("Pressed ENTER");

    }
}
```

jabrena, 2017-07-08:
Unit tests: **Passed**

@jartu, I liked so much your tests. I am working on this issue to improve the coverage of the whole project:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/209

I think that Unit Testing for Robotics projects is good in general.

jabrena, 2017-07-08:
In relation to your original question, the method: _simulateEvent_ is not relevant in the functionality of the class Button so we could skip. Now the usage of the Sensor Button for EV3 platform is complete.

Feel free to add new examples or collaborate in other areas. Your syntax is modern and I noticed you have a good professional Java background.

I am going to close this support issue.

closed 2017-07-08 by jabrena