ev3dev-lang-java/ev3dev-lang-java #354: [ROS] Develop a ROS node to Subscribe to topic: cmd_vel using a geometry_msgs.msg/Twist

jabrena, 2017-07-05:
Develop a ROS Node to subscribe to topic `cmd_vel` which use a Twist message

jabrena, 2017-07-05:
https://github.com/ihmcrobotics/ihmc-open-robotics-software/blob/develop/IHMCROSTools/src/us/ihmc/utilities/ros/publisher/RosOdometryPublisher.java
https://github.com/yamunamaccarana/TCA/blob/master/Navigator/src/tca/RobotDriver.java
https://github.com/jmacglashan/feedback_advantage/blob/master/src/main/java/experiments/turtlebot/TurtleExplorerControl.java
https://github.com/keroma08/lejos/tree/master/ros/nxt_lejos_apps/nxt_lejos_monitor
https://github.com/kylealanr/CS2133/blob/master/Assignment4/Robots/src/RobotInstructor.java
https://github.com/elmot/ev3javabrick/blob/master/ros-ev3-node/src/main/java/elmot/ros/nxt/NXTNode.java
https://github.com/elmot/ev3javabrick/blob/master/ros-ev3-node/src/main/java/elmot/ros/ev3/Ev3Node.java

``` java
        Subscriber<Twist> motorSpeedSubscriber = connectedNode.newSubscriber(namespace.join("cmd_vel"), Twist._TYPE);
        motorSpeedSubscriber.addMessageListener(new MessageListener<Twist>() {
            @Override
            public void onNewMessage(Twist msg) {
                double x = msg.getLinear().getX();
                double y = msg.getAngular().getZ();
                if (Math.abs(x) < 0.02) x = 0;
                if (Math.abs(y) < 0.02) y = 0;
                double s = x < 0 ? -1 : 1;
                double speed = Math.sqrt(x * x + y * y) / Math.sqrt(2);
                double twist = -200 * y / speed;
                speed *= 100 * s;
                try {
                    brick.MOTOR.timeSync(0, MotorFactory.MOTORSET.BC, (int) speed, (int) twist, 1000, MotorFactory.BRAKE.COAST);
                } catch (IOException e) {
                    Ev3Node.this.connectedNode.getLog().error("Motor error", e);
                }
            }
        });

```

jabrena, 2017-09-14:
http://docs.ros.org/api/geometry_msgs/html/msg/Twist.html

jabrena, 2017-09-14:
http://wiki.ros.org/geometry_msgs

jabrena, 2017-09-14:
Now, the bricks are able to be teleoperated in a easy way.

closed 2017-09-14 by jabrena

RaghavSaxena96, 2019-06-06:
how can I send a Twist geometry msg to Ros server  in a eaisest way without knowing much about ros

jabrena, 2019-06-06:
Hi @RaghavSaxena96, Can you create a new issue about the ROS topic?