ev3dev-lang-java/ev3dev-lang-java #358: Add methods: getRemoteCommand & getRemoteCommands on EV3IRSensor

jabrena, 2017-07-09:
Currently, the class EV3IRSensor has all modes implemented but in LeJOS API exist 2 methods pending to implement:

```
int | getRemoteCommand(int chan)Return the current remote command from the specified channel.
void | getRemoteCommands(byte[] cmds, int offset, int len)Obtain the commands associated with one or more channels.
```

It is necessary to add methods: getRemoteCommand & getRemoteCommands in the class: EV3IRSensor.

https://ev3dev-lang-java.github.io/docs/api/latest/ev3dev-lang-java/ev3dev/sensors/ev3/EV3IRSensor.html
http://www.lejos.org/ev3/docs/lejos/hardware/sensor/EV3IRSensor.html

jabrena, 2017-07-09:
@jartu, Can you review it?

jartu, 2017-07-09:
Sure, I'll take a look at it.

PatrickSalle, 2018-02-06:
I use Intelij to compile my programs. In the ev3dev-lang-java:0.7.0.jar 
I have not the getRemoteCommand methods.
What I have to do to get this method?
In my external librairies I refer Gradle:com.github.ev3dev-lang-java:0.7.0

Thank you if you can help me.
Patrick Sallé

jabrena, 2018-02-06:
Hi @PatrickSalle,

Give me few hours to review this issue, I will give you feedback today.

Juan Antonio

jabrena, 2018-02-06:
Hi @PatrickSalle, it is true but do you have one example that you need to use that commands?
Tell me.

Cheers

PatrickSalle, 2018-02-07:
Hi,
I just try to interact with the robot during the execution of program, to
modify
if necesary its behaviour.
Best regards.

jabrena, 2018-02-07:
Hi @PatrickSalle, 

do you have some example using that methods?

Juan Antonio

jabrena, 2018-02-07:
I found this discussion:
https://lejos.sourceforge.io/forum/viewtopic.php?t=7665

Maybe, it is a good point to begin.

Juan Antonio

jabrena, 2018-02-07:
https://github.com/mindstorms-cop/lejos-ev3/blob/master/ev3menutools/src/org/lejos/ev3/menutools/ircontrol/RemoteControl.java

jabrena, 2018-02-07:
https://sites.google.com/site/stanfordpcs/lejos/lejos-exploring-more

jabrena, 2018-02-07:
https://alwaysbusycorner.com/tag/lejos/

jabrena, 2018-02-07:
Maybe with this example is enough to add this feature.
I will try to implement this week.

PatrickSalle, 2018-02-07:
Thank you. I if I really use this feature in a significative example I will
post
the code.

PatrickSalle, 2018-04-09:
I travelled for a long time, and have not work on my project...
Have look at the getRemoteCommand?
Best regards.
Patrick.

jabrena, 2018-04-30:
One idea to implement the method:

``` java
		// Get the IR commands
			byte [] cmds = new byte[4];
			ir.getRemoteCommands(cmds, 0, cmds.length);
			
			// Figure out which channel is active:
			int command = 0;
			for(int i=0;i<4;i++) {
				if(cmds[i] > 0) {
					channel = i;
					command = cmds[i];
				}
			}
```

jabrena, 2018-04-30:
Original implementation:
https://github.com/ev3dev-lang-java/lejos-ev3-code-mirror/blob/master/ev3classes/src/lejos/hardware/sensor/EV3IRSensor.java

jabrena, 2018-04-30:
Hi @PatrickSalle,

I could implement the methods, the solution was not very complicated but I am not able to test in few days. If you have time, I could create a Snapshot to test in your project.

Do you use `Debian Jessie` or `Debian Stretch`?

Juan Antonio

jabrena, 2018-05-04:
Example to test the feature:

``` 
package example;

import ev3dev.sensors.ev3.EV3IRSensor;
import lejos.hardware.port.SensorPort;
import lejos.utility.Delay;

public class IRTest {

    public static void main(String[] args) {

        EV3IRSensor ir = new EV3IRSensor(SensorPort.S1);
        
        for(int i=0;i<40;i++) {
            System.out.println("cmd " + ir.getRemoteCommand(0)); // gets command from channel 1
            Delay.msDelay(500);
        }

    }

}
```

jabrena, 2018-05-04:
Example to test the method: `getRemoteCommands`

``` java
package example;

import ev3dev.sensors.ev3.EV3IRSensor;
import lejos.hardware.port.SensorPort;
import lejos.utility.Delay;

public class IRTest2 {

    public static void main(String[] args) {

        EV3IRSensor ir = new EV3IRSensor(SensorPort.S1);

        byte[] commands = new byte[EV3IRSensor.IR_CHANNELS];

        for(int i=0;i<40;i++) {
            ir.getRemoteCommands(commands, 0, commands.length);
            System.out.println("cmd1 " + commands[0]); // gets command from channel 1
            System.out.println("cmd2 " + commands[1]); // gets command from channel 2
            System.out.println("cmd3 " + commands[2]); // gets command from channel 3
            System.out.println("cmd4 " + commands[3]); // gets command from channel 4

            Delay.msDelay(500);
        }

    }

}
```

PatrickSalle, 2018-05-04:
Merci beaucoup. Comme je suis en voyage, je testerai à mon retour.
Cordialement, Patrick Sallé.

closed 2018-05-04 by jabrena

PatrickSalle, 2018-05-14:
Hello,
I have tried to create the EV3IRSensor class with the code
accessible at the address
https://github.com/ev3dev-lang-java/lejos-ev3-code-
mirror/blob/master/ev3classes/src/lejos/hardware/sensor/EV3IRSensor.java
I am not sure that it is the rigth method,
In fact I have no access to UARTPort.

import lejos.hardware.port.UARTPort;

Patrick Sallé.

jabrena, 2018-05-14:
Hi @PatrickSalle,

Can you paste the example that you are trying to test please?

PatrickSalle, 2018-05-15:
I am using Intellij Idea.
One of my projects is just a test of the main fonctionnalities available in
Ev3dev Lejos.

This project contains  two files :

MyThirdRobot.java with the code to test getRemoteCommand as you have
suggested
EV3IRSensor.java with the code of your implementation of getRemoteCommand.
In this file the

import lejos.hardware.port.UARTPort;

fail.

I enclose a tar of the project :