ev3dev-lang-java/ev3dev-lang-java #359: Error: Could not find or load main class

NielsAMG, 2017-07-09:
When I try to launch a programm it gives me this error: 
Error: Could not find or load main class.

jartu, 2017-07-09:
I assume you're launching the program with something like this?:

`java -jar my-jar-file.jar`

right? If that's the case, you have two options, either you can mention the class with the public static main method in the MANIFEST.MF file (typically in src\main\resources\META-INF folder) or you can launch it like this:

`java -cp my-jar-file.jar my.fully.qualified.Main`

NielsAMG, 2017-07-09:
@jartu Im trying to launch the program with

`java examples.MyFirstRobot` in /src/main/java

I also tried to launch it with that command

`java -cp . examples.MyFirstRobot`

jartu, 2017-07-09:
The problem with this approach

NielsAMG, 2017-07-09:
nhh?

jartu, 2017-07-09:
@NielsAMG (Sorry, that previous comment was an attempt to type the reply on my phone when I decided to wait till I come home to my computer but I somehow managed to post the unfinished reply...)

So here it is. The problem with launching the `MyFirstRobot` program like this is that:
1) The . (dot) that you're using refers to the "current directory" (i.e. the `/src/main/java` directory) where there are only Java source files, not the needed compiled *.class files. Those would typically be in the `/build/classes/java/main` directory.
2) But even if you specified the correct directory, you're still missing the necessary *.jar libraries that the program needs at runtime.

Moreover, I have just realized that I haven't asked you about where you actually attempt to run this. Are you running this on the EV3 brick through ssh console? Normally (e.g. for ev3dev-lang-java examples), you would do the following:

1) Download the examples to your computer.
2) Modify the `src\main\resources\META-INF\MANIFEST.MF` such that "Main-Class" entry points to the class that you want the jar to run by default.
3) Modify the `/config.gradle` file to have the "host" entry point to the IP address of your brick.
4) Run the
`gradle deploy`
which would build the whole thing into one "fat jar" (i.e. a jar that contains all the necessary library jars) and deploy it to your brick.
5) Establish an ssh connection to your brick (at the IP address of your brick and probably with the default robot/maker username and password).
6) Then execute in the default `/home/robot` directory the following:
java -jar your-jar-file.jar

... and that's it. If you don't want to keep modifying the MANIFEST.MF file every time you want to run a different class (with a public static main method), you can skip the step 2 and in step 6 execute:
java -cp your-jar-file.jar your.Main

Does this help?

jartu, 2017-07-09:
Plus I assume you went through the steps outlined in the "Roadmap" in https://github.com/ev3dev-lang-java/ev3dev-lang-java, correct?

NielsAMG, 2017-07-09:
@jartu First of all thank you for that, but can we write about that? To not spam this issue, Ive got Discord (SyntaxFehler#8423), Steam (https://steamcommunity.com/id/niels1006) and Skype (niels.hermsen@hotmail.com). Thank you :)

jartu, 2017-07-09:
OK, I've sent you an invitation on Skype. 

But for your own protection, I'd suggest you delete all your contacts from the above post as this is a public forum.

jabrena, 2017-07-09:
Hi @NielsAMG, did you solve the doubt?

Besides as @jartu say, if you execute the task: `deployAndRun` from the IDE (IntelliJ or Eclipse) you could execute the task and see the logs or console messages.

Maybe, it is necessary to improve the documentation.

Tell us.

Cheers

Juan Antonio

NielsAMG, 2017-07-09:
Hey @jabrena,

I just `gradle deploy` and then launched the program with `java -cp template_project_gradle-all-0.1.1.jar examples.MyFirstRobot`. 

My bad was, that I didn't know that I have to `gradle deploy` in the PowerShell.

Thank you all for your help 💯 

Cheers

jabrena, 2017-07-09:
Good point to improve the documentation. (Gradle Tasks)

I have just created a chat for everyone. :)
https://gitter.im/ev3dev-java-chat/main

Take a look the project and if you don´t like something or you have some doubt, create an issue.
The core API is stable, now exist 2 open

closed 2017-07-09 by jabrena