ev3dev-lang-java/ev3dev-lang-java #396: Expose some EV3 Sensor Constants to be used by ROS publishers

jabrena, 2017-09-15:


jabrena, 2017-09-15:
I am going to create a `feature` branch in order to add the solution.

aswin, 2017-09-15:
Hi, this is @aswin but I believe you have tagged the wrong person? :)

jabrena, 2017-09-15:
Hi @aswin, sorry mate. this is the @Asw1n for this matter.

Good weekend. :)

jabrena, 2017-09-15:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/347

jabrena, 2017-09-23:
No answer. Included.

closed 2017-09-23 by jabrena

Asw1n, 2017-09-26:
> This issue requires some analysis because expose some Sensor data in a
> public way to be used by Third party libraries.
>
> Some Sensors like EV3IRSensor or EV3UltrasonicSensor are used by the ROS
> library:
> https://github.com/ev3dev-lang-java/ros to send messages with the type:
> "sensor_msgs/Range"
> http://docs.ros.org/api/sensor_msgs/html/msg/Range.html
>
> This ROS Message, has the following constructor and it used to publish the
> information to ROS:
>
> final Message message = new Range(
>                     header,
>                     Range.INFRARED,
>                     0.5f,
>                     0.05f,
>                     0.55f,
>                     distance
>             );
>
>             final Message message = new Range(
>                     header,
>                     Range.ULTRASOUND,
>                     0.5f,
> [... 34 lines cut ...]
> The idea is to define constants for min/max variables like a Sensor
> definition and expose as a final static public in order to be used by
> everyone because is part of the definition and usage of the sensor.
>
> Hi @Asw1n <https://github.com/asw1n>, what is your opinion?

Hi Juan,

Sorry it took me so long to reply. I've been on holidays and also very much
occupied with other matters.

I think I would create a SampleproviderMetadata class. This class has a
LinkedList member that holds the actual metadata in named items. The class
has methods to maintain the list and to output it formatted for ROS. The
class is instantiated and filled during creation of a SampleProvider class.
So that there is always exactly one instance of the metadata class for each
type of SampleProvider. The SampleProvider interface should have a method
to retrieve the SampleproviderMetadata object.

Does this makes sense to you?

Beware that not all Sensors provide data that is easily described in
metadata.

Also take note that filters are also SampleProviders. They alter some of
the metadata and maybe even add metadata. Once you start doing metadata
you'll have to go all the way. It will require a lot of thinking. The
coding will be the easy part.

Best,
Aswin