ev3dev-lang-java/ev3dev-lang-java #402: Convert LDD model to stl format

jabrena, 2017-09-20:
https://youtu.be/7FNUxEfvd48

jabrena, 2017-09-20:
https://www.shapeways.com/tutorials/exporting-from-blender

jabrena, 2017-09-20:
https://www.shapeways.com/tutorials/prepping_blender_files_for_3d_printing

jabrena, 2017-09-20:
https://daler.github.io/blender-for-3d-printing/printing/export-stl.html

jabrena, 2017-09-20:
https://youtu.be/TaOeQpQMUQo

jabrena, 2017-09-20:
http://wiki.ldraw.org/index.php?title=LDraw_to_LDD_conversion

jabrena, 2017-09-20:
https://www.hitechnic.com/downloadnew.php?category=23

jabrena, 2017-09-20:
http://wiki.ros.org/urdf/XML/link

JakubVanek, 2018-09-06:
There is also a format called LDraw. There are many editors for this format. In particular, I think LDCad is better than LDD.

JakubVanek, 2018-09-06:
Quick google revealed this utility: https://github.com/kristov/ldraw2stl

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena