ev3dev-lang-java/ev3dev-lang-java #405: Build Openj9

jabrena, 2017-09-22:
https://www.eclipse.org/openj9/oj9_build.html

jabrena, 2017-09-24:
https://www.eclipse.org/openj9/oj9_build.html

```
docker build -t openj9 -f Dockerfile .
docker run -it openj9

docker build -t ev3dev-armel -f debian-jessie-armel.dockerfile .
docker run -t ev3dev-armel
```

jabrena, 2017-09-24:
Problem with jessie version:

``` 
  ev3dev-jessie git:(master) ✗ docker build -t ev3dev-armel -f debian-jessie-armel.dockerfile .

Sending build context to Docker daemon  20.99kB
Step 1/11 : FROM ev3dev/debian-jessie-armel-qemu-minbase
 ---> 342e80581d45
Step 2/11 : COPY apt.sources.list.debian /etc/apt/sources.list
 ---> Using cache
 ---> c42559e14819
Step 3/11 : RUN dpkg --add-architecture i386 &&     apt-key adv --keyserver pgp.mit.edu --recv-keys D57D95AF93178A7C &&     apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install --yes xz-utils         libc6:i386 libstdc++6:i386 libz1:i386 libncurses5:i386 libpython2.7:i386         build-essential cmake valac man-db bash-completion sudo vim nano less tree wget
 ---> Running in 9e2966c6888d
Executing: gpg --ignore-time-conflict --no-options --no-default-keyring --homedir /tmp/tmp.8CyvKXhwQs --no-auto-check-trustdb --trust-model always --primary-keyring /etc/apt/trusted.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-stable.gpg --keyserver pgp.mit.edu --recv-keys D57D95AF93178A7C
gpg: requesting key 93178A7C from hkp server pgp.mit.edu
gpg: key 93178A7C: public key "ppa.lechnology.com - Archive Signing Key <david@lechnology.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
Ign http://ftp.debian.org jessie InRelease
Get:1 http://security.debian.org jessie/updates InRelease [63.1 kB]
Get:2 http://ftp.debian.org jessie Release.gpg [2373 B]
Get:3 http://archive.ev3dev.org jessie InRelease [8418 B]
Get:4 http://emdebian.org jessie InRelease [30.7 kB]
Get:5 http://ftp.debian.org jessie Release [148 kB]
Get:6 http://security.debian.org jessie/updates/main Sources [259 kB]
Get:7 http://security.debian.org jessie/updates/contrib Sources [1298 B]
Get:8 http://security.debian.org jessie/updates/non-free Sources [20 B]
Get:9 http://security.debian.org jessie/updates/main armel Packages [537 kB]
Get:10 http://archive.ev3dev.org jessie/main Sources [21.2 kB]
Get:11 http://security.debian.org jessie/updates/contrib armel Packages [994 B]
Get:12 http://security.debian.org jessie/updates/non-free armel Packages [20 B]
Get:13 http://security.debian.org jessie/updates/main i386 Packages [546 kB]
Get:14 http://archive.ev3dev.org jessie/main armel Packages [48.5 kB]
Get:15 http://security.debian.org jessie/updates/contrib i386 Packages [2369 B]
Get:16 http://security.debian.org jessie/updates/non-free i386 Packages [20 B]
Get:17 http://archive.ev3dev.org jessie/main i386 Packages [15.4 kB]
Ign http://emdebian.org jessie InRelease
Get:18 http://emdebian.org jessie/main Sources [6458 B]
Get:19 http://emdebian.org jessie/main armel Packages [8215 B]
Get:20 http://emdebian.org jessie/main i386 Packages [14.6 kB]
Get:21 http://ftp.debian.org jessie/main Sources [9166 kB]
Get:22 http://ftp.debian.org jessie/contrib Sources [58.9 kB]
Get:23 http://ftp.debian.org jessie/non-free Sources [119 kB]
Get:24 http://ftp.debian.org jessie/main armel Packages [8868 kB]
Get:25 http://ftp.debian.org jessie/contrib armel Packages [42.9 kB]
Get:26 http://ftp.debian.org jessie/non-free armel Packages [71.1 kB]
Get:27 http://ftp.debian.org jessie/main i386 Packages [9065 kB]
Get:28 http://ftp.debian.org jessie/contrib i386 Packages [58.1 kB]
Get:29 http://ftp.debian.org jessie/non-free i386 Packages [97.1 kB]
Fetched 29.3 MB in 31s (920 kB/s)
Reading package lists...
W: GPG error: http://emdebian.org jessie InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7DE089671804772E
Reading package lists...
Building dependency tree...
The following extra packages will be installed:
  binutils bsdmainutils bzip2 ca-certificates cmake-data cpp cpp-4.9 dpkg-dev
  fakeroot file g++ g++-4.9 gcc gcc-4.9 gcc-4.9-base:i386 groff-base ifupdown
  iproute2 isc-dhcp-client isc-dhcp-common krb5-locales libalgorithm-c3-perl
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libarchive-extract-perl libarchive13 libasan1 libatm1 libatomic1
  libbz2-1.0:i386 libc-dev-bin libc6-dev libc6-i686:i386 libcgi-fast-perl
  libcgi-pm-perl libclass-c3-perl libclass-c3-xs-perl libcloog-isl4
  libcpan-meta-perl libcurl3 libdata-optlist-perl libdata-section-perl
  libdb5.3:i386 libdns-export100 libdpkg-perl libelfg0 libexpat1
  libexpat1:i386 libfakeroot libfcgi-perl libffi6 libffi6:i386
  libfile-fcntllock-perl libgcc-4.9-dev libgcc1:i386 libgdbm3 libglib2.0-0
  libglib2.0-bin libglib2.0-data libglib2.0-dev libgmp10 libgnutls-deb0-28
  libgomp1 libgpm2 libgpm2:i386 libgssapi-krb5-2 libhogweed2 libicu52 libidn11
  libirs-export91 libisc-export95 libisccfg-export90 libisl10 libk5crypto3
  libkeyutils1 libkrb5-3 libkrb5support0 libldap-2.4-2 liblog-message-perl
[... 59 lines cut ...]
  libsasl2-modules-db libsoftware-license-perl libsqlite3-0 libsqlite3-0:i386
  libssh2-1 libssl1.0.0 libssl1.0.0:i386 libstdc++-4.9-dev libstdc++6:i386
  libsub-exporter-perl libsub-install-perl libtasn1-6 libterm-ui-perl
[... cut ...]
The command '/bin/sh -c dpkg --add-architecture i386 &&     apt-key adv --keyserver pgp.mit.edu --recv-keys D57D95AF93178A7C &&     apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install --yes xz-utils         libc6:i386 libstdc++6:i386 libz1:i386 libncurses5:i386 libpython2.7:i386         build-essential cmake valac man-db bash-completion sudo vim nano less tree wget' returned a non-zero code: 100
➜  ev3dev-jessie git:(master) ✗ docker run -t ev3dev-armel 
Unable to find image 'ev3dev-armel:latest' locally
docker: Error response from daemon: repository ev3dev-armel not found: does not exist or no pull access.
See 'docker run --help'.
➜  ev3dev-jessie git:(master) ✗ 
```

jabrena, 2017-09-24:
Running Stretch version:

```
docker build -t ev3dev-armel -f debian-stretch-armel.dockerfile .
```

jabrena, 2017-09-24:
Later, it is necessary to add the following dependencies used to build `openj9`

```
# Install required OS tools
RUN 
    g++-4.8 \
    gcc-4.8 \

apt-get update \
  && apt-get install -qq -y --no-install-recommends \
    autoconf \
    ca-certificates \
    ccache \
    cpio \
    file \
    git \
    git-core \
    libasound2-dev \
    libcups2-dev \
    libelf-dev \
    libfreetype6-dev \
    libnuma-dev \
    libx11-dev \
    libxext-dev \
    libxrender-dev \
    libxt-dev \
    libxtst-dev \
    make \
    openjdk-8-jdk \
    pkg-config \
    realpath \
    ssh \
    unzip \
    wget \
    zip \
  && rm -rf /var/lib/apt/lists/*
```

jabrena, 2017-09-24:
Good idea:
https://github.com/markehammons/languageshootout-jmh

jabrena, 2017-09-24:
http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

jabrena, 2017-09-24:
https://github.com/ev3dev-lang-java/docker-cross/blob/master/ev3dev-stretch/toolchain.cmake.armel

```
#RUN 
ln -s g++ /usr/bin/c++ \
  && ln -s g++-4.8 /usr/bin/g++ \
  && ln -s gcc /usr/bin/cc \
  && ln -s gcc-4.8 /usr/bin/gcc

ln -s gcc /opt/gcc-linaro-6.3.1-2017.02-x86_64_arm-linux-gnueabi/bin/arm-linux-gnueabi-gcc
ln -s g++ /opt/gcc-linaro-6.3.1-2017.02-x86_64_arm-linux-gnueabi/bin/arm-linux-gnueabi-g++
```

jabrena, 2017-09-24:
``` 
root@7d8f0e450da5:/home/compiler/openj9-openjdk-jdk9# /bin/bash  ./configure --with-freemarker-jar=/home/compiler/freemarker.jar
Warning: The generated configure file contains changes not present in the custom generated file.
Running autogen.sh to correct the situation
Using autoconf at /usr/bin/autoconf [autoconf (GNU Autoconf) 2.69]
Generating generated-configure.sh
Generating custom generated-configure.sh
Running custom generated-configure.sh
configure: Configuration created at Sun Sep 24 19:06:59 UTC 2017.
configure: configure script generated at timestamp 1506279925.
checking for basename... /usr/bin/basename
checking for bash... /bin/bash
checking for cat... /bin/cat
checking for chmod... /bin/chmod
checking for cmp... /usr/bin/cmp
checking for comm... /usr/bin/comm
checking for cp... /bin/cp
checking for cut... /usr/bin/cut
checking for date... /bin/date
checking for gdiff... no
checking for diff... /usr/bin/diff
checking for dirname... /usr/bin/dirname
checking for echo... /bin/echo
checking for expr... /usr/bin/expr
checking for file... /usr/bin/file
checking for find... /usr/bin/find
checking for head... /usr/bin/head
checking for gunzip... /bin/gunzip
checking for pigz... no
checking for gzip... /bin/gzip
checking for ln... /bin/ln
checking for ls... /bin/ls
checking for mkdir... /bin/mkdir
checking for mktemp... /bin/mktemp
checking for mv... /bin/mv
checking for nawk... /usr/bin/nawk
checking for printf... /usr/bin/printf
checking for rm... /bin/rm
checking for rmdir... /bin/rmdir
checking for sh... /bin/sh
checking for sort... /usr/bin/sort
checking for tail... /usr/bin/tail
checking for gtar... no
checking for tar... /bin/tar
checking for tee... /usr/bin/tee
checking for touch... /usr/bin/touch
checking for tr... /usr/bin/tr
checking for uname... /bin/uname
checking for uniq... /usr/bin/uniq
checking for wc... /usr/bin/wc
checking for which... /usr/bin/which
checking for xargs... /usr/bin/xargs
checking for gawk... no
checking for mawk... mawk
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for a sed that does not truncate output... /bin/sed
checking for cygpath... no
checking for greadlink... no
checking for readlink... /bin/readlink
checking for df... /bin/df
checking for cpio... /bin/cpio
checking for nice... /usr/bin/nice
checking for pandoc... no
checking build system type... armv7l-unknown-linux-gnueabi
checking host system type... armv7l-unknown-linux-gnueabi
checking target system type... armv7l-unknown-linux-gnueabi
checking openjdk-build os-cpu... linux-arm
checking openjdk-target os-cpu... linux-arm
checking compilation type... native
checking for top-level directory... /home/compiler/openj9-openjdk-jdk9
checking if custom source is suppressed (openjdk-only)... no
checking which variant of the JDK to build... normal
checking which debug level to use... release
checking which variants of the JVM to build... server
configure: error: unsupported OpenJ9 cpu armv7l
configure exiting with result code 1
```

jabrena, 2017-09-24:
Asking help to `Openj9` community

JakubVanek, 2017-12-26:
Hi,
I think OpenJDK 9 contains open-sourced ARM ports from Oracle, i.e. continuation of the discontinued EJDK8. Hope that helps.
Jakub Vaněk

JakubVanek, 2017-12-26:
Oh, I haven't noticed the #232.

JakubVanek, 2017-12-27:
It might be possible to use OpenJDK9 build flags for OpenJ9. I'll check that tomorrow (technically today, it's 2:20 AM here...)

jabrena, 2017-12-27:
Later, we could compare the results between OpenJDK vs OpenJ9 here:
https://github.com/ev3dev-lang-java/jvm-benchmark

JakubVanek, 2017-12-27:
Hmm, the hacking process so far includes even patching out vector load/store instructions in assembler helpers (armnathelp.s)...

JakubVanek, 2017-12-29:
I've succeeded with the build itself, but the JVM segfaults, more in eclipse/openj9#66.

JakubVanek, 2017-12-29:
Hmm, maybe I should have turned the debug mode on... :smile:

jabrena, 2017-12-31:
Hi @JakubVanek, 
I reviewed the issue: https://github.com/eclipse/openj9/issues/66 and I saw that you generated a build about OpenJ9: https://drive.google.com/file/d/1Fc-pTDjXu3nCqlaVhcowvPa2A2WUDCnm/view

Is it working for EV3?
Did you test on the Brick?

Merry Chrismas

Juan Antonio

JakubVanek, 2017-12-31:
Hi @jabrena,

I don't know yet. I'll get my old EV3 microSD card returned from a friend around Thursday. I'm probably going to test it next weekend.

Happy New Year

Jakub Vanek

JakubVanek, 2018-01-03:
Hmm, I've tested it and it doesn't work. It needs some further hacking of the build system/runtime/JIT.

jabrena, 2018-03-28:
Rejected. LEAN Criteria

closed 2018-03-28 by jabrena