ev3dev-lang-java/ev3dev-lang-java #410: Add support for NXT sensors

ammerzon, 2018-01-30:
Adding support for the following NXT sensors which are already available in leJOS:

- NXT light sensor 9844
- NXT color sensor 9694

jabrena, 2018-02-06:
Hi @ammerzon, 

Sorry by the delay. I will review the EV3Dev support in that area.
Why use NXT Sensors and not use EV3 Sensors?

Cheers

Juan Antonio

ammerzon, 2018-02-08:
Thank you for your reply. 

We use some EV3 sensors but also have the requirement to use NXT sensors in our project.

jabrena, 2018-02-09:
Oki, 
I will review if I have that sensors at home this weekend.

ammerzon, 2018-02-21:
Yes, I could test the Java prototype in my local environment.

Unfortunately I don't have the time right now to implement the support for the two sensor on my own.

viniciusccarvalho, 2019-02-20:
Any updates on this? I have those sensors I can help testing out any code

jabrena, 2019-02-21:
Hi @viniciusccarvalho,

This feature is stopped at the moment. 
What sensors are interested?

viniciusccarvalho, 2019-02-21:
I noticed that the NXT motors are not working. I'm new to ev3dev, but I ran the commands on linux to test motor speed, and stop/reset commands and they worked fine, so it has to be something with the java bindings.
This weekend I'll try to start from scratch on a fresh new install of ev3 stretch and see if the problem persists.

I have not tested the sensors, my bad I meant motors. But I can also test them since I have all sensors for NXT, but I believe the java bindings are not there for the sensors right?

Thank you

jabrena, 2019-02-22:
Hi, interesting your comment. I recommend to use the following snapshot version for everything:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2019-02-18/snapshot-ev3dev-stretch-ev3-generic-2019-02-18.img.xz

Why: at the end of 2018, we could integrate OpenJDK11 on EV3Dev. Indeed, I recommend to read the following article to increase the CPU:
https://lechnology.com/2018/06/overclocking-lego-mindstorms-ev3-part-2/

Juan Antonio

viniciusccarvalho, 2019-02-22:
@jabrena I'll give it a try, but any chance I could just clone the repo and use it instead? I'd rather just install it on my local maven repo than importing a set of jars. 

Any particular commit hash I should be pointing, or HEAD should be fine?

Regards

viniciusccarvalho, 2019-02-22:
@jabrena Still the same issue with the image, and even after overclocking. It really may be an NXT implementation issue here.

viniciusccarvalho, 2019-02-22:
@jabrena Just found out some interesting results (albeit still not clear to me why)

So, I know for a fact that running:

```
echo reset > $MC/command
echo 500 > $MC/speed_sp
echo 2000 > $MC/time_sp
echo run-timed > $MC/command
```

Works with the NXT motors.

So I looked into https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/actuators/lego/motors/BaseRegulatedMotor.java#L95

Which seems to be the source of my problems :)

And then checking https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/src/main/java/ev3dev/utils/Sysfs.java#L39

It seems that checking for permissions on the file is causing the problem, when I check the file permissions:
```
robot@ev3dev:/sys/class/tacho-motor/motor0$ ls -la
total 0
drwxr-xr-x 5 root root      0 Jan  1  1970 .
drwxr-xr-x 3 root root      0 Jan  1  1970 ..
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 address
--w--w---- 1 root ev3dev 4096 Feb 22 20:55 command
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 commands
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 count_per_rot
lrwxrwxrwx 1 root root      0 Feb 22 20:50 device -> ../../../ev3-ports:outA:lego-ev3-l-motor
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 driver_name
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 duty_cycle
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 duty_cycle_sp
drwxr-xr-x 2 root root      0 Feb 22 19:08 hold_pid
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 max_speed
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 polarity
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 position
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 position_sp
drwxr-xr-x 2 root root      0 Feb 22 19:08 power
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 ramp_down_sp
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 ramp_up_sp
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 speed
drwxr-xr-x 2 root root      0 Feb 22 19:08 speed_pid
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 20:55 speed_sp
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 state
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 stop_action
-r--r--r-- 1 root ev3dev 4096 Feb 22 19:08 stop_actions
lrwxrwxrwx 1 root root      0 Jan  1  1970 subsystem -> ../../../../../../../../../class/tacho-motor
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 20:55 time_sp
-rw-rw-r-- 1 root ev3dev 4096 Feb 22 19:08 uevent
robot@ev3dev:/sys/class/tacho-motor/motor0$
```

I'm not sure why would the `file.canWrite` return false here, as the permissions show it as writable to the group ev3dev (I'm assuming the app runs with a user on that group)

But I can tell you this, I just wrote this simple test:

```kotlin
  val motorPath = "/sys/class/tacho-motor/motor0"
    val commandFile = File("$motorPath/command")
    val speedFile = File("$motorPath/speed_sp")
    val timeFile = File("$motorPath/time_sp")
    println("Reseting motor")
    commandFile.writeText("reset")
    Delay.msDelay(800)
    println("Setting speed")
    speedFile.writeText("500")
    Delay.msDelay(800)
    println("Setting Time")
    timeFile.writeText("2000")
    Delay.msDelay(800)
    println("Running")
    commandFile.writeText("run-timed")
    Delay.msDelay(800)
```

And it works just fine. So the question I believe is: **why is file.canWrite returning false**, when we can clearly write to the file?

Hope this helps, I'm for one am relieved that at worst case I could just patch the ev3dev jar in order to make my bot to work

viniciusccarvalho, 2019-02-22:
One more thing to add to all this, I ran 7 times (why 7 I don't know) the runDeploy/undeploy procedure, without changing the code.

And to my surprise it worked 3 out of those 7 times, so the problem does seems to be non deterministic.

I have also noticed that my motors on the filesystem kept changing from mapped folders (motor0, motor2, motor8, motor7) I did not take notes if there was a correlation between the mount folder and when the code worked.

Besides this strange behavior I have a few points to highlight and maybe @jabrena you could help me out.

1. I noticed the code runs much slower than on a regular Lejos system. I'm talking about 5-10x slower here. I could probably be doing something wrong, but the time between a touch button press and my motor reacting on lejos is almost instant, but on the new code it took a good 2-3 seconds for it to respond to the push event. Also, only worked once ... :(

2. I am loading sounds and playing them (wav samples mono, 20kb in size) also took a good 4-5 seconds between the time I pressed the button and the time the sound came out of the speakers.

As reference, I'm using lejos way of polling for events on the sensor framework. Basically I have a thread that loops forever on all sensors:
```kotlin
fun start() {
        service.submit {
            while(true){
                buttonListeners.forEach {
                    it.readInput()
                }
            }
        }
    }

class ButtonListener(private val index: Int, private val sensorPort: Port, private val callback: (Event) -> Unit) : SensorListener{

    private val touchSensor =  EV3TouchSensor(sensorPort)
    private val sensorMode = touchSensor.touchMode

    override fun readInput() {
        val sample = FloatArray(sensorMode.sampleSize())
        sensorMode.fetchSample(sample, 0)
        val touched = sample[0] == 1.0f
        if(touched){
            callback(ButtonPressedEvent(index))
        }
    }

}
``` 
So if I detect a button press from the samples, I notify the main controller using a `ButtonPressedEvent`

And for sound this is the code I'm using (also runs on a separate thread)
```kotlin
class MediaManager(private val executor: ExecutorService) {

    val sound = Sound.getInstance()
    fun playSound(){

        executor.submit {
            sound.volume = 100
            sound.playSample(File("/home/robot/java/samples/coin_2.wav"))
        }
    }
}
```

Finally regarding the sound, in one of the executions of the program I got a weird error about endianness of the filesystem and the wav file, saying it could not play it. That was strange, happened only once and after it happened sound would not play anymore.

Wow, lots of things, but I hope I could have provided enough information for some debugging on your end.

JakubVanek, 2019-02-23:
Hmmm, `canWrite` looks familiar. We already had a similar issue here: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/554

viniciusccarvalho, 2019-02-23:
After some googling, it seems that canXXX methods are actually broken in Java ( on my mobile now, I can update later with the Sun decade old bug later ).
So why not change that method to just try to write and catch an exception instead? 
In case file is writable ( most cases) it would avoid at least one branch execution.
Seems like an easy change that should not impact anything

jabrena, 2019-02-24:
Hi @viniciusccarvalho, 

Can you create an issue about this analysis?

Juan Antonio

jabrena, 2019-08-18:
Rejected by LEAN

closed 2019-08-18 by jabrena