ev3dev-lang-java/ev3dev-lang-java #411: Combine rplidar and gps in raspberry pi 3 to obtain data

JJOCEAN1, 2018-03-10:
@jabrena

jabrena, 2018-03-11:
Doubts about your question:

+ What version do you have about `RPLIDAR`, A1 or A2?
+ What is the goal with the final document? Are you going to process in some way?

How to begin with your development?

This project has been designed to offer a Java solution for Lego Mindstorms EV3. In this generation other provides like Dexter Industries or Mindsensors has created some raspberry products which are compatible with LEGO sensors. All products are compatible with EV3Dev a custom Debian distro. I recommend to start your project installing EV3Dev in your Raspberry Pi 3 board:
http://www.ev3dev.org/docs/getting-started/
http://www.ev3dev.org/downloads/
Download the image for Raspberry Pi 2, it is compatible for Raspberry Pi 3.

jabrena, 2018-03-11:
How to begin with your development?

This project has been designed to offer a Java solution for Lego Mindstorms EV3. In this generation other provides like Dexter Industries or Mindsensors has created some raspberry products which are compatible with LEGO sensors. All products are compatible with EV3Dev a custom Debian distro. I recommend to start your project installing EV3Dev in your Raspberry Pi 3 board:
http://www.ev3dev.org/docs/getting-started/
http://www.ev3dev.org/downloads/

# Step 1
Download the image for Raspberry Pi 2, it is compatible for Raspberry Pi 3.

JJOCEAN1, 2018-03-11:
Thanks for your help, first i have an RPLIDAR A1M1-R3, i think about how to create each line in the document, i want (Angle;Distance;latitude;longitude) of each point, it is possible? with that format, separate with (;), the rest of the procces is with this document, but at that moment i only need to create de .txt document

jabrena, 2018-03-11:
Yes, it is possible combine both async sensors.

jabrena, 2018-03-11:
Comment in this issue if you find some problem when you try to install EV3Dev for Raspberry Pi 3.

jabrena, 2018-03-12:
Good morning @JJOCEAN1, 
what OS do you have installed in your laptop? WIN | OSX | Linux

jabrena, 2018-03-12:
Example to use RPLIDAR:

``` java
package examples;

import ev3dev.sensors.slamtec.RPLidarA1;
import ev3dev.sensors.slamtec.model.Scan;
import lombok.extern.slf4j.Slf4j;

public @Slf4j class Demo {

    public static void main(String[] args) throws Exception {

        log.info("Testing RPLidar on a EV3 Brick with Java");
        final String USBPort = "/dev/ttyUSB0";
        final RPLidarA1 lidar = new RPLidarA1(USBPort);
        lidar.init();

        for(int x = 0; x <= 5; x++){

            final Scan scan = lidar.scan();
            log.info("Iteration: {}, Measures: {}", x, scan.getDistances().size());
            scan.getDistances()
                .stream()
                .filter((measure) -> measure.getQuality() > 10)
                .filter((measure) -> (measure.getAngle() >= 345 || measure.getAngle() <= 15))
                .filter((measure) -> measure.getDistance() <= 50)
                .forEach(System.out::println);
        }

        lidar.close();
        log.info("End demo");
        System.exit(0);
    }
}
```

jabrena, 2018-03-12:
In the issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/412

You have the info about GPS

jabrena, 2018-03-15:
Example to use USB GPS:

``` java
package examples;

import ev3dev.sensors.gps.GenericGPS;
import lombok.extern.slf4j.Slf4j;

@Slf4j
public class GenericGPSTest {

	public static void main(String[] args) throws Exception {

		final String port = "/dev/ttyACM0";
		final GenericGPS gps = new GenericGPS(port);
		gps.init();

		//This method block main thread 10 seconds
		timeCounter(10);

		gps.close();
		log.info("Closing connection with the USB GPS Device");

		log.info("LAT: {} {}, LAT: {} {}, ALT: {}", gps.getLatitude(), gps.getLatitudeDirection(), gps.getLongitude(), gps.getLongitudeDirection(), gps.getAltitude());
		log.info("DATE: {}, TIMESTAMP: {}", gps.getDate(), gps.getTimeStamp());
		log.info("NSTAT: {}", gps.getSatellitesTracked());

	}

	private static void timeCounter(final int seconds) throws InterruptedException {
		log.info("Start reading");
		for(int x = 0; x <= seconds; x++){
			log.info("Iteration: {} \n\n", x);
			Thread.sleep(1000);
		}
	}
}
```

jabrena, 2018-03-24:
Hi @JJOCEAN1, any progress?

JJOCEAN1, 2018-04-10:
Hola @jabrena creo que he conseguido instalar el ev3dev en mi raspberry 3, también a través del software Putty creo que he podido establecer conexión, siguiendo los pasos de http://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/ , se supone que ¿estando en la misma red wifi es suficiente?, ¿Cuando escribes algún comando dentro de putty, se tiene que actualizar automáticamente y poder ver lo mismo en la raspberry por hdmi?

jabrena, 2018-04-10:
Hi @JJOCEAN1, 

When you have a `SSH` connection, you could see the result of your commands from `Putty`.
It is not necessary to connect any display to your `Raspberry Pi`.

Did you connect the `RPLIDAR A1` to your Raspberry Pi?

Juan Antonio

jabrena, 2018-04-10:
On internet exist many Java tutorials, maybe it is a good issue to develop in the future. In part, it is a goal of this educational project in a Robotics context.

To steps to do are:

1. Install Java 10 in local. I suppose that you are using Windows.
2. Install Maven and Gradle
3. Install IntelliJ
4. Download the project example: https://github.com/ev3dev-lang-java/template_project_gradle
5. Open the project on IntelliJ

When you have the previous steps completed, we will continue with more steps.

Juan Antonio

JJOCEAN1, 2018-04-10:
ok i think i am ready

JJOCEAN1, 2018-04-11:
Como se ejecuta el codigo de los zip de github y que se ejecute en la raspberry?

jabrena, 2018-04-11:
Hi @JJOCEAN1, I don't undertand when you say zips of github.

Can you explain in detail your doubt?

If you have the template project in your computer. Open the project with IDEA and add the previous examples that I showed.

In order to deploy the example, you will have to upgrade `MANIFEST-INF` with the name of the class that you want to run and you will have to upgrade the IP of the raspberry pi in the file: `config.gradle`.

Using the Gradle task: `deployAndRun` you can deploy the example in the `Rasperry Pi` and run.

did you install any `JVM` in the Raspberry Pi?

Juan Antonio

JJOCEAN1, 2018-04-12:
no he instalado ninguna JVM en la raspberry

jabrena, 2018-04-12:
You sould install a JVM in your board in that case.

```
    apt-key adv --recv-key --keyserver keyserver.ubuntu.com EEA14886
    echo "deb http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main" | sudo tee -a /etc/apt/sources.list
    echo "deb-src http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main" | sudo tee -a /etc/apt/sources.list
    sudo apt-get update
    sudo apt-get install oracle-java8-installer
```

jabrena, 2018-04-15:
Any progress?

jabrena, 2018-04-23:
Latest version of the library:

```
https://jitpack.io/#ev3dev-lang-java/usb-devices/0.2.0-SNAPSHOT
```

JakubVanek, 2018-04-23:
Another way of installing Java (taken from installer):
```sh
    cat >/etc/apt/preferences.d/jdk <<EOF
Package: *
Pin: release a=stable
Pin-Priority: 200
Package: *
Pin: release a=testing
Pin-Priority: 100
EOF
    # workaround for cyclic dependency
    ln -sf "/usr/lib/jvm/java-10-openjdk-armhf/bin/java" "/usr/bin/java"

    # add repo, update
    echo "deb http://ftp.debian.org/debian buster main contrib non-free" | sudo tee "/etc/apt/sources.list.d/jdk.list"
    apt-get update

    # install package
    #  (the symlink above gets discarded, but
    #   it is needed during the installation)
    apt-get install --yes --no-install-recommends "openjdk-10-jre-headless"
```
This process was intended to be run on Debian Stretch, though it may work on Jessie too (just change 'stable' to 'oldstable' in the preferences/jdk part).

jabrena, 2018-04-23:
Moved question to next release

jabrena, 2018-05-19:
Any progress @JJOCEAN1 ?

jabrena, 2018-05-30:
Closed by lack of feedback.

closed 2018-05-30 by jabrena