ev3dev-lang-java/ev3dev-lang-java #412: Support for USB GPS

jabrena, 2018-03-11:
It is necessary to create a support for USB GPS.

[image]

In leJOS, exist a good package to parse all NMEA messages and the pending development is a driver to connect and process the incoming stream.
https://github.com/ev3dev-lang-java/lejos-ev3-code-mirror/tree/master/ev3classes/src/lejos/hardware/gps

JJOCEAN1, 2018-03-11:
I think to buy this gps-usb: https://www.amazon.es/GlobalSat-BU-353-S4-Receptor-SiRF-Star/dp/B008200LHW/ref=sr_1_3?ie=UTF8&qid=1520765197&sr=8-3&keywords=gps+usb
Will it be compatible?

jabrena, 2018-03-11:
Hi @JJOCEAN1,

This evening, I will review more models. Besides, at home I have USB GPS, I will try to know if I am able to connect with it.

Juan Antonio

jabrena, 2018-03-11:
Links of interest:
https://www.java-forums.org/new-java/52375-receiving-gps-data-through-comm-port.html
http://www.oracle.com/webfolder/technetwork/tutorials/obe/java/RaspberryPi_GPS/RaspberryPi_GPS.html
https://github.com/alus-it/GPSreceiver
https://github.com/sardylan/gpsutility
https://github.com/kallaspriit/Java-Serial-NMEA-GPS

jabrena, 2018-03-11:
My USB GPS Device is detected in EV3Dev in the following way:

```
robot@brick1:~$ lsusb
Bus 001 Device 005: ID 2001:3308 D-Link Corp. DWA-121 802.11n Wireless N 150 Pico Adapter [Realtek RTL8188CUS]
Bus 001 Device 004: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 003: ID 1546:01a6 U-Blox AG 
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

```
robot@brick1:~$ ls /dev
autofs           full     log               mmcblk0p1           rfkill  tty    tty16  tty24  tty32  tty40  tty49  tty57  tty8     tty_in2    vcs1   vcsa3
block            fuse     loop-control      mmcblk0p2           rtc     tty0   tty17  tty25  tty33  tty41  tty5   tty58  tty9     tty_in3    vcs2   vcsa4
bus              hidraw0  loop0             mqueue              rtc0    tty1   tty18  tty26  tty34  tty42  tty50  tty59  ttyACM0  tty_in4    vcs3   vcsa5
char             hwrng    loop1             network_latency     serial  tty10  tty19  tty27  tty35  tty43  tty51  tty6   ttyS0    ttyprintk  vcs4   vcsa6
console          i2c-1    loop2             network_throughput  shm     tty11  tty2   tty28  tty36  tty44  tty52  tty60  ttyS1    uinput     vcs5   watchdog
cpu_dma_latency  initctl  loop3             null                snd     tty12  tty20  tty29  tty37  tty45  tty53  tty61  ttyS2    urandom    vcs6   watchdog0
disk             input    mem               ptmx                stderr  tty13  tty21  tty3   tty38  tty46  tty54  tty62  ttySU0   usbmon0    vcsa   zero
fb0              kmem     memory_bandwidth  pts                 stdin   tty14  tty22  tty30  tty39  tty47  tty55  tty63  ttySU1   usbmon1    vcsa1  zram0
fd               kmsg     mmcblk0           random              stdout  tty15  tty23  tty31  tty4   tty48  tty56  tty7   tty_in1  vcs        vcsa2
```

jabrena, 2018-03-12:
Hi @JJOCEAN1,

Tonight, I will cleanup a bit the example but it is running my `POC`:

``` java
java -Djava.library.path=/usr/lib/jni/ -server -jar /home/robot/usb-devices-all-0.1.3-SNAPSHOT.jar
ev3dev#1|SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
ev3dev#1|SLF4J: Defaulting to no-operation (NOP) logger implementation
ev3dev#1|SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
ev3dev#1|Stable Library
ev3dev#1|=========================================
ev3dev#1|Native lib Version = RXTX-2.2pre2
ev3dev#1|Java lib Version   = RXTX-2.1-7
ev3dev#1|WARNING:  RXTX Version mismatch
ev3dev#1|	Jar version = RXTX-2.1-7
ev3dev#1|	native lib Version = RXTX-2.2pre2
ev3dev#1|$GPTXT,01,01,02,u-blox ag - www.u-blox.com*50
ev3dev#1|
ev3dev#1|$GPTXT,01,01,02,HW  UBX-G60xx  00040007 FFF1FFFFp*55
ev3dev#1|
ev3dev#1|$GPTXT,01,01,02,ROM CORE 7.03 (45969) Mar 17 2011 16:18:34*59
ev3dev#1|
ev3dev#1|$GPTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*20
ev3dev#1|
ev3dev#1|$GPTXT,01,01,02,ANTSTATUS=OK*3B
ev3dev#1|
ev3dev#1|$GPRMC,191545.00,A,5200.67674,N,00045.23592,W,0.868,,120318,,,A*6D
ev3dev#1|
ev3dev#1|$GPVTG,,T,,M,0.868,N,1.608,K,A*2A
ev3dev#1|
ev3dev#1|$GPGGA,191545.00,5200.67674,N,00045.23592,W,1,05,1.50,99.9,M,46.5,M,,*71
ev3dev#1|
ev3dev#1|$GPGSA,A,3,31,27,29,26,16,,,,,,,,2.71,1.50,2.26*0B
ev3dev#1|
ev3dev#1|$GPGSV,3,1,12,04,87,170,08,05,14,041,,09,07,332,,16,41,297,18*73
ev3dev#1|
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,18,23,08,298,,25,15,122,09*7E
ev3dev#1|
ev3dev#1|$GPGSV,3,3,12,26,73,284,20,27,14,253,21,29,41,067,11,31,37,201,20*79
ev3dev#1|
ev3dev#1|$GPGLL,5200.67674,N,00045.23592,W,191545.00,A,A*7B
ev3dev#1|
ev3dev#1|$GPRMC,191546.00,A,5200.67727,N,00045.23632,W,0.982,,120318,,,A*65
ev3dev#1|$GPVTG,,T,,M,0.982,N,1.820,K,A*2B
ev3dev#1|$GPGGA,191546.00,5200.67727,N,00045.23632,W,1,06,1.50,100.6,M,46.5,M,,*41
ev3dev#1|$GPGSA,A,3,31,27,29,26,25,16,,,,,,,2.71,1.50,2.26*0C
ev3dev#1|$GPGSV,3,1,12,04,87,170,07,05,14,041,,09,07,332,,16,41,297,20*77
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,19,23,08,298,,25,15,122,10*77
ev3dev#1|$GPGSV,3,3,12,26,73,284,21,27,14,253,23,29,41,067,10,31,37,201,21*7A
ev3dev#1|$GPGLL,5200.67727,N,00045.23632,W,191546.00,A,A*76
ev3dev#1|$GPRMC,191547.00,A,5200.67680,N,00045.23736,W,0.728,,120318,,,A*63
ev3dev#1|$GPVTG,,T,,M,0.728,N,1.349,K,A*21
ev3dev#1|$GPGGA,191547.00,5200.67680,N,00045.23736,W,1,05,1.64,101.9,M,46.5,M,,*43
ev3dev#1|$GPGSA,A,3,31,27,26,25,16,,,,,,,,2.53,1.64,1.92*0C
ev3dev#1|$GPGSV,3,1,12,04,87,170,07,05,14,041,,09,07,332,,16,41,297,22*75
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,20,23,08,298,,25,15,122,11*7C
ev3dev#1|$GPGSV,3,3,12,26,73,284,22,27,14,253,24,29,41,067,09,31,37,201,22*75
ev3dev#1|$GPGLL,5200.67680,N,00045.23736,W,191547.00,A,A*7E
ev3dev#1|$GPRMC,191548.00,A,5200.67697,N,00045.23951,W,1.154,,120318,,,A*69
ev3dev#1|$GPVTG,,T,,M,1.154,N,2.137,K,A*25
ev3dev#1|$GPGGA,191548.00,5200.67697,N,00045.23951,W,1,05,1.64,104.2,M,46.5,M,,*4B
ev3dev#1|$GPGSA,A,3,31,27,26,25,16,,,,,,,,2.53,1.64,1.92*0C
ev3dev#1|$GPGSV,3,1,12,04,87,170,07,05,14,041,,09,07,332,,16,41,297,20*77
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,19,23,08,298,,25,15,122,10*77
ev3dev#1|$GPGSV,3,3,12,26,73,284,22,27,14,253,24,29,41,067,09,31,36,201,22*74
ev3dev#1|$GPGLL,5200.67697,N,00045.23951,W,191548.00,A,A*78
ev3dev#1|$GPRMC,191549.00,V,,,,,,,120318,,,N*75
ev3dev#1|$GPVTG,,,,,,,,,N*30
ev3dev#1|$GPGGA,191549.00,,,,,0,04,20.14,,,,,,*64
ev3dev#1|$GPGSA,A,1,31,27,26,16,,,,,,,,,28.97,20.14,20.81*3C
[... 33 lines cut ...]
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,21,23,08,298,,25,15,122,14*78
ev3dev#1|$GPGSV,3,3,12,26,73,284,22,27,14,253,25,29,41,067,14,31,36,201,22*79
ev3dev#1|$GPGLL,5200.67714,N,00045.24801,W,191553.00,A,A*7B
ev3dev#1|$GPRMC,191554.00,A,5200.67708,N,00045.24861,W,1.628,,120318,,,A*6A
ev3dev#1|$GPVTG,,T,,M,1.628,N,3.015,K,A*29
ev3dev#1|$GPGGA,191554.00,5200.67708,N,00045.24861,W,1,06,1.33,112.6,M,46.5,M,,*46
ev3dev#1|$GPGSA,A,3,31,27,29,26,25,16,,,,,,,2.33,1.33,1.92*03
ev3dev#1|$GPGSV,3,1,12,04,87,170,13,05,14,041,,09,07,332,,16,41,297,20*72
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,21,23,08,298,,25,15,122,14*78
ev3dev#1|$GPGSV,3,3,12,26,73,284,22,27,14,253,25,29,41,067,14,31,36,201,22*79
ev3dev#1|$GPGLL,5200.67708,N,00045.24861,W,191554.00,A,A*77
ev3dev#1|$GPRMC,191555.00,A,5200.67729,N,00045.24926,W,1.377,,120318,,,A*65
ev3dev#1|$GPVTG,,T,,M,1.377,N,2.551,K,A*22
ev3dev#1|$GPGGA,191555.00,5200.67729,N,00045.24926,W,1,06,1.33,112.9,M,46.5,M,,*49
ev3dev#1|$GPGSA,A,3,31,27,29,26,25,16,,,,,,,2.33,1.33,1.92*03
ev3dev#1|$GPGSV,3,1,12,04,87,170,13,05,14,041,,09,07,332,,16,41,297,21*73
ev3dev#1|$GPGSV,3,2,12,20,20,102,,21,58,154,21,23,08,298,,25,15,122,14*78
ev3dev#1|$GPGSV,3,3,12,26,73,284,22,27,14,253,25,29,41,067,14,31,36,201,22*79
ev3dev#1|$GPGLL,5200.67729,N,00045.24926,W,191555.00,A,A*77
```

Later, I will upgrade `lejos-commons` to add the package about GPS from this repo:
https://github.com/ev3dev-lang-java/lejos-ev3-code-mirror/tree/master/ev3classes/src/lejos/hardware/gps

Juan Antonio

jabrena, 2018-03-12:
Upgrade lejos-commons with GPS package:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/414

jabrena, 2018-03-12:
Hi @JJOCEAN1, 

Take a look this Git Branch:
https://github.com/ev3dev-lang-java/usb-devices/tree/feature/%23412

You have an example to connect with a USB GPS Device:

``` java
package examples;

import ev3dev.sensors.gps.GenericGPS;
import lombok.extern.slf4j.Slf4j;

@Slf4j
public class GenericGPSTest {

	public static void main(String[] args) throws Exception {

		final String port = "/dev/ttyACM0";
		final GenericGPS gps = new GenericGPS(port);
		gps.init();

		//This method block main thread 10 seconds
		timeCounter(10);

		gps.close();
		log.info("Closing connection with the USB GPS Device");
	}

	private static void timeCounter(final int seconds) throws InterruptedException {
		log.info("Start reading");
		for(int x= 0; x<=seconds; x++){
			log.info("Iteration: {} \n\n", x);
			Thread.sleep(1000);
		}
	}
}
```

This class at the moment, connect with the device and write the information directly using the method: `serialEvent`:
https://github.com/ev3dev-lang-java/usb-devices/blob/feature/%23412/src/main/java/ev3dev/sensors/gps/GenericGPS.java

``` java
	/**
	 * Handle an event on the serial port. Read the data and print it.
	 */
	@Override
	public synchronized void serialEvent(SerialPortEvent oEvent) {
		if (oEvent.getEventType() == SerialPortEvent.DATA_AVAILABLE) {
			try {
				final String inputLine=input.readLine();

				System.out.println(inputLine);

			} catch (Exception e) {
				log.error(e.getLocalizedMessage());
			}
		}
	}
```

If you like, fork the repo and try to parse the inputline with the help of this classes:
https://github.com/ev3dev-lang-java/lejos-commons/tree/master/src/main/java/lejos/hardware/gps

Later, it is necessary to create some get methods for the fields that you need.

I will verify your merge request.

Cheers

Juan Antonio

jabrena, 2018-03-15:
Hi @JJOCEAN1,

I have upgrade the class with the complete NMEA support.
Today, I will do some integration tests.

jabrena, 2018-03-15:
Using the following example, it is possible to use a USB GPS:

``` java
package examples;

import ev3dev.sensors.gps.GenericGPS;
import lombok.extern.slf4j.Slf4j;

@Slf4j
public class GenericGPSTest {

	public static void main(String[] args) throws Exception {

		final String port = "/dev/ttyACM0";
		final GenericGPS gps = new GenericGPS(port);
		gps.init();

		//This method block main thread 10 seconds
		timeCounter(10);

		gps.close();
		log.info("Closing connection with the USB GPS Device");

		log.info("LAT: {} {}, LAT: {} {}, ALT: {}", gps.getLatitude(), gps.getLatitudeDirection(), gps.getLongitude(), gps.getLongitudeDirection(), gps.getAltitude());
		log.info("DATE: {}, TIMESTAMP: {}", gps.getDate(), gps.getTimeStamp());
		log.info("NSTAT: {}", gps.getSatellitesTracked());

	}

	private static void timeCounter(final int seconds) throws InterruptedException {
		log.info("Start reading");
		for(int x = 0; x <= seconds; x++){
			log.info("Iteration: {} \n\n", x);
			Thread.sleep(1000);
		}
	}
}
```

Log output:

```
ev3dev#11|2018-03-15 18:47:26 [main] INFO  examples.GenericGPSTest - Closing connection with the USB GPS Device
ev3dev#11|2018-03-15 18:47:26 [main] INFO  examples.GenericGPSTest - LAT: 40.741895 N, LAT: -73.989308 W, ALT: 102.8
ev3dev#11|2018-03-15 18:47:26 [main] INFO  examples.GenericGPSTest - DATE: Thu Mar 15 18:47:25 UTC 2018, TIMESTAMP: 184725
ev3dev#11|2018-03-15 18:47:26 [main] INFO  examples.GenericGPSTest - NSTAT: 4
```

Using this website, you could check GPS data in a map
https://www.gps-coordinates.net/

closed 2018-03-15 by jabrena