ev3dev-lang-java/ev3dev-lang-java #413: Project Issue review

jabrena, 2018-03-12:
For this year, it is necessary to review:

- Backlog
- Open pull requests

With that information, it is necessary to reschedule milestones for 2018 in order to put priorities.

jabrena, 2018-03-12:
In the short term, the main goals for the project in 2018 are:

**Continous JDK Support**
http://openjdk.java.net/jeps/322
https://mreinhold.org/blog/forward-faster

it is necessary to build Java 10 for the project:
http://openjdk.java.net/projects/jdk/10/

It is necessary to maintain have a continous process to release custom JRE for the whole community.

It is necessary to automate increase the automation activities to reduce the time used to install the whole environment.

jabrena, 2018-03-12:
Any suggestion? @JakubVanek @jartu

ammerzon, 2018-03-12:
Support for [NXT color sensor 9694](https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/410)?

jabrena, 2018-03-14:
Hi @ammerzon, I will add your idea as part of the roadmap but with low priority. Anyway, I will create a Git branch with a prototype to test in your side. At home, I only found here NXT Sound Sensors. 

For this year, I observe more possibilities with `Arduino Sensors`, `USB Sensors` and `Grove Sensors`. `EV3` and `Raspberry Pi` Boards has USB ports ready to use.

jabrena, 2018-03-31:
Any suggestion before closing this issue to define the goals for this year? @chgandm, @jaakkos, @jeperez, @jweijers, @MartinKvicala, @Mayt38, @DMyTryC, @mominbilal, @nmatra, @Quingu, @submars

JakubVanek, 2018-03-31:
Maybe #446 .

jabrena, 2018-03-31:
Hi @JakubVanek, in what scenario do you like to use Bluetooth?

For Communications we have in the list:

- ROS Comms
- RMI Remoting
- Bluetooth comms ?

JakubVanek, 2018-03-31:
Hmmm, RMI over BT PAN might be enough for some stuff.
I meant some BlueZ wrapper API so that the brick could itself connect to some other device or brick.

jabrena, 2018-03-31:
This package is a huge challenge:
https://github.com/ev3dev-lang-java/lejos-ev3-code-mirror/tree/master/ev3classes/src/lejos/remote/ev3

You are doing a great effor in JVM layer. Maybe others could help in this area. In my case, I have to continue with this library:
https://github.com/ev3dev-lang-java/ros

To provide `SLAM` features to `EV3` Robots with `Java`.
https://www.youtube.com/watch?v=c_rkFlHtAYM&t=90s

jabrena, 2018-03-31:
In my case, I have to continue with this library about ROS to provide `SLAM` features to `EV3` Robots with `Java`. Last year, I could integrate son sensor data on ROS and this year, I only should integrate that data on `Cartographer`.
https://github.com/ev3dev-lang-java/ros

Idea:
https://www.youtube.com/watch?v=c_rkFlHtAYM

JakubVanek, 2018-04-01:
If it gets accepted, I vote for #453 too.

jabrena, 2018-04-02:
The idea to create a Spike to research if it is possible to move LeJOS runtime to EV3Dev world could be nice. When I create the project, I discovered that the whole project was fixed in a OS environment with few capabilities (No USB support, No package manager, No complete OS). You could review multiple comments on LeJOS forum in last 4 years:
https://lejos.sourceforge.io/forum/

So, in terms of backward compatibility, it could be a great opportunity but... It requires a good sum of time to run everything (6 months) so, if you like we could Schedule the execution of the Spike in the proper way to combine effort. Besides for that issue, when we execute, I would like to take in account `Andy` because indeed, he is maintaining the `LeJOS forum` and we need to coordinate with him.

This project provide a solid integration with EV3Dev (Support for 3 platforms: EV3 Brick, BrickPi & PiStorms + USB Support) but of course original LeJOS, has a strong library for EV3.

So my answer is yes, but we should schedule:

**v0.8.0:** April 2018

+ v0.8.0 Jessie Support + Jessie Installer + Oracle JRE 8
+ v2.1.0 Stretch Support + Stretch Installer + OpenJDK 9/10
+ v0.2.0 USB Devices (Adding GPS Support)

**v0.9.0:** June 2018

+ v2.2.0 Stretch Support + Stretch Installer + OpenJDK 10
+ Initial Spike to move LeJOS runtime to https://github.com/ev3dev/lms2012-compat
+ Continue with ROS development
+ Early build for OpenJDK 11

**v0.10.0:** September 2018

+ v2.3.0 Internal code refactoring
+ Initial Spike to move LeJOS runtime to https://github.com/ev3dev/lms2012-compat
+ Continue with ROS development
+ Initial Spike to add Remote Support

But remember that it is `Java Robotics project` so, we have to create balance in terms of efforts (Personal spare time) to:

+ Maintain the solution
+ Develop new robotics features (ROS, Remote, Vision)
+ Platform research (Raspberry Pi 3 Model B+, Raspberry Pi 4, Depth Cameras, LIDAR, Grove Pi, PivotPi, Arduino, EV3Dev upgrades)
+ Technical Debt (Code refactoring, Improve Installer syntax, etc...)

If you notice, LEGO offer few sensors, but with a EV3 Brick, the most powerful sensors has no a LEGO nature: LIDAR, IMU, Webcam, etc... It is a pity that Raspberry Pi 3 doesn´t have support for Depth cameras, every serious models uses USB C, so in 2019, BrickPi3 is the best candidate to use them.

Juan Antonio

JakubVanek, 2018-04-02:
Yes, I agree.

gloomyandy, 2018-04-02:
Hi,
I'd just like to make sure that you all understand exactly what the
relationship is between leJOS on the EV3 and the original LEGO
implementation....

leJOS does not run on top of the LEGO VM nor does it use any of the user
space implementation. What it does do is use a number of highly modified
versions of the LEGO kernel modules to provide low level support for i2c,
sound, LEGO serial sensors and access to I/O ports. It also uses direct
access to the (now rather old) Linux implementations of Bluetooth, USB and
networking. All of the middle and higher level code is written in Java and
tends to be rather dependant on the low level functionality provided by the
various kernel implementations. So I'm not sure that lms2014-compact
actually gets you anything in terms of moving the EV3 version of leJOS to
ev3dev.

On a general note I've never been in favour of a leJOS port to ev3dev.
Ev3dev uses a very different driver model to leJOS (in places better, in
places worse) and trying to make the leJOS model work on ev3dev has never
made much sense to me. The only really good argument is that it would allow
leJOS programs to run on ev3dev. But I suspect that the large differences
in the low level driver model (particularly for things like motor control)
would mean that in reality very few complex programs would simply run. Far
better to simply provide a pure ev3dev interface for Java programs that
follows the same device model as other languages on ev3dev. There are
really very few parts of leJOS that would bring anything new to ev3dev
(without ditching some of the very low level parts of ev3dev, which is
unlikely to happen). Perhaps the Chassis implementation (though I'm not
sure how well that would work using the ev3dev motor implementation) and
maybe the new sensor framework. Most of the other "high level robotics"
stuff is really not worth porting, it tends to have many limitations
inherited from the NXT that make it hard to fix/improve without a rewrite
and there are better solutions available with things like ROS. Things like
the remote capabilities of leJOS have always had problems (as do most
remote APIs), particularly when trying to use them with multiple threads,
they are fundamentally broken. I've always argued that the correct way to
do "remote control" or "daisy chaining" is to create a task specific high
level remote API and use that not try to make use of low level generic
interfaces, which inevitably suffer timing/bandwidth/portability issues.

leJOS was always about doing as much as possible in Java and providing a
system that allowed you to simply write Java robotics programs in a
professional level IDE and run them on LEGO hardware. But that is not
really what ev3dev is about, it is more about allowing the use of
components written using different languages run on different types of
hardware using whatever development tools the user wishes to use. These
different aims do not fit well together. It makes far more sense to provide
a good robust Java implementation of the existing ev3dev device API
sticking as closely as possible to the model used in other languages on the
EV3. That way the Java API can evolve along with ev3dev as it moves to new
and different hardware platforms.

I really think it is time to move on from leJOS, it served its purpose
well, but it has had its day.

Good luck and have fun!

Andy

jabrena, 2018-04-02:
Hi @gloomyandy,

many thanks for your comments, the reply was a bit extended so I will try to reply by parts:

**What is your opinion to run LeJOS runtime with lms2014-compact?**

> Hi,
> I'd just like to make sure that you all understand exactly what the
> relationship is between leJOS on the EV3 and the original LEGO
> implementation....
>
> leJOS does not run on top of the LEGO VM [...]

I think that the Spike is interesting. Maybe @JakubVanek or others in the community discover some way to run everything with modern version of Linux, etc... It is a good Engineering exercise. At the end this kind of projects has a educational goal so why not try. Other questions is the solution is valid in terms of many different factors. In my case, I don´t have that `C` skills. :(

jabrena, 2018-04-02:
**It is possible to run LeJOS interfaces on EV3Dev?**

> On a general note I've never been in favour of a leJOS port to ev3dev.
> Ev3dev uses a very different driver model to leJOS (in places better, in
> places worse) and trying to make the leJOS model work [...]

Well, you should try latest versions of the libraries. The results are fine for me. To test the library, I used many examples from Internet in order to understand what is the kind of syntax that people uses in their programs for **NXT/EV3** and using this library it is possible to execute.

It is true that it is necessary to change some imports for some classes because I consider that some classes not related with the implementation goes in `lejos-commons`:
https://github.com/ev3dev-lang-java/lejos-commons

but the concrete implementation goes here:
https://github.com/ev3dev-lang-java/ev3dev-lang-java

Anyway, I know that you are not super fan of this library, but I it is necessary to continue development in some way... :)

One point that it is not possible to implement in the same way is the motor synchronisation. In latest version of LeJOS:
https://sourceforge.net/projects/ev3.lejos.p/files/0.9.1-beta/

Motor synchronisation goes in the runtime but in EV3Dev, that action is delegated to the libraries.

At the moment, for local navigation, I have isolated some parts in this library:
https://github.com/ev3dev-lang-java/lejos-navigation

> Perhaps the Chassis implementation (though I'm not
> sure how well that would work using the ev3dev motor implementation) and
> maybe the new sensor framework. 

But chassis is pending. I think that I need to test in detail more current Pilot implementation before add Chassis. It is true that the new way to navigate is more elegant, but I need tot test in detail before releasing.

jabrena, 2018-04-02:
I**s it possible to continue using "high level robotics" concepts from LeJOS?**

> Most of the other "high level robotics"
> stuff is really not worth porting, it tends to have many limitations
> inherited from the NXT that make it hard to fix/improve without a rewrite
> and there are better solutions available with things like ROS. 

For me, the concepts from lejos-commons continue be valid for me. In EV3 Space, the most trending development is the python library for ev3 but the level of abstraction reached with LeJOS is not compared with latest version.

It is true that they have some example using Daisy chain with RPyC but remote package from lejos is more integral in my opinion but you have more expertise about the limits of that package. But it is a good opportunity to migrate.
https://github.com/ev3dev/ev3dev/issues/1031

**Is ROS a holy grail?**

ROS is not the holy grail. In 2010 Lawrie & me implemented a ROS solution for NXT and this is the same idea for EV3. The main difference between both version is because now, it is possible to run the ROS link on the Brick, not in the PC, this is the innovation. If you observe Turtlebot3 project on Github requires some kind of development:
https://github.com/ROBOTIS-GIT/turtlebot3

So, using Turtlebot 3 or using EV3, it is necessary to develop and do many tests.

jabrena, 2018-04-02:
**Is EV3Dev a poliglot environment?**

> leJOS was always about doing as much as possible in Java and providing a
> system that allowed you to simply write Java robotics programs in a
> professional level IDE and run them on LEGO hardware. But [...]

It is true, EV3Dev is a bit poliglot but tell me from a Java perspective, does exist another alternative with active developers coding new stuff? The answer is no. So EV3Dev is a good opportunity to continue developing Java software for robots. It is not perfect, I am not sure, but it allows run new bytecodes.

I agree that in the process to develop this library and others interesting like `USB Devices`:
https://github.com/ev3dev-lang-java/usb-devices

In general, _it is a pity that you are sad with the usage of LeJOS in the future_, from my point of view, it is valid the Robotics concepts. It shows many concepts about robotics with Java and that concepts are universals so, for me it has the enough value to continue maintaining that interfaces now, in a **poliglot** environment. 

Modern consultants say: `Evolutionary architectures`
https://www.thoughtworks.com/books/building-evolutionary-architectures

I will call you to discuss some aspects to to improve the API in next months.

Cheers

Juan Antonio

gloomyandy, 2018-04-02:
Again I'm not sure you understand how limited and buggy the current
implementation is. For instance the movement model only supports a
move/stop/turn model it does not work with continuous motion, nor is it
easy to use the model to do things like fuse different sensor data
together. It has never been updated to work well with the sensor framework
and has many limitations from the old NXT system that should no longer
apply. As to the remote API well just try using it with almost anything
other than a trivial program. Any attempt to use multiple threads will fail
in unpredictable and frustrating ways. There are also large parts of the
API which are either not implemented or which do not work when in a remote
context.

As to ROS well a lot has change since 2010! Is it the holy grail of course
not, but for doing anything like sensor fusion, SLAM, visualisation tools
and the rest it is way beyond what is supported by leJOS. Other advantages
is that it has a well define mechanism for splitting functionality between
the robot and say a PC, for leJOS you have to do all of that work yourself.
The one major drawback is the complexity.

jabrena, 2018-04-02:
Hi @gloomyandy,

The points that you say about `acceleration` and `velocity` are true, that aspects are limitation between different implementations but to run a Pilot is Ok:

**Example:**
https://github.com/ev3dev-lang-java/examples/blob/master/ev3dev-lang-java/src/main/java/lejos/navigation/navigator/NavigatorTest2.java

``` java
package lejos.navigation.navigator;

import ev3dev.sensors.Battery;
import lejos.navigation.pilot.PilotConfig;
import lejos.robotics.localization.OdometryPoseProvider;
import lejos.robotics.navigation.DifferentialPilot;
import lejos.robotics.navigation.Navigator;

public class NavigatorTest2 {

    public static void main(final String[] args) {

        PilotConfig pilotConf = new PilotConfig();
        final DifferentialPilot pilot = pilotConf.getPilot();

        OdometryPoseProvider pp = new OdometryPoseProvider(pilot);
        Navigator navigator = new Navigator(pilot, pp);

        navigator.goTo(0,0);
        navigator.goTo(100,0);
        navigator.goTo(50,50);
        navigator.goTo(100,-50);
        navigator.goTo(0,0);
        navigator.followPath();
        navigator.waitForStop();

        System.out.println(pp.getPose());

        System.out.println("Voltage: " + Battery.getInstance().getVoltage());

    }

}
```

I didn´t see many repos on Github using the concepts that you say. It is a possible weakness, maybe yes, but the general usage of leJOS doesn´t include that concepts. Please, search on github to review how many repos use that methods with this projects. I accept the point but it is not determinant.

In relation to I2C, totally agree with you, but EV3Dev provides others ways to communicate with the sensors. One example is NXTCamV5:

Implementation to EV3Dev integration:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/src/main/java/ev3dev/sensors/mindsensors/NXTCamV5.java

Implementation to LeJOS:
https://github.com/ev3dev-lang-java/lejos-support/blob/develop/src/main/java/lejos/hardware/device/NXTCamV5.java

The first one uses EV3Dev /sys/class interface and leJOS uses i2c integration. Take a look the EV3Dev sensor support is massive:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html

jabrena, 2018-04-02:
If you like @gloomyandy, when I finish this iteration, we could review in team ROS implementation and add all suggestions that we discuss together. I appreciate your technical comments. At the end, the idea is to have a SLAM solution described in your article :p 
https://lejosnews.wordpress.com/2017/07/15/slam/

Juan Antonio

gloomyandy, 2018-04-02:
> master/ev3dev-lang-java/src/main/java/lejos/navigation/
> navigator/NavigatorTest2.java
> blob/develop/src/main/java/ev3dev/sensors/mindsensors/NXTCamV5.java
> develop/src/main/java/lejos/hardware/device/NXTCamV5.java
> en/ev3dev-jessie/sensors.html

Your point about the coverage in ev3dev of sensors is exactly the issue I'm
trying to raise here. Rather than trying to make the leJOS API work on
ev3dev, why not simply provide a Java API that mirrors that used by other
ev3dev languages (use the API generation tool that ev3dev has and you can
easily track changes). That way you will get full access to all of the
ev3dev sensor and device support and will not have to spend time writing a
large number of classes to bridge between the leJOS API and ev3dev. Provide
a Java implementation that makes the most of ev3dev rather than looking
back at the past.

If there are some classes (like the pilots) which you think are useful then
by all means take them and convert them to use this ev3dev Java API, though
you may well be better spending your time providing a better
pilot/navigator system to be honest.

jabrena, 2018-04-02:
Because why invent a new API if Java people feel comfortable with leJOS API?
Why reinvent again the same concepts?

One alternative is start from scratch with the design and another alternative is the implementation of the LeJOS interfaces from: 
https://github.com/ev3dev-lang-java/lejos-commons

And the second one was my choice and now is a bit stable. :)

It is amazing when you respect the interfaces and you use this library:
https://github.com/ev3dev-lang-java/lejos-navigation

and the Differential pilot run again :)

This is the reason to not start from the beginning, now we have:

- Modern OS and upgraded: Debian Stretch
- Docker environments to cross-compile software: OpenJDK 9 and I wish OpenJDK 10
- Libraries to interact with LEGO and non LEGO sensors
- Libraries to provide local navigation (Not perfect with classic Differential pilot)

But anyway, if current implementation don´t like in some aspect from last release (November of 2015), I would like to add the changes in this project. 

Juan Antonio

JakubVanek, 2018-04-02:
Hi all,

thanks for all the comments!

I haven't thought about whether the porting of leJOS to ev3dev would be a future-proof solution. I was thinking about smoothly bridging these two ecosystems, leJOS with old system underneath on one side, ev3dev with Debian on the other side. You're probably right that it would be cleaner to write a pure ev3dev-style API. But the leJOS API seemed also quite good to me.

Regarding the leJOS runtime: I'm aware that leJOS isn't running on the lms2012 VM. What I was linking to was rather a consumer of the compatibility layer (https://github.com/ev3dev/lego-linux-drivers/tree/ev3dev-stretch/lms2012), sorry for that. It is a few of the original LEGO kernel drivers (not the ev3dev-style ones) ported to ev3dev. Now I'm realizing that the compatibility layer might stiffen development with non-LEGO peripherals and non-EV3 bricks, something I also haven't thought about. I also didn't know that leJOS uses heavily modified versions of the stock drivers. Then it would make sense why `lms_dcm` code wasn't working on these ported drivers.

Regarding the leJOS bugs: I wasn't aware of those issues in the codebase. If it's like this, it might be better to start from scratch, building maybe a leJOS-like API, or to port only the necessary classes. Well, basically what you and @jabrena are proposing and what ev3dev-lang-java is today.

> Provide a Java implementation that makes the most of ev3dev rather than looking back at the past. 

That's also an interesting viewpoint. I agree with that on one hand, however the idea that an API similar to the one on NXT would run even on ev3dev is also appealing.

>  I really think it is time to move on from leJOS, it served its purpose well, but it has had its day.

I'm a bit sad to hear that. I was also hoping that leJOS development would continue someday. Though I might just be a bit nostalgic, leJOS was my first experience with Java and robots.

Anyway, my past doesn't matter. Maybe we can create something greater than leJOS.

Regards,

Jakub

jabrena, 2018-04-02:
So, we will continue with the iteration v0.8.0 and and later, in the next release, we will review again the issue: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/453

@JakubVanek, Do you know if in some way the original motor model could run Debian Stretch? Maybe running that module we could have the best of both worlds:

+ EV3Dev Sensor support: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html
+ LeJOS Motor model to run Complex navigation system with `acceleration` and `velocity` concepts included.

Meanwhile, our DifferentialPilot will continue running on the robots! :)

In next iteration, we could discuss with @gloomyandy how to improve the LeJOS API from his perspective.

Cheers

Juan Antonio

juan-medina, 2018-04-03:
I going to throw my two cents in this discussion, do not take it to seriously since I'm not an expert neither in robots, neither in the platforms that they are run on.

Considering this:

- Lejos is probably well design but is quite p;d. the way to work with it is like java code develop 10 years ago.
- Attracting Lejos only developers shouldn't be a goal for the project.
- ev3dev already define an API that is already available in several languages
- Attracting ev3dev developer, regardless of what technology they use currently, should be a goal for the project.
- Java has evolve a loot modern API are really design very different, fluent APIs is a clear example, like many apache projects have already implemented. for example: https://hc.apache.org/httpcomponents-client-ga/tutorial/html/fluent.html, or even DSL : https://docs.spring.io/spring-integration/reference/html/java-dsl.html

My view, that could be totally wrong should be, to have 3 projects.

- ev3dev-lang-java-core : Designed following the ev3dev API specification, so it will make very easy to attract other developers that already use that API in other language, for example in python. They could easy get up to speed with the java implementation, and should be update when either new change on the API or new hardware/OS support is added.
- ev3dev-lang-java-fluent (or ev3dev-lang-java-edge or something else) : Provides high level API use modern patterns, fluents APIs/DSL, infinite streams, event driven, etc. Is our candy to attract to new developers, they could easy use us if they know ev3dev-lang API but we offer new shiny things. (It uses eve3dev-lang-java-core)
- ev3dev-langa-java-Lejos : A wrapper around Lejos classes to make easy to port existing Lejo java projects, this project should at the very end disappear.  (It uses eve3dev-lang-java-core)

This is a way to separate the concerns, and attract developers.

If something is working in adding support to some new hardware or OS other could be either update the stream implementation or the lejos versions.

jabrena, 2018-04-03:
Many thanks @juan-medina by the comment from a Software engineering perspective. The analysis that you commented put the effort in one specific library and not in the whole project, I will explain the whole project to clarify:

- Modern OS and upgraded: Debian Jessie/Stretch OK (Thanks @dlech)
- Docker environments to cross-compile software: OpenJDK 9 / 10 <- @JakubVanek is doing a great effort in this area.
- Library to interact with LEGO sensors and third party companies: <- your comments.
- Library to interact with USB Sensors: <- Stable and It is required to get Sensor data for SLAM with ROS
- Libraries to provide local navigation: Stable but without a support of Chassis and the concepts that @gloomyandy say about `acceleration` and `velocity`
- ROS Integration for SLAM: In progress

So, the areas where should put the main efforts are:

- OpenJDK 10/11
- ROS Integration
- Remote support

Later, we could discuss how to evolve some APIs and some Gaps. 

Adding efforts to create more libraries for the same stuff, doesn´t generate innovation.
When we have more Robotics features, we could improve some parts of it.

Well, I continue with v0.8.0 issues

Juan Antonio

gloomyandy, 2018-04-03:
>    httpcomponents-client-ga/tutorial/html/fluent.html
>    java-dsl.html

I think Juan Medina's comments make a lot of sense, and to a large degree
reflect my own thoughts. I'm going to make one final attempt to contribute
to things...

As a software engineer of 30+ years experience (much of that with Sun
Microsystems and Oracle) the thought of creating a Java system for ev3dev
that does not publish the same API as that used by other languages on the
platform seems strange to me. This is especially the case given that ev3dev
provides a tool that will automatically generate the code,  support it and
importantly update it. By not going this route you will be faced with
constantly having to update things to track developments in the platform
and will often be lacking support for new hardware etc. As someone that has
spent a lot of time working and managing  projects that end up in this
loop, it is not a fun way to spend your time! In addition it simply does
not make sense to me to sacrifice the ability to easily share documentation
and knowledge gained from the other programming environments.

To lose all of the above to provide compatibility with leJOS which as has
been mentioned is an arguably outdated and certainly heavily compromised
(because of limitations of the NXT) system does not seem a good compromise
to me. I see very little evidence of a large user base wishing to move
existing code from the existing leJOS platform to ev3dev and in reality
there are very few substantive leJOS projects out there. Most are small
projects (often school/collage based that the authors have long moved on
from) that would arguably benefit from being rewritten if such a move was
thought worthwhile. I think the usefulness of many of the leJOS classes is
being over-estimated and the difficulty of porting the ones that do have
value to an ev3dev based API is really not that great.

At the end of the day it is up to the people who will work on this code to
do things that interest them, after all it is their own time they are
investing. I've spent a lot of my own time over the last few years using
leJOS as a tool to explore robotics and in the process have contributed to
that project. Feel free to ignore my comments, but you did ask for them. As
I've already said I think that leJOS as a project has had its day. If you
wish to continue using Java on future robotics platforms then I really
think it is time to start again with a fresh look at the problem. If I was
working on this project today (and I have no real interest in doing so,
these days I'd rather ride my bike!), I would be doing as little work as
possible on the basic API classes (generate them with the tools available),
these have already been defined by the platform. Instead I'd be wanting to
improve things like the motion models, the support for modern sensors, the
ability to statistically process results, fuse information and in
particular looking at the higher level problems of SLAM/Robotic
Vision/Control systems etc. Basically making it possible for people to
explore the interesting robotics problems that are out there with low cost
or "toy" hardware.

Whatever you decide to do with your time, have fun and good luck!

Andy

jabrena, 2018-04-03:
@gloomyandy, the issue is about how to evolve this project, not about other stuff... :)

Take a look this Hardware:
http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143452239825

Maybe it could be useful for the Computational problems that you want to solve.

In my case, I have enough with this hardware environment including BrickPi boards.

Cheers

Juan Antonio

jabrena, 2018-04-14:
For the next 6 months, this is a possible roadmap:

**v0.8.0: April 2018**

- Build for OpenJDK 9
- Build for OpenJDK 10
- v0.8.0 Jessie Support + Jessie Installer + Oracle JRE 8
- v2.1.0 Stretch Support + Stretch Installer + OpenJDK 9/10
- v0.2.0 USB Devices (Adding GPS Support)

**v2.2.0: July 2018**

- v2.2.0 Stretch Support + Stretch Installer + OpenJDK 10 + BrickPi3 support
- Release ROS library v0.1.0
- Early build for OpenJDK 11
- Migration openjdk-ev3 to adoptopenjdk CI

**v2.3.0: October 2018**

v2.3.0 Internal code refactoring
- Release ROS library v0.2.0
- Initial Spike to add Remote Support

closed 2018-04-14 by jabrena

JakubVanek, 2018-04-15:
Hmmm, would it be possible to release jigsaw sooner, even in a non-refactored, somewhat dumb (single module) form? I'm curious about the performance boosts offered by AppCDS and eventually preloading.

Also, OpenJDK 11 won't be available until September (early release will be in August), https://github.com/ev3dev-lang-java/openjdk-ev3/issues/21#issuecomment-380823294.

jabrena, 2018-04-15:
Yes, in next iteration, we could try Jigsaw as a first iteration. :)