ev3dev-lang-java/ev3dev-lang-java #416: Release OpenJDK 10 JRE

jabrena, 2018-03-13:
It is necessary to release OpenJDK 10:
https://github.com/ev3dev-lang-java/openjdk-ev3/tree/openjdk-10

jabrena, 2018-03-25:
Result of the first compilation:

```
Juans-MBP:build jabrena$ sudo docker run --rm -it -v /Users/jabrena/Desktop/ ev3dev-lang-java:jdk-build
Password:
compiler@bc57e0902df3:/opt/jdkcross$ ./fetch.sh 
abort: Permission denied: '/build/jdk'
compiler@bc57e0902df3:/opt/jdkcross$ sudo chown compiler:compiler /build/jdk
chown: cannot access '/build/jdk': No such file or directory
compiler@bc57e0902df3:/opt/jdkcross$ exit
exit
Juans-MBP:build jabrena$ sudo docker run --rm -it -v /Users/jabrena/Desktop:/build ev3dev-lang-java:jdk-build
compiler@bb814cf26cdf:/opt/jdkcross$ ./fetch.sh 
requesting all changes
adding changesets
adding manifests                                                                                                                                                   
adding file changes                                                                                                                                                
added 2815 changesets with 4779 changes to 489 files                                                                                                               
updating to branch default
341 files updated, 0 files merged, 0 files removed, 0 files unresolved                                                                                             
# Repositories:  corba jaxp jaxws langtools jdk hotspot nashorn 
                corba:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/corba corba
                 jaxp:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/jaxp jaxp
                 jaxp:   requesting all changes
                corba:   requesting all changes
                 jaxp:   adding changesets
                corba:   adding changesets
                corba:   adding manifests
                 jaxp:   adding manifests
                corba:   adding file changes
                 jaxp:   adding file changes
                corba:   added 925 changesets with 5504 changes to 2597 files
                corba:   updating to branch default
                corba:   1201 files updated, 0 files merged, 0 files removed, 0 files unresolved
                jaxws:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/jaxws jaxws
                jaxws:   requesting all changes
                jaxws:   adding changesets
                jaxws:   adding manifests
                jaxws:   adding file changes
                 jaxp:   added 1217 changesets with 15333 changes to 8488 files
                 jaxp:   updating to branch default
                 jaxp:   3343 files updated, 0 files merged, 0 files removed, 0 files unresolved
            langtools:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/langtools langtools
            langtools:   requesting all changes
            langtools:   adding changesets
            langtools:   adding manifests
                jaxws:   transaction abort!
            langtools:   adding file changes
                jaxws:   rollback completed
                jaxws:   abort: stream ended unexpectedly (got 33036 bytes, expected 50056)
                  jdk:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/jdk jdk
                  jdk:   requesting all changes
                  jdk:   adding changesets
            langtools:   transaction abort!
                  jdk:   adding manifests
            langtools:   rollback completed
            langtools:   abort: stream ended unexpectedly (got 69131 bytes, expected 295536)
              hotspot:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/hotspot hotspot
              hotspot:   requesting all changes
              hotspot:   adding changesets
              hotspot:   adding manifests
                  jdk:   adding file changes
              hotspot:   adding file changes
              hotspot:   added 13550 changesets with 85505 changes to 16280 files
              hotspot:   updating to branch default
              hotspot:   9230 files updated, 0 files merged, 0 files removed, 0 files unresolved
              nashorn:   hg clone http://hg.openjdk.java.net/jdk10/jdk10/nashorn nashorn
              nashorn:   requesting all changes
              nashorn:   adding changesets
              nashorn:   adding manifests
[... 186 lines cut ...]
checking for jtreg... no
checking for jtreg test harness... no, not found
checking for jib... no
checking if @file is supported by gcc... yes
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking stdio.h usability... yes
checking stdio.h presence... yes
checking for stdio.h... yes
checking size of int *... 4
checking for target address size... 32 bits
checking whether byte ordering is bigendian... no
checking for ABI profle... arm-ev3
configure: error: Invalid ABI profile: "arm-ev3"
configure exiting with result code 1
compiler@bb814cf26cdf:/opt/jdkcross$ 
```

jabrena, 2018-03-25:
New compilation:

```
sudo docker run --rm -it -v /Users/jabrena/Desktop/openjdk10:/build ev3dev-lang-java:jdk10-build
```

JakubVanek, 2018-03-25:
https://github.com/ev3dev-lang-java/openjdk-ev3/releases/tag/v0.4.2 for Stretch

jabrena, 2018-03-25:
Meanwhile I finish the compilation, I will try to run a HelloWorld example on the EV3 Brick with Debian Stretch for EV3Dev

jabrena, 2018-03-25:
```
sudo apt-get install p7zip-full
7z x jre-ev3.zip
robot@ev3dev:~$ ls
jre-ev3  jre-ev3.zip
robot@ev3dev:~$ cd jre-ev3
robot@ev3dev:~/jre-ev3$ ls
bin  conf  legal  lib  release
robot@ev3dev:~/jre-ev3$ cd bin/
robot@ev3dev:~/jre-ev3/bin$ ls
java  javac  jdb  jrunscript  jshell  keytool  rmid  rmiregistry  serialver
robot@ev3dev:~/jre-ev3/bin$ ./java -version
``` 

@JakubVanek , take a look this message:

``` 
robot@ev3dev:~/jre-ev3/bin$ ./java -version
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xb6b28af8, pid=2217, tid=2218
#
# JRE version:  (10.0+46) (build )
# Java VM: OpenJDK Client VM (10+46, mixed mode, sharing, serial gc, linux-)
# Problematic frame:
# V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/robot/jre-ev3/bin/hs_err_pid2217.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

[error occurred during error reporting (), id 0x4]
``

jabrena, 2018-03-25:
Logs:

``` 
robot@ev3dev:~/jre-ev3/bin$ cat hs_err_pid2217.log 
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xb6b28af8, pid=2217, tid=2218
#
# JRE version:  (10.0+46) (build )
# Java VM: OpenJDK Client VM (10+46, mixed mode, sharing, serial gc, linux-)
# Problematic frame:
# V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  S U M M A R Y ------------

Command Line: 

Host: ARM, 1 cores, 56M, ev3dev-stretch
Time: Sun Mar 25 21:40:25 2018 UTC elapsed time: 0 seconds (0d 0h 0m 0s)

---------------  T H R E A D  ---------------

Current thread is native thread

Stack: [0xb64fa000,0xb654a000],  sp=0xb6548c74,  free space=315k
Native frames: (J=compiled Java code, A=aot compiled Java code, j=interpreted, Vv=VM code, C=native code)
V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
C  0xb6548c7c
V  [libjvm.so+0x5f6aec]  Threads::create_vm(JavaVMInitArgs*, bool*)+0x1cc
V  [libjvm.so+0x3a2bc8]  JNI_CreateJavaVM+0x78

siginfo: si_signo: 4 (SIGILL), si_code: 1 (ILL_ILLOPC), si_addr: 0xb6b28af8

Registers:
  r0  = 0xb6308750
  r1  = 0x00001de8
  r2  = 0x00000000
  r3  = 0x00000000
  r4  = 0xb6548cb0
  r5  = 0x00000000
  r6  = 0xb6d02000
  r7  = 0xb6548cf0
  r8  = 0xb6d0d088
  r9  = 0xb6548dd3
  r10 = 0x00000000
  fp  = 0xb6548c74
  r12 = 0x00000004
  sp  = 0xb6548c74
  lr  = 0xb695b858
  pc  = 0xb6b28af8
  cpsr = 0x80000010

Top of Stack: (sp=0xb6548c74)
0xb6548c74:   b6548c7c b6548dbc b6bfaaec 00000000
0xb6548c84:   00000000 00000000 00000000 00000000
0xb6548c94:   00000000 00000000 00000000 00000000
0xb6548ca4:   00000000 b6d17dbc b6d21aec 00000001
0xb6548cb4:   00000000 5f1af775 00000162 00000000
0xb6548cc4:   00000000 00000000 00000000 b6f33738
0xb6548cd4:   b6548d48 b6d3a8d8 000003dc b6548d4c
0xb6548ce4:   b6f0e788 00000001 b6d3fa14 00000100 

Instructions: (pc=0xb6b28af8)
[... 163 lines cut ...]

CPU:total 1 (initial active 1) 

/proc/cpuinfo:
processor	: 0
model name	: ARM926EJ-S rev 5 (v5l)
BogoMIPS	: 148.88
Features	: swp half thumb fastmult edsp java 
CPU implementer	: 0x41
CPU architecture: 5TEJ
CPU variant	: 0x0
CPU part	: 0x926
CPU revision	: 5

Hardware	: LEGO MINDSTORMS EV3
Revision	: 0000
Serial		: 0000000000000000

Memory: 4k page, physical 57408k(1444k free), swap 98300k(94380k free)

vm_info: OpenJDK Client VM (10+46) for linux--sflt JRE (10+46), built on Mar 25 2018 20:16:11 by "" with gcc 6.3.0 20170516

END.
robot@ev3dev:~/jre-ev3/bin$ 
```

JakubVanek, 2018-03-27:
Could you try running Java with the `-Xint` flag?

jabrena, 2018-03-27:
**Test 2:**

``` 
robot@ev3dev:~/jre-ev3-openjdk10/bin$ ./java -Xint HelloWorld
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xb6bacaf8, pid=9879, tid=9880
#
# JRE version:  (10.0+46) (build )
# Java VM: OpenJDK Client VM (10+46, interpreted mode, sharing, serial gc, linux-)
# Problematic frame:
# V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/robot/jre-ev3-openjdk10/bin/hs_err_pid9879.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

[error occurred during error reporting (), id 0x4]

Aborted
robot@ev3dev:~/jre-ev3-openjdk10/bin$ 
```

jabrena, 2018-03-27:
```
robot@ev3dev:~/jre-ev3-openjdk10/bin$ ./java -Xint HelloWorld
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xb6b8daf8, pid=9890, tid=9891
#
# JRE version:  (10.0+46) (build )
# Java VM: OpenJDK Client VM (10+46, interpreted mode, sharing, serial gc, linux-)
# Problematic frame:
# V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
#
# Core dump will be written. Default location: /home/robot/jre-ev3-openjdk10/bin/core
#
# An error report file with more information is saved as:
# /home/robot/jre-ev3-openjdk10/bin/hs_err_pid9890.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

[error occurred during error reporting (), id 0x4]

Aborted (core dumped)
robot@ev3dev:~/jre-ev3-openjdk10/bin$ cat hs_err_pid9890.log 
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xb6b8daf8, pid=9890, tid=9891
#
# JRE version:  (10.0+46) (build )
# Java VM: OpenJDK Client VM (10+46, interpreted mode, sharing, serial gc, linux-)
# Problematic frame:
# V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
#
# Core dump will be written. Default location: /home/robot/jre-ev3-openjdk10/bin/core
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  S U M M A R Y ------------

Command Line: -Xint HelloWorld

Host: ARM, 1 cores, 56M, ev3dev-stretch
Time: Tue Mar 27 17:59:15 2018 UTC elapsed time: 0 seconds (0d 0h 0m 0s)

---------------  T H R E A D  ---------------

Current thread is native thread

Stack: [0xb655f000,0xb65af000],  sp=0xb65adc74,  free space=315k
Native frames: (J=compiled Java code, A=aot compiled Java code, j=interpreted, Vv=VM code, C=native code)
V  [libjvm.so+0x524af8]  perfMemory_init()+0x68
C  0xb65adc7c
V  [libjvm.so+0x5f6aec]  Threads::create_vm(JavaVMInitArgs*, bool*)+0x1cc
V  [libjvm.so+0x3a2bc8]  JNI_CreateJavaVM+0x78

siginfo: si_signo: 4 (SIGILL), si_code: 1 (ILL_ILLOPC), si_addr: 0xb6b8daf8

Registers:
  r0  = 0xb6408840
  r1  = 0x00001de8
  r2  = 0x00000000
  r3  = 0x00000000
  r4  = 0xb65adcb0
  r5  = 0x00000000
  r6  = 0xb6d67000
  r7  = 0xb65adcf0
[... 184 lines cut ...]
CmaFree:               0 kB

CPU:total 1 (initial active 1) 

/proc/cpuinfo:
processor	: 0
model name	: ARM926EJ-S rev 5 (v5l)
BogoMIPS	: 148.88
Features	: swp half thumb fastmult edsp java 
CPU implementer	: 0x41
CPU architecture: 5TEJ
CPU variant	: 0x0
CPU part	: 0x926
CPU revision	: 5

Hardware	: LEGO MINDSTORMS EV3
Revision	: 0000
Serial		: 0000000000000000

Memory: 4k page, physical 57408k(1576k free), swap 98300k(92788k free)

vm_info: OpenJDK Client VM (10+46) for linux--sflt JRE (10+46), built on Mar 25 2018 20:16:11 by "" with gcc 6.3.0 20170516

END.
```

jabrena, 2018-03-27:
https://packages.debian.org/buster/openjdk-10-jdk

jabrena, 2018-03-27:
https://packages.debian.org/buster/java/openjdk-10-jre

jabrena, 2018-03-27:
https://packages.debian.org/buster/openjdk-10-jre-headless

jabrena, 2018-03-27:
sudo apt-get install openjdk-10-jre-headless --no-install-recommends

jabrena, 2018-03-27:
sudo apt-get install openjdk-11-jre-headless --no-install-recommends

jabrena, 2018-03-27:
`tar -xvzf community_images.tar.gz`

jabrena, 2018-03-27:
Available modules with Java 9:

``` 
java version "9.0.1"
Java(TM) SE Runtime Environment (build 9.0.1+11)
Java HotSpot(TM) 64-Bit Server VM (build 9.0.1+11, mixed mode)
Juans-MBP:~ jabrena$ java --list-modules
java.activation@9.0.1
java.base@9.0.1
java.compiler@9.0.1
java.corba@9.0.1
java.datatransfer@9.0.1
java.desktop@9.0.1
java.instrument@9.0.1
java.jnlp@9.0.1
java.logging@9.0.1
java.management@9.0.1
java.management.rmi@9.0.1
java.naming@9.0.1
java.prefs@9.0.1
java.rmi@9.0.1
java.scripting@9.0.1
java.se@9.0.1
java.se.ee@9.0.1
java.security.jgss@9.0.1
java.security.sasl@9.0.1
java.smartcardio@9.0.1
java.sql@9.0.1
java.sql.rowset@9.0.1
java.transaction@9.0.1
java.xml@9.0.1
java.xml.bind@9.0.1
java.xml.crypto@9.0.1
java.xml.ws@9.0.1
java.xml.ws.annotation@9.0.1
javafx.base@9.0.1
javafx.controls@9.0.1
javafx.deploy@9.0.1
javafx.fxml@9.0.1
javafx.graphics@9.0.1
javafx.media@9.0.1
javafx.swing@9.0.1
javafx.web@9.0.1
jdk.accessibility@9.0.1
jdk.attach@9.0.1
jdk.charsets@9.0.1
jdk.compiler@9.0.1
jdk.crypto.cryptoki@9.0.1
jdk.crypto.ec@9.0.1
jdk.deploy@9.0.1
jdk.deploy.controlpanel@9.0.1
jdk.dynalink@9.0.1
jdk.editpad@9.0.1
jdk.hotspot.agent@9.0.1
jdk.httpserver@9.0.1
jdk.incubator.httpclient@9.0.1
jdk.internal.ed@9.0.1
jdk.internal.jvmstat@9.0.1
jdk.internal.le@9.0.1
jdk.internal.opt@9.0.1
jdk.internal.vm.ci@9.0.1
jdk.jartool@9.0.1
jdk.javadoc@9.0.1
jdk.javaws@9.0.1
jdk.jcmd@9.0.1
jdk.jconsole@9.0.1
jdk.jdeps@9.0.1
jdk.jdi@9.0.1
jdk.jdwp.agent@9.0.1
jdk.jfr@9.0.1
jdk.jlink@9.0.1
jdk.jshell@9.0.1
jdk.jsobject@9.0.1
jdk.jstatd@9.0.1
jdk.localedata@9.0.1
jdk.management@9.0.1
jdk.management.agent@9.0.1
jdk.management.cmm@9.0.1
jdk.management.jfr@9.0.1
jdk.management.resource@9.0.1
jdk.naming.dns@9.0.1
jdk.naming.rmi@9.0.1
jdk.net@9.0.1
jdk.pack@9.0.1
jdk.packager@9.0.1
jdk.packager.services@9.0.1
jdk.plugin@9.0.1
jdk.plugin.dom@9.0.1
jdk.plugin.server@9.0.1
jdk.policytool@9.0.1
jdk.rmic@9.0.1
jdk.scripting.nashorn@9.0.1
jdk.scripting.nashorn.shell@9.0.1
jdk.sctp@9.0.1
jdk.security.auth@9.0.1
jdk.security.jgss@9.0.1
jdk.snmp@9.0.1
jdk.unsupported@9.0.1
jdk.xml.bind@9.0.1
jdk.xml.dom@9.0.1
jdk.xml.ws@9.0.1
jdk.zipfs@9.0.1
oracle.desktop@9.0.1
oracle.net@9.0.1
```

JakubVanek, 2018-03-27:
@jabrena well, that's the full OpenJDK distribution. EV3 JRE is (intentionally) jlinked to only `java.se` and `jdk.jdwp.agent`. I expect `java.se` to be a sort of a metapackage, so some more packages will be available.

JakubVanek, 2018-03-28:
Hmm, grep from jimage tool prints this on jre-ev3/lib/modules:
```
Module: java.base
Module: java.compiler
Module: java.datatransfer
Module: java.desktop
Module: java.instrument
Module: java.logging
Module: java.management.rmi
Module: java.management
Module: java.naming
Module: java.prefs
Module: java.rmi
Module: java.scripting
Module: java.se
Module: java.security.jgss
Module: java.security.sasl
Module: java.sql.rowset
Module: java.sql
Module: java.xml.crypto
Module: java.xml
Module: jdk.jdwp.agent
```

jabrena, 2018-03-28:
Running OpenJDK 10 on EV3!

```
robot@ev3dev:~/jre-ev3/bin$ ./java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46)
OpenJDK Client VM (build 10+46, mixed mode)
robot@ev3dev:~/jre-ev3/bin$ cd ../../
robot@ev3dev:~$ ls
HelloWorld.class  HelloWorld.java  jre-ev3  jre-ev3-openjdk10  jre-ev3-openjdk10.zip  jre-ev3.tar.gz  jre-ev3.zip
robot@ev3dev:~$ cp HelloWorld.class ./jre-ev3/bin/
robot@ev3dev:~$ cd jre-ev3/bin/
robot@ev3dev:~/jre-ev3/bin$ ./java HelloWorld       
Hello, World
```

jabrena, 2018-03-28:
It is time to automate the installation

jabrena, 2018-03-28:
Yesterday, I forked some repos from AdoptOpenJDK to do some LoadTests on the Brick.
In next days, I will integrate Installer project in the Template project to install the new OpenJDK for EV3 Brick in an automatic way. I will test the library in a heavy way. I will inform in this issue about the progress.

Yes, we should test Zero too.

Besides, we could create a folder for OpenJDK 11 to avoid surprises.

Amazing Job @JakubVanek !!!

JakubVanek, 2018-03-28:
https://github.com/ev3dev-lang-java/openjdk-ev3/commit/17ae3430c0aa463d807963dbab413e414938c868 adds a way how to build Zero.

JakubVanek, 2018-03-28:
Latest jtreg download: https://adopt-openjdk.ci.cloudbees.com/job/jtreg/lastSuccessfulBuild/artifact/

https://github.com/ev3dev-lang-java/jvm-benchmark isn't a working benchmark, the negative durations we got with openjdk9 are simply wrong. I suppose the processor cannot go back in time. So finding a new benchmark is a great idea.

JakubVanek, 2018-03-28:
OpenJDK SystemTest looks also promising.

jabrena, 2018-03-29:
http://openjdk.java.net/jtreg/

jabrena, 2018-03-30:
I am going to execute some JVM Benchmarks libraries in the brick to check the installation of the JRI.

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/220

jabrena, 2018-03-30:
**jvm-benchmark:**

``` 
java -server -jar /home/robot/jvm-benchmark-0.1.0-all.jar 
ev3dev#1|GeneralBench 1.2
ev3dev#1|       6 byte[16384] manual copies:       113 ms     869946 bytes/sec
ev3dev#1|    1220 byte[16384] arraycopies:         217 ms   92112811 bytes/sec
ev3dev#1|      24 int[4096] manual copies:         146 ms    2693260 bytes/sec
ev3dev#1|    1220 int[4096] arraycopies:           187 ms  106890267 bytes/sec
ev3dev#1|  200000 byte add:                        121 ms    1652892 ops/sec
ev3dev#1|  200000 byte sub:                          3 ms   66666666 ops/sec
ev3dev#1|  200000 byte mul:                          1 ms  200000000 ops/sec
ev3dev#1|  200000 byte div:                         46 ms    4347826 ops/sec
ev3dev#1|  200000 short add:                       130 ms    1538461 ops/sec
ev3dev#1|  200000 short sub:                         2 ms  100000000 ops/sec
ev3dev#1|  200000 short mul:                         1 ms  200000000 ops/sec
ev3dev#1|  200000 short div:                        60 ms    3333333 ops/sec
ev3dev#1|  200000 char add:                        128 ms    1562500 ops/sec
ev3dev#1|  200000 char sub:                          3 ms   66666666 ops/sec
ev3dev#1|  200000 char mul:                          3 ms   66666666 ops/sec
ev3dev#1|  200000 char div:                         54 ms    3703703 ops/sec
ev3dev#1|  200000 int add:                         131 ms    1526717 ops/sec
ev3dev#1|  200000 int sub:                           1 ms  200000000 ops/sec
ev3dev#1|  200000 int mul:                           9 ms   22222222 ops/sec
ev3dev#1|  200000 int div:                          47 ms    4255319 ops/sec
ev3dev#1|  200000 long add:                        330 ms     606060 ops/sec
ev3dev#1|  200000 long sub:                        319 ms     626959 ops/sec
ev3dev#1|  200000 long mul:                         -7 ms  -28571428 ops/sec
ev3dev#1|  200000 long div:                        133 ms    1503759 ops/sec
ev3dev#1|  200000 float add:                       184 ms    1086956 ops/sec
ev3dev#1|  200000 float sub:                        -1 ms -200000000 ops/sec
ev3dev#1|  200000 float mul:                         1 ms  200000000 ops/sec
ev3dev#1|  200000 float div:                         1 ms  200000000 ops/sec
ev3dev#1|  200000 double add:                      211 ms     947867 ops/sec
ev3dev#1|  200000 double sub:                        2 ms  100000000 ops/sec
ev3dev#1|  200000 double mul:                        5 ms   40000000 ops/sec
ev3dev#1|  200000 double div:                        1 ms  200000000 ops/sec
ev3dev#1|  200000 method calls:                    124 ms    1612903 ops/sec
ev3dev#1|  200000 static method calls:             114 ms    1754385 ops/sec
ev3dev#1|    2000 string concats:                  786 ms       2544 ops/sec
ev3dev#1|   20000 string compares (easy):           62 ms     322580 ops/sec
ev3dev#1|    1000 string compares (hard):            2 ms     500000 ops/sec
ev3dev#1|   20000 object creations:                150 ms     133333 ops/sec
ev3dev#1| 6242048 Total Loop Executions:          7253 ms     860616 loops/sec
ev3dev#1|Note: each Loop Execution includes multiple Java operations
:deployAndRun
```

jabrena, 2018-03-30:
Fixed issue on project: vmbenchmarks
https://github.com/google/auto/issues/334

In the execution on EV3, I detected the problem with the package: sun/misc/Unsafe

```
robot@ev3dev:~$ java -jar commonscenario-baseline-uber.jar 
Exception in thread "main" java.lang.NoClassDefFoundError: sun/misc/Unsafe
	at org.openjdk.jmh.util.Utils.<clinit>(Utils.java:48)
	at org.openjdk.jmh.runner.Runner.createOutputFormat(Runner.java:108)
	at org.openjdk.jmh.runner.Runner.<init>(Runner.java:89)
	at org.openjdk.jmh.Main.main(Main.java:43)
Caused by: java.lang.ClassNotFoundException: sun.misc.Unsafe
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(Unknown Source)
	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
	... 4 more
```

Testing `JDK` release

jabrena, 2018-03-30:
`update-alternatives --config java
`

jabrena, 2018-03-30:
Modules added in Java 9:

``` 
jdk/jmods/
jdk/jmods/java.transaction.jmod
jdk/jmods/jdk.incubator.httpclient.jmod
jdk/jmods/jdk.accessibility.jmod
jdk/jmods/jdk.crypto.cryptoki.jmod
jdk/jmods/jdk.jshell.jmod
jdk/jmods/jdk.attach.jmod
jdk/jmods/jdk.charsets.jmod
jdk/jmods/jdk.scripting.nashorn.shell.jmod
jdk/jmods/jdk.compiler.jmod
jdk/jmods/jdk.jdi.jmod
jdk/jmods/jdk.security.jgss.jmod
jdk/jmods/jdk.rmic.jmod
jdk/jmods/java.sql.jmod
jdk/jmods/jdk.internal.ed.jmod
jdk/jmods/java.management.jmod
jdk/jmods/jdk.internal.vm.ci.jmod
jdk/jmods/java.datatransfer.jmod
jdk/jmods/java.desktop.jmod
jdk/jmods/jdk.net.jmod
jdk/jmods/jdk.editpad.jmod
jdk/jmods/java.xml.bind.jmod
jdk/jmods/jdk.naming.rmi.jmod
jdk/jmods/jdk.jstatd.jmod
jdk/jmods/jdk.pack.jmod
jdk/jmods/java.instrument.jmod
jdk/jmods/jdk.jartool.jmod
jdk/jmods/jdk.scripting.nashorn.jmod
jdk/jmods/java.se.ee.jmod
jdk/jmods/jdk.jdwp.agent.jmod
jdk/jmods/jdk.crypto.ec.jmod
jdk/jmods/java.sql.rowset.jmod
jdk/jmods/jdk.xml.ws.jmod
jdk/jmods/jdk.sctp.jmod
jdk/jmods/java.smartcardio.jmod
jdk/jmods/java.rmi.jmod
jdk/jmods/java.logging.jmod
jdk/jmods/java.management.rmi.jmod
jdk/jmods/java.prefs.jmod
jdk/jmods/jdk.jdeps.jmod
jdk/jmods/jdk.xml.dom.jmod
jdk/jmods/java.xml.jmod
jdk/jmods/jdk.zipfs.jmod
jdk/jmods/java.corba.jmod
jdk/jmods/jdk.httpserver.jmod
jdk/jmods/java.security.sasl.jmod
jdk/jmods/jdk.management.jmod
jdk/jmods/java.base.jmod
jdk/jmods/jdk.jsobject.jmod
jdk/jmods/java.xml.ws.annotation.jmod
jdk/jmods/jdk.internal.jvmstat.jmod
jdk/jmods/jdk.jlink.jmod
jdk/jmods/jdk.naming.dns.jmod
jdk/jmods/java.security.jgss.jmod
jdk/jmods/java.naming.jmod
jdk/jmods/java.xml.crypto.jmod
jdk/jmods/jdk.internal.le.jmod
jdk/jmods/jdk.hotspot.agent.jmod
jdk/jmods/jdk.unsupported.jmod
jdk/jmods/jdk.javadoc.jmod
jdk/jmods/jdk.jconsole.jmod
jdk/jmods/jdk.policytool.jmod
jdk/jmods/java.activation.jmod
jdk/jmods/jdk.management.agent.jmod
jdk/jmods/java.se.jmod
jdk/jmods/jdk.dynalink.jmod
jdk/jmods/java.xml.ws.jmod
jdk/jmods/jdk.security.auth.jmod
jdk/jmods/jdk.localedata.jmod
jdk/jmods/java.compiler.jmod
jdk/jmods/jdk.internal.opt.jmod
jdk/jmods/java.scripting.jmod
jdk/jmods/jdk.jcmd.jmod
jdk/jmods/jdk.xml.bind.jmod
```

We should add this module: `jdk/jmods/jdk.unsupported.jmod` in next compilation to add support for reflection features. This Module add the controversial package: `sun/misc/Unsafe`

jabrena, 2018-03-30:
How to test the project in the Brick:
https://github.com/ev3dev-lang-java/vmbenchmarks -f 1

scp ./target/vmbenchmarks.jar robot@192.168.1.88:/home/robot

`jmh` tutorial:
https://github.com/guozheng/jmh-tutorial/blob/master/README.md

jabrena, 2018-03-30:
```
robot@ev3dev:~$ java -jar benchmarks.jar -f 1
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/benchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
=============== DEBUG MESSAGE: Atomic load(jlong) unsupported on this platform ================

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

=============== DEBUG MESSAGE: Atomic load(jlong) unsupported on this platform ================

[thread 4430 also had an error]
[error occurred during error reporting ((null)), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing fatal error message), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing type of error), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing exception/signal name), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing current thread and pid), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing error message), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing Java version string), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing problematic frame), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing core file information), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing bug submit message), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing summary), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing VM option summary), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing summary machine and OS info), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing date and time), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[... 47 lines cut ...]

[error occurred during error reporting (printing VM operation), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing process), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing all threads), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing VM state), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing owned locks on error), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[Too many errors, abort]
Aborted

```

jabrena, 2018-03-30:
Similar Bug:
https://bugs.openjdk.java.net/browse/JDK-8146540

jabrena, 2018-03-30:
https://docs.oracle.com/javase/7/docs/webnotes/tsg/TSG-VM/html/clopts.html

jabrena, 2018-03-30:
It is necessary to review the Build for OpenJDK10 to check ARM support for Concurrency:

```
robot@ev3dev:~$ sudo rm /tmp/jmh.lock 
robot@ev3dev:~$ sudo update-alternatives --config java
There are 4 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                        Priority   Status
------------------------------------------------------------
  0            /opt/jdk-9-build-46/bin/java                 1         auto mode
  1            /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java   1         manual mode
  2            /opt/jdk-10-build-46/bin/java                1         manual mode
* 3            /opt/jdk-9-build-46/bin/java                 1         manual mode
  4            /opt/jri-10-build-46/bin/java                1         manual mode

Press <enter> to keep the current choice[*], or type selection number: 2
update-alternatives: using /opt/jdk-10-build-46/bin/java to provide /usr/bin/java (java) in manual mode
robot@ev3dev:~$ java -jar benchmarks.jar -wi 1 -i 1 -f 1
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/benchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
=============== DEBUG MESSAGE: Atomic load(jlong) unsupported on this platform ================

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting ((null)), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing fatal error message), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing type of error), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing exception/signal name), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing current thread and pid), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing error message), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing Java version string), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing problematic frame), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing core file information), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing bug submit message), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing summary), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================
[... 63 lines cut ...]

[error occurred during error reporting (printing VM operation), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing process), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing all threads), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing VM state), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing owned locks on error), id 0xe0000000]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[Too many errors, abort]
Aborted
robot@ev3dev:~$ 
```

JakubVanek, 2018-03-31:
Hmmm, that's really strange. cmpxchg64 should work. It was added back in [2011](http://lists.infradead.org/pipermail/linux-arm-kernel/2011-June/053581.html), kernel version 3.1. And it seems that OpenJDK 9 uses it without any problems.

jabrena, 2018-03-31:
Yes, OpenJDK 9, run fine with the tests. 

Maybe we should review the same file between both versions to check the differences:

OpenJDK 9:
https://github.com/AdoptOpenJDK/openjdk-jdk9-backup/blob/master/hotspot/src/cpu/arm/vm/assembler_arm_32.hpp

Path for OpenJDK 9:
https://github.com/ev3dev-lang-java/openjdk-ev3/blob/master/scripts/jdk9.patch

Function in OpenJDK9:

``` cpp
  void dmb(DMB_Opt opt, Register reg) {
    if (VM_Version::arm_arch() >= 7) {
      emit_int32(0xF57FF050 | opt);
    } else {
      bool preserve_tmp = (reg == noreg);
      if(preserve_tmp) {
        reg = Rtemp;
        str(reg, Address(SP, -wordSize, pre_indexed));
      }
      mov(reg, 0);
      // DataMemoryBarrier
      emit_int32(0xe << 28 |
                0xe << 24 |
                0x7 << 16 |
                reg->encoding() << 12  |
                0xf << 8  |
                0xb << 4  |
                0xa);
      if(preserve_tmp) {
        ldr(reg, Address(SP, wordSize, post_indexed));
      }
    }
  }
```

OpenJDK 10:

jabrena, 2018-03-31:
I checked both entire files and the unique difference between 10 in compare with 9 is this method:

``` cpp
void udf(int imm_16) {
    assert((imm_16 >> 16) == 0, "encoding constraint");
    emit_int32(0xe7f000f0 | (imm_16 & 0xfff0) << 8 | (imm_16 & 0xf));
}
```

jabrena, 2018-03-31:
Maybe we should check the differences between the whole folder:
http://hg.openjdk.java.net/jdk/jdk10/file/b09e56145e11/src/hotspot/cpu/arm
https://github.com/AdoptOpenJDK/openjdk-jdk9-backup/tree/master/hotspot/src/cpu/arm/vm

Besides as you commented, we could try with Zero too:
http://hg.openjdk.java.net/jdk/jdk10/file/b09e56145e11/src/hotspot/cpu/zero

Juan Antonio

JakubVanek, 2018-03-31:
There is no problem with memory barrier. The problem is with atomic operations.

jabrena, 2018-03-31:
To compare 2 folders:

`diff -rq folder-1 folder-2`
https://github.com/yousseb/meld/releases/

jabrena, 2018-03-31:
File: stubGenerator_arm.cpp

``` cpp
    BarrierSet* bs = Universe::heap()->barrier_set();
    if (bs->has_write_ref_pre_barrier()) {
      assert(bs->has_write_ref_array_pre_opt(),
             "Else unsupported barrier set.");

      assert( addr->encoding() < callee_saved_regs, "addr must be saved");
      assert(count->encoding() < callee_saved_regs, "count must be saved");

```

file: vm_version_arm32.cpp

`AllocatePrefetchDistance = 128;`

jabrena, 2018-03-31:
I am going to publish a bug in OpenJDK, in parallel to this research. 

If in OpenJDK9, Atomic Operations run, In OpenJDK10 should run too.
https://bugs.openjdk.java.net/secure/Dashboard.jspa

JakubVanek, 2018-03-31:
They aren't going to fix it, ARMv5 isn't supported. Moreover supports_cx8() should return true - the kernel should support kuser arm helpers >=5.
GDB will tell us more.

jabrena, 2018-03-31:
Bug reported.

``` 

We will review your report and have assigned it an internal review ID : 9053189 . Depending upon the completeness of the report and our ability to reproduce the problem, either a new bug will be posted, or we will contact you for further information.
```

jabrena, 2018-04-04:
Today, I will try to subscribe to some mailing lists from OpenJDK to explain the case about the `ARM5` support.
http://mail.openjdk.java.net/mailman/listinfo

jabrena, 2018-04-04:
Email sent to:
http://mail.openjdk.java.net/pipermail/build-dev/

jabrena, 2018-04-04:
http://mail.openjdk.java.net/pipermail/build-dev/2018-April/021479.html

jabrena, 2018-04-05:
https://bugs.openjdk.java.net/browse/JDK-8200580

jabrena, 2018-04-06:
Eureka @JakubVanek !!!

Using the VM parameter from David, the benchmark run!!!

Take a look the output:

```
robot@ev3dev:~$ Connection to 192.168.1.88 closed by remote host.
Connection to 192.168.1.88 closed.
Juans-MacBook-Pro:openjdk10 jabrena$ ssh robot@192.168.1.88
ssh: connect to host 192.168.1.88 port 22: Operation timed out
Juans-MacBook-Pro:openjdk10 jabrena$ ssh robot@192.168.43.88
The authenticity of host '192.168.43.88 (192.168.43.88)' can't be established.
ECDSA key fingerprint is SHA256:sNNNlUWTtDtWX0CbCLcC83h8Yohuql2x2f6elfXKf+c.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.43.88' (ECDSA) to the list of known hosts.
Password: 
Linux ev3dev 4.9.58-ev3dev-1.6.0-ev3 #1 PREEMPT Tue Oct 31 22:16:58 CDT 2017 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Sun Apr  1 09:00:24 2018 from 192.168.1.97
robot@ev3dev:~$ ls
0                 HelloWorld.java  ejdk-8-fcs-b132-linux-arm-sflt-03_mar_2014.tar.gz  installer                    nod_low_power.wav
HelloWorld.class  benchmarks.jar   ev3dev-lang-java-2.1.0-SNAPSHOT-all.jar            jvm-benchmark-0.1.0-all.jar  vmbenchmarks.jar
robot@ev3dev:~$ sudo update-alternatives --config java
[sudo] password for robot: 
There are 4 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                        Priority   Status
------------------------------------------------------------
  0            /opt/jdk-9-build-46/bin/java                 1         auto mode
  1            /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java   1         manual mode
  2            /opt/jdk-10-build-46/bin/java                1         manual mode
* 3            /opt/jdk-9-build-46/bin/java                 1         manual mode
  4            /opt/jri-10-build-46/bin/java                1         manual mode

Press <enter> to keep the current choice[*], or type selection number: 2
update-alternatives: using /opt/jdk-10-build-46/bin/java to provide /usr/bin/java (java) in manual mode
robot@ev3dev:~$ java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode)
robot@ev3dev:~$ ls
0                 HelloWorld.java  ejdk-8-fcs-b132-linux-arm-sflt-03_mar_2014.tar.gz  installer                    nod_low_power.wav
HelloWorld.class  benchmarks.jar   ev3dev-lang-java-2.1.0-SNAPSHOT-all.jar            jvm-benchmark-0.1.0-all.jar  vmbenchmarks.jar
robot@ev3dev:~$ java -XX:-AssumeMP HelloWorld 
OpenJDK Client VM warning: Option AssumeMP was deprecated in version 10.0 and will likely be removed in a future release.
Hello, World
robot@ev3dev:~$ java -XX:-AssumeMP -jar vmbenchmarks.jar -wi 1 -i 1 -f 1
OpenJDK Client VM warning: Option AssumeMP was deprecated in version 10.0 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/vmbenchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
# JMH version: 1.20
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jdk-10-build-46/bin/java
# VM options: -XX:-AssumeMP
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 1 thread, will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: com.github.vmbenchmarks.DynamicLanguages.groovy
[... 618 lines cut ...]
Benchmark                                       Mode  Cnt        Score   Error  Units
DynamicLanguages.groovy                        thrpt             0.075          ops/s
ExceptionsBenchMark.throwRuntimeException      thrpt          1030.309          ops/s
GeneratePoiWorkBook.generatePoiXLSWorkBook     thrpt             2.902          ops/s
GeneratePoiWorkBook.generatePoiXLSXWorkBook    thrpt             0.092          ops/s
GeneratePrimeNumbersWithForLoop.generatePrime  thrpt          3429.985          ops/s
Jackson.jsonMashal                             thrpt             1.806          ops/s
Jackson.jsonUnmashal                           thrpt             2.182          ops/s
Jackson.xmlMashalJackson                       thrpt             1.603          ops/s
Jackson.xmlMashalJacksonStatic                 thrpt             0.590          ops/s
Jackson.xmlUnmashalJackson                     thrpt             0.710          ops/s
Jackson.xmlUnmashalJacksonStatic               thrpt             9.750          ops/s
MicroBenchMarks.arrayListForeach               thrpt         71821.297          ops/s
MicroBenchMarks.arrayListIterator              thrpt        111942.761          ops/s
MicroBenchMarks.calculateLogPi                 thrpt         38142.079          ops/s
MicroBenchMarks.calculateLogPiCodeElimination  thrpt         42947.611          ops/s
MicroBenchMarks.deadCodeElimination            thrpt       6713946.581          ops/s
MicroBenchMarks.loopThatCanBeUnrolled          thrpt           906.742          ops/s
MicroBenchMarks.manualUnrollOfLoop             thrpt          1040.036          ops/s
MicroBenchMarks.noDeadCodeElimination          thrpt         33852.666          ops/s
Reflection.doReflection                        thrpt          2395.990          ops/s
Reflection.withoutReflection                   thrpt       3230871.698          ops/s
RegularExpressions.validateEmail               thrpt            18.078          ops/s
robot@ev3dev:~$ 
```

jabrena, 2018-04-06:
We should apply the localPath to include this parameter enabled by default. :D
We are near!!!

jabrena, 2018-04-06:
Second level of testing:

```
robot@ev3dev:~$ java -XX:-AssumeMP -jar benchmarks.jar -wi 1 -i 1 -f 1  
OpenJDK Client VM warning: Option AssumeMP was deprecated in version 10.0 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/benchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
# JMH 1.16 (released 508 days ago, please consider updating!)
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jdk-10-build-46/bin/java
# VM options: -XX:-AssumeMP
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 10 threads (1 group; 5x "testAtomicLongCounterRead", 5x "testAtomicLongCounterWrite" in each group), will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: net.learn.java.concurrency.counter.CounterPerfTest.AtomicLongCounter

# Run progress: 0.00% complete, ETA 00:01:06
# Fork: 1 of 1
OpenJDK Client VM warning: Option AssumeMP was deprecated in version 10.0 and will likely be removed in a future release.
# Warmup Iteration   1: 1627.154 ops/s
Iteration   1: 1587.519 ops/s
                 testAtomicLongCounterRead:  1136.452 ops/s
                 testAtomicLongCounterWrite: 451.067 ops/s

Result "AtomicLongCounter":
  1587.519 ops/s

Secondary result "testAtomicLongCounterRead":
  1136.452 ops/s

Secondary result "testAtomicLongCounterWrite":
  451.067 ops/s

# JMH 1.16 (released 508 days ago, please consider updating!)
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jdk-10-build-46/bin/java
# VM options: -XX:-AssumeMP
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 10 threads (1 group; 5x "testFairReentrantLockCounterRead", 5x "testFairReentrantLockCounterWrite" in each group), will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: net.learn.java.concurrency.counter.CounterPerfTest.FairReentrantLockCounter

# Run progress: 3.03% complete, ETA 00:30:20
# Fork: 1 of 1
OpenJDK Client VM warning: Option AssumeMP was deprecated in version 10.0 and will likely be removed in a future release.
# Warmup Iteration   1: 508.353 ops/s
Iteration   1: 650.404 ops/s
                 testFairReentrantLockCounterRead:  356.770 ops/s
                 testFairReentrantLockCounterWrite: 293.634 ops/s

Result "FairReentrantLockCounter":
  650.404 ops/s

Secondary result "testFairReentrantLockCounterRead":
  356.770 ops/s

Secondary result "testFairReentrantLockCounterWrite":
  293.634 ops/s

# JMH 1.16 (released 508 days ago, please consider updating!)
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jdk-10-build-46/bin/java
# VM options: -XX:-AssumeMP
[... 2777 lines cut ...]
CounterPerfTest.LongAdderCounter                                                                          ss           0.009            s/op
CounterPerfTest.LongAdderCounter:testLongAdderCounterRead                                                 ss           0.006            s/op
CounterPerfTest.LongAdderCounter:testLongAdderCounterWrite                                                ss           0.012            s/op
CounterPerfTest.OptimisticStampedLockCounter                                                              ss           0.008            s/op
CounterPerfTest.OptimisticStampedLockCounter:testOptimisticStampedLockCounterRead                         ss           0.004            s/op
CounterPerfTest.OptimisticStampedLockCounter:testOptimisticStampedLockCounterWrite                        ss           0.012            s/op
CounterPerfTest.ReentrantLockCounter                                                                      ss           0.010            s/op
CounterPerfTest.ReentrantLockCounter:testReentrantLockCounterRead                                         ss           0.009            s/op
CounterPerfTest.ReentrantLockCounter:testReentrantLockCounterWrite                                        ss           0.012            s/op
CounterPerfTest.ReentrantReadWriteLockCounter                                                             ss           0.009            s/op
CounterPerfTest.ReentrantReadWriteLockCounter:testReentrantReadWriteLockCounterRead                       ss           0.006            s/op
CounterPerfTest.ReentrantReadWriteLockCounter:testReentrantReadWriteLockCounterWrite                      ss           0.011            s/op
CounterPerfTest.SemaphoreCounter                                                                          ss           0.007            s/op
CounterPerfTest.SemaphoreCounter:testSemaphoreCounterRead                                                 ss           0.004            s/op
CounterPerfTest.SemaphoreCounter:testSemaphoreCounterWrite                                                ss           0.011            s/op
CounterPerfTest.StampedLockCounter                                                                        ss           0.010            s/op
CounterPerfTest.StampedLockCounter:testStampedLockCounterRead                                             ss           0.008            s/op
[... cut ...]
CounterPerfTest.VolatileCounter:testSynchronizedCounterRead                                               ss           0.020            s/op
CounterPerfTest.VolatileCounter:testSynchronizedCounterWrite                                              ss           0.020            s/op
CounterPerfTest.VolatileCounter:testVolatileCounterRead                                                   ss           0.014            s/op
CounterPerfTest.VolatileCounter:testVolatileCounterWrite                                                  ss           0.017            s/op
robot@ev3dev:~$ 
```

jabrena, 2018-04-06:
Third level of testing:

``` 
robot@ev3dev:~$ java -XX:-AssumeMP -jar jvm-benchmark-0.1.0-all.jar 
OpenJDK Client VM warning: Option AssumeMP was deprecated in version 10.0 and will likely be removed in a future release.
GeneralBench 1.2
       6 byte[16384] manual copies:       114 ms     862315 bytes/sec
    1220 byte[16384] arraycopies:         159 ms  125713710 bytes/sec
      24 int[4096] manual copies:         146 ms    2693260 bytes/sec
    1220 int[4096] arraycopies:           155 ms  128957935 bytes/sec
  200000 byte add:                        131 ms    1526717 ops/sec
  200000 byte sub:                          2 ms  100000000 ops/sec
  200000 byte mul:                          3 ms   66666666 ops/sec
  200000 byte div:                         51 ms    3921568 ops/sec
  200000 short add:                       129 ms    1550387 ops/sec
  200000 short sub:                         3 ms   66666666 ops/sec
  200000 short mul:                         7 ms   28571428 ops/sec
  200000 short div:                        45 ms    4444444 ops/sec
  200000 char add:                        123 ms    1626016 ops/sec
  200000 char sub:                          1 ms  200000000 ops/sec
  200000 char mul:                         -3 ms  -66666666 ops/sec
  200000 char div:                         39 ms    5128205 ops/sec
  200000 int add:                         121 ms    1652892 ops/sec
  200000 int sub:                           1 ms  200000000 ops/sec
  200000 int mul:                           1 ms  200000000 ops/sec
  200000 int div:                          63 ms    3174603 ops/sec
  200000 long add:                        431 ms     464037 ops/sec
  200000 long sub:                        397 ms     503778 ops/sec
  200000 long mul:                        138 ms    1449275 ops/sec
  200000 long div:                        144 ms    1388888 ops/sec
  200000 float add:                       196 ms    1020408 ops/sec
  200000 float sub:                        -3 ms  -66666666 ops/sec
  200000 float mul:                        -3 ms  -66666666 ops/sec
  200000 float div:                         3 ms   66666666 ops/sec
  200000 double add:                      216 ms     925925 ops/sec
  200000 double sub:                        3 ms   66666666 ops/sec
  200000 double mul:                        1 ms  200000000 ops/sec
  200000 double div:                        1 ms  200000000 ops/sec
  200000 method calls:                    100 ms    2000000 ops/sec
  200000 static method calls:              87 ms    2298850 ops/sec
    2000 string concats:                  689 ms       2902 ops/sec
   20000 string compares (easy):          123 ms     162601 ops/sec
    1000 string compares (hard):            2 ms     500000 ops/sec
   20000 object creations:                159 ms     125786 ops/sec
 6242048 Total Loop Executions:          8143 ms     766553 loops/sec
Note: each Loop Execution includes multiple Java operations
robot@ev3dev:~$ 
```

jabrena, 2018-04-07:
```
scp foobar.txt your_username@remotehost.edu:/some/remote/directory
sudo update-alternatives --config java
```

```
robot@ev3dev:~$ java HelloWorld 
Hello, World
robot@ev3dev:~$ java -jar jvm-benchmark-0.1.0-all.jar 
GeneralBench 1.2
       6 byte[16384] manual copies:       108 ms     910222 bytes/sec
    1220 byte[16384] arraycopies:         178 ms  112294831 bytes/sec
      24 int[4096] manual copies:         139 ms    2828892 bytes/sec
    1220 int[4096] arraycopies:           169 ms  118275029 bytes/sec
  200000 byte add:                        123 ms    1626016 ops/sec
  200000 byte sub:                          2 ms  100000000 ops/sec
  200000 byte mul:                          4 ms   50000000 ops/sec
  200000 byte div:                         39 ms    5128205 ops/sec
  200000 short add:                       119 ms    1680672 ops/sec
  200000 short sub:                         1 ms  200000000 ops/sec
  200000 short mul:                         1 ms  200000000 ops/sec
  200000 short div:                        52 ms    3846153 ops/sec
  200000 char add:                        122 ms    1639344 ops/sec
  200000 char sub:                          3 ms   66666666 ops/sec
  200000 char mul:                          4 ms   50000000 ops/sec
  200000 char div:                         52 ms    3846153 ops/sec
  200000 int add:                         114 ms    1754385 ops/sec
  200000 int sub:                           1 ms  200000000 ops/sec
  200000 int mul:                           2 ms  100000000 ops/sec
  200000 int div:                          69 ms    2898550 ops/sec
  200000 long add:                        357 ms     560224 ops/sec
  200000 long sub:                        321 ms     623052 ops/sec
  200000 long mul:                          2 ms  100000000 ops/sec
  200000 long div:                        136 ms    1470588 ops/sec
  200000 float add:                       187 ms    1069518 ops/sec
  200000 float sub:                        -1 ms -200000000 ops/sec
  200000 float mul:                        -1 ms -200000000 ops/sec
  200000 float div:                         1 ms  200000000 ops/sec
  200000 double add:                      210 ms     952380 ops/sec
  200000 double sub:                       -1 ms -200000000 ops/sec
  200000 double mul:                        1 ms  200000000 ops/sec
  200000 double div:                        2 ms  100000000 ops/sec
  200000 method calls:                    123 ms    1626016 ops/sec
  200000 static method calls:             126 ms    1587301 ops/sec
    2000 string concats:                  769 ms       2600 ops/sec
   20000 string compares (easy):          141 ms     141843 ops/sec
    1000 string compares (hard):            6 ms     166666 ops/sec
   20000 object creations:                150 ms     133333 ops/sec
 6242048 Total Loop Executions:          7217 ms     864908 loops/sec
Note: each Loop Execution includes multiple Java operations
robot@ev3dev:~$ 

```

```
java -jar vmbenchmarks.jar -wi 1 -i 1 -f 1
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/vmbenchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
# JMH version: 1.20
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jri-10-build-46-v2/bin/java
# VM options: <none>
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 1 thread, will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: com.github.vmbenchmarks.DynamicLanguages.groovy

# Run progress: 0.00% complete, ETA 00:00:52
# Fork: 1 of 1
```

```
java -jar benchmarks.jar -wi 1 -i 1 -f 1
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/benchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
# JMH 1.16 (released 510 days ago, please consider updating!)
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jri-10-build-46-v2/bin/java
# VM options: <none>
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 10 threads (1 group; 5x "testAtomicLongCounterRead", 5x "testAtomicLongCounterWrite" in each group), will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: net.learn.java.concurrency.counter.CounterPerfTest.AtomicLongCounter

# Run progress: 0.00% complete, ETA 00:01:06
# Fork: 1 of 1
```

JakubVanek, 2018-04-07:
Hmmm, we should introduce some way of sharing logs without making the issue threads unbelievably long. Do you think a GitHub wiki page, GitHub Gists or a separate pastebin-like repo would be good for that?

JakubVanek, 2018-04-07:
EV3 is travelling back in time again:
```
  200000 float sub:                        -1 ms -200000000 ops/sec
  200000 float mul:                        -1 ms -200000000 ops/sec
  200000 double sub:                       -1 ms -200000000 ops/sec
```
We should call it Doctor Who benchmark.

jabrena, 2018-04-07:
```
robot@ev3dev:/opt$ time java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Zero VM (build 10+46-ev3, interpreted mode)

real	0m48.478s
user	0m38.040s
sys	0m0.010s

robot@ev3dev:~$ time java HelloWorld 
Hello, World

real	1m36.040s
user	0m0.000s
sys	1m12.120s

robot@ev3dev:~$ sudo update-alternatives --config java
There are 6 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                        Priority   Status
------------------------------------------------------------
  0            /opt/jri-10-build-46-zero/bin/java           1         auto mode
  1            /opt/ejdk1.8.0/linux_arm_sflt/jre/bin/java   1         manual mode
  2            /opt/jdk-10-build-46/bin/java                1         manual mode
  3            /opt/jdk-9-build-46/bin/java                 1         manual mode
  4            /opt/jri-10-build-46-v2/bin/java             1         manual mode
* 5            /opt/jri-10-build-46-zero/bin/java           1         manual mode
  6            /opt/jri-10-build-46/bin/java                1         manual mode

robot@ev3dev:~$ time java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode)

real	0m7.771s
user	0m4.890s
sys	0m0.000s
robot@ev3dev:~$ time java HelloWorld
Hello, World

real	0m7.311s
user	0m5.480s
sys	0m0.010s

robot@ev3dev:~$ time java -version
java version "1.8.0"
Java(TM) SE Embedded Runtime Environment (build 1.8.0-b132, headless)
Java HotSpot(TM) Embedded Client VM (build 25.0-b70, mixed mode)

real	0m5.988s
user	0m3.850s
sys	0m0.000s
```

jabrena, 2018-04-07:
``` 
robot@ev3dev:/opt$ time java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode)

real	0m8.715s
user	0m5.940s
sys	0m0.010s
robot@ev3dev:/opt$ cd /home/robot/
robot@ev3dev:~$ time java HelloWorld 
Hello, World

real	0m8.978s
user	0m0.000s
sys	0m6.790s
robot@ev3dev:~$ java -jar jvm-benchmark-0.1.0-all.jar 
GeneralBench 1.2
       6 byte[16384] manual copies:       108 ms     910222 bytes/sec
    1220 byte[16384] arraycopies:         189 ms  105759153 bytes/sec
      24 int[4096] manual copies:         126 ms    3120761 bytes/sec
    1220 int[4096] arraycopies:           162 ms  123385679 bytes/sec
  200000 byte add:                        107 ms    1869158 ops/sec
  200000 byte sub:                          1 ms  200000000 ops/sec
  200000 byte mul:                          5 ms   40000000 ops/sec
  200000 byte div:                         42 ms    4761904 ops/sec
  200000 short add:                       114 ms    1754385 ops/sec
  200000 short sub:                         2 ms  100000000 ops/sec
  200000 short mul:                         1 ms  200000000 ops/sec
  200000 short div:                        38 ms    5263157 ops/sec
  200000 char add:                        122 ms    1639344 ops/sec
  200000 char sub:                          5 ms   40000000 ops/sec
  200000 char mul:                          5 ms   40000000 ops/sec
  200000 char div:                         42 ms    4761904 ops/sec
  200000 int add:                         121 ms    1652892 ops/sec
  200000 int sub:                           4 ms   50000000 ops/sec
  200000 int mul:                           2 ms  100000000 ops/sec
  200000 int div:                          59 ms    3389830 ops/sec
  200000 long add:                        358 ms     558659 ops/sec
  200000 long sub:                        350 ms     571428 ops/sec
  200000 long mul:                        157 ms    1273885 ops/sec
  200000 long div:                        142 ms    1408450 ops/sec
  200000 float add:                       172 ms    1162790 ops/sec
  200000 float sub:                         4 ms   50000000 ops/sec
  200000 float mul:                         1 ms  200000000 ops/sec
  200000 float div:                         3 ms   66666666 ops/sec
  200000 double add:                      214 ms     934579 ops/sec
  200000 double sub:                        1 ms  200000000 ops/sec
  200000 double mul:                       -2 ms -100000000 ops/sec
  200000 double div:                        2 ms  100000000 ops/sec
  200000 method calls:                     88 ms    2272727 ops/sec
  200000 static method calls:              84 ms    2380952 ops/sec
    2000 string concats:                  700 ms       2857 ops/sec
   20000 string compares (easy):          116 ms     172413 ops/sec
    1000 string compares (hard):            5 ms     200000 ops/sec
   20000 object creations:                136 ms     147058 ops/sec
 6242048 Total Loop Executions:          7741 ms     806361 loops/sec
Note: each Loop Execution includes multiple Java operations
robot@ev3dev:~$ java -jar vmbenchmarks.jar -wi 1 -i 1 -f 1
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/vmbenchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
# JMH version: 1.20
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jdk-10-build-46-v2/bin/java
# VM options: <none>
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 1 thread, will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: com.github.vmbenchmarks.DynamicLanguages.groovy

# Run progress: 0.00% complete, ETA 00:00:52
# Fork: 1 of 1
^Crobot@ev3dev:~$ java -jar benchmarks.jar -wi 1 -i 1 -f 1
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.openjdk.jmh.util.Utils (file:/home/robot/benchmarks.jar) to field java.io.Console.cs
WARNING: Please consider reporting this to the maintainers of org.openjdk.jmh.util.Utils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
# JMH 1.16 (released 510 days ago, please consider updating!)
# VM version: JDK 10, VM 10+46-ev3
# VM invoker: /opt/jdk-10-build-46-v2/bin/java
# VM options: <none>
# Warmup: 1 iterations, 1 s each
# Measurement: 1 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 10 threads (1 group; 5x "testAtomicLongCounterRead", 5x "testAtomicLongCounterWrite" in each group), will synchronize iterations
# Benchmark mode: Throughput, ops/time
# Benchmark: net.learn.java.concurrency.counter.CounterPerfTest.AtomicLongCounter

# Run progress: 0.00% complete, ETA 00:01:06
# Fork: 1 of 1
^Crobot@ev3dev:~$ 
```

jabrena, 2018-04-07:
Hi @JakubVanek ,

Yes, paste logs are a bit ugly. We could think in one solution.

jabrena, 2018-04-07:
Definitively, JRI 10 is the winner in compare to a JDK version. Less class to load, always is better.

@JakubVanek, you are a Super Czech Hacker! :)

[7 more replies]

closed 2018-04-07 by jabrena