ev3dev-lang-java/ev3dev-lang-java #417: Why java is so slow on ev3

RareScrap, 2018-03-16:
I installed ev3dev-lang-Java according to the instructions, and I started this program:
EV3LargeRegulatedMotor motorLeft = new EV3LargeRegulatedMotor(MotorPort.A);
        motorLeft.brake();

        for (int motorSpeed = 200; motorSpeed <= 1000; motorSpeed+= 300) {
            motorLeft.setSpeed(motorSpeed);

            motorLeft.forward();
            Delay.msDelay(2000);
            motorLeft.stop();

            motorLeft.backward();
            Delay.msDelay(2000);
            motorLeft.stop();
        }
        System.exit(0);

 The program for a very long time creates the EV3LargeRegulatedMotor motorLeft, but in the future it works well. Why is this happening and how to fix it?

jartu, 2018-03-16:
I have the same problem. Unfortunately, so far, I haven't found a way to speed things up. I believe one suggestion was to create a Java MicroEdition-like profile to limit the size of the executable (by excluding a lot of the packages one wouldn't normally need on small devices) but I haven't tried that yet.

jabrena, 2018-03-16:
Hi mate, I will do a quick answer but this weekend I will extend the answer with some logs examples.

Using the logs example from this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/178

```
java -jar /home/robot/ev3dev-lang-java-all-0.4.0-SNAPSHOT.jar
ev3dev#5|2017-02-18 12:13:41 [main] INFO  e.actuators.motors.LargeMotorDemo - Starting motors on A
ev3dev#5|2017-02-18 12:13:41 [main] TRACE ev3dev.hardware.EV3DevPlatform - Detected platform: EV3BRICK
ev3dev#5|2017-02-18 12:13:41 [main] DEBUG e.a.motors.BaseRegulatedMotor - Detecting motor on port: outA
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: outA
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port0/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port1/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port2/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port3/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port4/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port4/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG e.a.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: outA
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/tacho-motor
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor1/address
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/tacho-motor/motor1/address
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - echo reset > /sys/class/tacho-motor/motor1/command
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - echo 500 > /sys/class/tacho-motor/motor1/speed_sp
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - echo brake > /sys/class/tacho-motor/motor1/stop_action
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - echo normal > /sys/class/tacho-motor/motor1/polarity
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - echo 500 > /sys/class/tacho-motor/motor1/speed_sp
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - echo run-forever > /sys/class/tacho-motor/motor1/command
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor1/state
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor1/speed_sp
ev3dev#5|2017-02-18 12:13:43 [main] INFO  e.actuators.motors.LargeMotorDemo - Large Motor is moving: true at speed 500
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG lejos.utility.Delay - Delay: 2000
ev3dev#5|2017-02-18 12:13:45 [main] DEBUG ev3dev.utils.Sysfs - echo stop > /sys/class/tacho-motor/motor1/command
ev3dev#5|2017-02-18 12:13:45 [main] INFO  e.actuators.motors.LargeMotorDemo - Stopped motors
```

Oracle JRE8, takes more or less 8 seconds to load all classes in memory and process with the Small CPU included with the EV3. Later, the library execute the following instructions to put the motor in the right status:

```
java -jar /home/robot/ev3dev-lang-java-all-0.4.0-SNAPSHOT.jar
ev3dev#5|2017-02-18 12:13:41 [main] INFO  e.actuators.motors.LargeMotorDemo - Starting motors on A
ev3dev#5|2017-02-18 12:13:41 [main] TRACE ev3dev.hardware.EV3DevPlatform - Detected platform: EV3BRICK
ev3dev#5|2017-02-18 12:13:41 [main] DEBUG e.a.motors.BaseRegulatedMotor - Detecting motor on port: outA
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: outA
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port0/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port1/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port2/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port3/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-port/port4/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port4/address
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG e.a.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG ev3dev.utils.Sysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
ev3dev#5|2017-02-18 12:13:42 [main] DEBUG lejos.utility.Delay - Delay: 500
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: outA
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/tacho-motor
ev3dev#5|2017-02-18 12:13:43 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor1/address
[... cut ...]
In relation to this point, this year, I will try to optimize. Another way to boost the java process execution is:

- execute in the EV3:

```
java -Xshare:dump
```

https://docs.oracle.com/javase/7/docs/technotes/guides/vm/class-data-sharing.html

This is something that I couldn't automate, but it is something that I will do.

- Another initiative to improve the execution is the creation of Runtime images with Java 9.

https://www.developer.com/java/data/what-is-a-custom-runtime-image-in-java-9.html

**Note:** This weekend I will upgrade this comment with LOGS from latest version of the library.

jabrena, 2018-03-24:
@RareScrap, Did you solve the question?

RareScrap, 2018-04-08:
I will try "java -Xshare:dump", thanks.

jabrena, 2018-04-10:
Take a look the metrics:

``` 
robot@ev3dev:~$ java -Xshare:dump
Allocated shared space: 268435456 bytes at 0x80000000
Loading classes to share ...
Loading classes to share: done.
Rewriting and linking classes ...
Rewriting and linking classes: done
Number of classes 1231
    instance classes   =  1170
    obj array classes  =    53
    type array classes =     8
Updating ConstMethods ... done. 
Removing unshareable information ... done. 
Scanning all metaspace objects ... 
Allocating RW objects ... 
Allocating RO objects ... 
Relocating embedded pointers ... 
Relocating external roots ... 
Dumping symbol table ...
Relocating SystemDictionary::_well_known_klasses[] ... 
Removing java_mirror ... done. 
mc  space:      2924 [  0.0% of total] out of      4096 bytes [ 71.4% used] at 0x80000000
rw  space:   2069528 [ 18.2% of total] out of   2072576 bytes [ 99.9% used] at 0x80001000
ro  space:   4617204 [ 40.7% of total] out of   4620288 bytes [ 99.9% used] at 0x801fb000
md  space:      3080 [  0.0% of total] out of      4096 bytes [ 75.2% used] at 0x80663000
od  space:   4655900 [ 41.0% of total] out of   4657152 bytes [100.0% used] at 0x80664000
total    :  11348636 [100.0% of total] out of  11358208 bytes [ 99.9% used]
robot@ev3dev:~$ 
robot@ev3dev:~$ time java -version
openjdk version "10" 2018-03-20
OpenJDK Runtime Environment (build 10+46-ev3)
OpenJDK Client VM (build 10+46-ev3, mixed mode, sharing)

real	0m4.650s
user	0m2.780s
sys	0m0.000s
robot@ev3dev:~$ time java HelloWorld 
Hello, World

real	0m4.129s
user	0m0.000s
sys	0m3.140s
robot@ev3dev:~$ 
``` 

The previous example uses the following configuration:

- Debian Stretch for EV3Dev
- OpenJDK 10

Juan Antonio

JakubVanek, 2018-04-10:
This depends on #441, #444, #445 and https://github.com/ev3dev-lang-java/installer/issues/1.

jabrena, 2018-04-23:
Moved question to next release.

JakubVanek, 2018-12-02:
In recent changes to the template project we achieved the following:
* Slf4j-simple is now used instead of logback, which eliminates quite a lot of the delay.
* Only a small jar is deployed, libraries are stored on the brick.
* We added some optimization flags to make the startup faster.
* We explored the usage AppCDS, but we haven't found a big benefit so far.

For now, I'm closing this issue. In the future, we might focus on the following issues:
* https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/445 - Manually preload some classes.
* https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/600 - Use cached libraries in a daemon process.

closed 2018-12-02 by JakubVanek