ev3dev-lang-java/ev3dev-lang-java #418: Replace Date object by a LocalDateTime builder

jabrena, 2018-03-16:
Example:
https://www.mkyong.com/java8/java-8-how-to-convert-string-to-localdate/

jabrena, 2018-04-23:
Possible solution to test:

https://docs.oracle.com/javase/8/docs/api/index.html?java/time/LocalDateTime.html

``` java
	/**
	 * Update Time values
	 */
	private void updateTime(){

		int timeStamp = ggaSentence.getTime();

		if (timeStamp > 0) {
			int hh = timeStamp / 10000;
			int mm = (timeStamp / 100) % 100;
			int ss = timeStamp % 100;

			if(date != null) {
				date = LocalDateTime.of(
						date.getYear(),
						date.getMonthValue(),
						date.getDayOfMonth(),
						hh,
						mm,
						sss);
			}
			
		}
	}

	/**
	 * Update Date values
	 */
	private void updateDate(){
		int dateStamp = rmcSentence.getDate();

		if(dateStamp > 0) {
			int dd = dateStamp / 10000;
			int mm = (dateStamp / 100) % 100;
			int yy = dateStamp % 100;

			date = LocalDateTime.of(yy, mm, dd, 0, 0, 0);
		}
	}
```

The new object follow Inmutable principles.

jabrena, 2018-04-23:
Fixed

closed 2018-04-23 by jabrena

jabrena, 2018-04-23:
Interesting logs related with the installation of rxtxJava

``` 
robot@ev3dev:~$ sudo apt-get install --yes --no-install-recommends librxtx-java
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree        
Reading state information... Done
The following NEW packages will be installed:
  librxtx-java
0 upgraded, 1 newly installed, 0 to remove and 352 not upgraded.
2 not fully installed or removed.
Need to get 158 kB of archives.
After this operation, 925 kB of additional disk space will be used.
Get:1 http://cdn-fastly.deb.debian.org/debian buster/main armel librxtx-java armel 2.2pre2+dfsg1-1 [158 kB]
Fetched 158 kB in 3s (51.9 kB/s) 
Selecting previously unselected package librxtx-java.
(Reading database ... 33917 files and directories currently installed.)
Preparing to unpack .../librxtx-java_2.2pre2+dfsg1-1_armel.deb ...
Unpacking librxtx-java (2.2pre2+dfsg1-1) ...
Setting up librxtx-java (2.2pre2+dfsg1-1) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Setting up ca-certificates-java (20170930) ...
Adding debian:IdenTrust_Commercial_Root_CA_1.pem
Adding debian:Equifax_Secure_CA.pem
Adding debian:QuoVadis_Root_CA_2.pem
Adding debian:AffirmTrust_Networking.pem
Adding debian:AffirmTrust_Premium.pem
Adding debian:Verisign_Class_1_Public_Primary_Certification_Authority_-_G3.pem
Adding debian:Comodo_Secure_Services_root.pem
Warning: there was a problem reading the certificate file /etc/ssl/certs/NetLock_Arany_=Class_Gold=_F?tan?s?tv?ny.pem. Message:
  /etc/ssl/certs/NetLock_Arany_=Class_Gold=_F?tan?s?tv?ny.pem (No such file or directory)
Adding debian:GlobalSign_Root_CA.pem
Adding debian:Hellenic_Academic_and_Research_Institutions_ECC_RootCA_2015.pem
Adding debian:Entrust.net_Premium_2048_Secure_Server_CA.pem
Adding debian:AddTrust_External_Root.pem
Adding debian:Network_Solutions_Certificate_Authority.pem
Adding debian:Actalis_Authentication_Root_CA.pem
Adding debian:AddTrust_Qualified_Certificates_Root.pem
Adding debian:EE_Certification_Centre_Root_CA.pem
Adding debian:OISTE_WISeKey_Global_Root_GB_CA.pem
Adding debian:TURKTRUST_Certificate_Services_Provider_Root_2007.pem
Adding debian:T-TeleSec_GlobalRoot_Class_3.pem
Adding debian:TWCA_Root_Certification_Authority.pem
Adding debian:AffirmTrust_Premium_ECC.pem
Adding debian:DST_Root_CA_X3.pem
Adding debian:TWCA_Global_Root_CA.pem
Adding debian:Go_Daddy_Root_Certificate_Authority_-_G2.pem
Adding debian:GlobalSign_Root_CA_-_R3.pem
Adding debian:WellsSecure_Public_Root_Certificate_Authority.pem
Adding debian:Taiwan_GRCA.pem
Adding debian:Chambers_of_Commerce_Root_-_2008.pem
Adding debian:QuoVadis_Root_CA.pem
Adding debian:Swisscom_Root_CA_1.pem
Adding debian:GeoTrust_Universal_CA.pem
Adding debian:Security_Communication_RootCA2.pem
Adding debian:Sonera_Class_2_Root_CA.pem
Adding debian:CFCA_EV_ROOT.pem
Adding debian:DST_ACES_CA_X6.pem
Adding debian:thawte_Primary_Root_CA_-_G3.pem
Adding debian:SwissSign_Gold_CA_-_G2.pem
Adding debian:Comodo_Trusted_Services_root.pem
Adding debian:COMODO_Certification_Authority.pem
Adding debian:DigiCert_Assured_ID_Root_G2.pem
Adding debian:Certplus_Root_CA_G2.pem
Adding debian:Certigna.pem
Adding debian:GeoTrust_Primary_Certification_Authority_-_G2.pem
Adding debian:GeoTrust_Global_CA.pem
Adding debian:AddTrust_Public_Services_Root.pem
Adding debian:Security_Communication_Root_CA.pem
[... 122 lines cut ...]
Adding debian:Starfield_Services_Root_Certificate_Authority_-_G2.pem
Adding debian:DigiCert_Global_Root_CA.pem
Adding debian:EC-ACC.pem
Adding debian:certSIGN_ROOT_CA.pem
done.
Processing triggers for ca-certificates (20161130+nmu1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...

done.
done.
Setting up openjdk-10-jre-headless:armel (10~46-2) ...
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/rmid to provide /usr/bin/rmid (rmid) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/keytool to provide /usr/bin/keytool (keytool) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/jjs to provide /usr/bin/jjs (jjs) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/pack200 to provide /usr/bin/pack200 (pack200) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/rmiregistry to provide /usr/bin/rmiregistry (rmiregistry) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/unpack200 to provide /usr/bin/unpack200 (unpack200) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/orbd to provide /usr/bin/orbd (orbd) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/servertool to provide /usr/bin/servertool (servertool) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/bin/tnameserv to provide /usr/bin/tnameserv (tnameserv) in auto mode
update-alternatives: using /usr/lib/jvm/java-10-openjdk-armel/lib/jexec to provide /usr/bin/jexec (jexec) in auto mode
FATAL -> Failed to fork.
```

JakubVanek, 2018-04-23:
Hmmm, `apt` has probably failed to finish installation the previous time it was run. That's why it is configuring cacerts + Java now.

`FATAL -> Failed to fork.` might indicate an out-of-memory error. I don't think ev3dev has a ulimit on the number of processes set.

`apt-get install -f` should do something useful.