ev3dev-lang-java/ev3dev-lang-java #423: Add a Fake System to simulate the whole ev3dev filesystem

jabrena, 2018-03-17:
In order to improve the Code coverage support, it is necessary the way that we are testing.
https://martinfowler.com/articles/mocksArentStubs.html

How to create folders
https://www.mkyong.com/java/how-to-create-directory-in-java/

jabrena, 2018-03-17:
Idea to remove the fake directory recursively:
http://www.baeldung.com/java-delete-directory

jabrena, 2018-03-24:
Adding Support for Touch Sensors:

```
ev3dev#43|2017-02-18 13:28:57 [main] TRACE ev3dev.hardware.EV3DevPlatform - Detected platform: EV3BRICK
ev3dev#43|2017-02-18 13:28:57 [main] TRACE ev3dev.hardware.EV3DevPlatform - Detected platform: EV3BRICK
ev3dev#43|2017-02-18 13:28:57 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting motors with port: in1
ev3dev#43|2017-02-18 13:28:58 [main] DEBUG ev3dev.utils.Sysfs - ls /sys/class/lego-sensor
ev3dev#43|2017-02-18 13:28:58 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor3/address
ev3dev#43|2017-02-18 13:28:58 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-sensor/sensor3/address
ev3dev#43|2017-02-18 13:28:58 [main] DEBUG ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor3/value0
ev3dev#43|2017-02-18 13:28:58 [main] INFO  examples.sensors.ev3.TouchSensorDemo - Iteration: 0
ev3dev#43|2017-02-18 13:28:58 [main] INFO  examples.sensors.ev3.TouchSensorDemo - Touch: 0
```

jabrena, 2018-03-24:
Added support.

Reviewed example from example repo:

``` java
package ev3dev.sensors.ev3;

import ev3dev.sensors.Battery;
import lejos.hardware.port.SensorPort;
import lejos.robotics.SampleProvider;
import lejos.utility.Delay;

public class TouchSensorDemo {

	//Robot Configuration
	private static EV3TouchSensor touch1 = new EV3TouchSensor(SensorPort.S1);
	
	//Configuration
	private static int HALF_SECOND = 500;
	
	public static void main(String[] args) {

		final SampleProvider sp = touch1.getTouchMode();
		int touchValue = 0;

        //Control loop
        final int iteration_threshold = 20;
        for(int i = 0; i <= iteration_threshold; i++) {

        	float [] sample = new float[sp.sampleSize()];
            sp.fetchSample(sample, 0);
            touchValue = (int)sample[0];

			System.out.println("Iteration: {}" + i);
			System.out.println("Touch: {}" + touchValue);
            
            Delay.msDelay(HALF_SECOND);
        }

        System.out.println(Battery.getInstance().getVoltage());
		
	}

}
```
https://github.com/ev3dev-lang-java/examples/blob/master/ev3dev-lang-java/src/main/java/ev3dev/sensors/ev3/TouchSensorDemo.java

JakubVanek, 2018-03-28:
IIRC this might be done using service providers, too.
Extra-advanced option would be to write a custom FUSE filesystem creating an illusion of the real brick.

jabrena, 2018-04-02:
Hi @JakubVanek, current solution is Java based and it is easy to maintain.
Current solution is based on a static file system simulation for the tests. 
For `ev3dev-lang-python` exist a similar solution:
https://github.com/ev3dev/ev3dev-lang-fake-sys

I think that current implementation is Ok for static sensor simulations.

jabrena, 2018-04-02:
Development complete.
It is necessary to review current implementation to try to refactor de code a bit.

jabrena, 2018-04-05:
Refactored code.

In the future, this development should be isolated in a new repo with new features like:

- Dynamic Sensor behaviours

But for the moment is Ok.

closed 2018-04-05 by jabrena