ev3dev-lang-java/ev3dev-lang-java #430: Add support for EV3Dev Stretch

jabrena, 2018-03-18:


jabrena, 2018-03-25:
The snapshots version of Debian Stretch are located here:
http://www.ev3dev.org/downloads/
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-03-16/

jabrena, 2018-03-25:
Burning a new SD for EV3 with a Debian Stretch Snapshot.

The tool Etcher is compatible with Debian Stretch Snapshot versions:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-03-16/snapshot-ev3dev-stretch-ev3-generic-2018-03-16.img.xz

jabrena, 2018-03-25:
Running SD Card on EV3.

jabrena, 2018-03-25:
Reviewing /sys/class path to detect some change:

```
Linux ev3dev 4.9.58-ev3dev-1.6.0-ev3 #1 PREEMPT Tue Oct 31 22:16:58 CDT 2017 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
robot@ev3dev:~$ ls
robot@ev3dev:~$ cd /sys/class/
robot@ev3dev:/sys/class$ ls
backlight  board-info  graphics     ieee80211  lego-sensor  mtd           pwm        scsi_device   sound        udc        watchdog
bdi        dc-motor    hidraw       input      mem          net           regulator  scsi_disk     spi_master   usbmon     zram-control
block      dma         i2c-adapter  leds       misc         phy           rfkill     scsi_generic  tacho-motor  vc
bluetooth  gpio        i2c-dev      lego-port  mmc_host     power_supply  rtc        scsi_host     tty          vtconsole
robot@ev3dev:/sys/class$ cd power_supply/
robot@ev3dev:/sys/class/power_supply$ ls
lego-ev3-battery
robot@ev3dev:/sys/class/power_supply$ cd lego-ev3-battery
robot@ev3dev:/sys/class/power_supply/lego-ev3-battery$ ls
current_now  device  power  scope  subsystem  technology  type  uevent  voltage_max_design  voltage_min_design  voltage_now
robot@ev3dev:/sys/class/power_supply/lego-ev3-battery$ 
```

jabrena, 2018-03-25:
**EV3 US Sensors:** No change

```
robot@ev3dev:/sys/class$ cd lego-sensor/
robot@ev3dev:/sys/class/lego-sensor$ ls
sensor0
robot@ev3dev:/sys/class/lego-sensor$ cd sensor0
robot@ev3dev:/sys/class/lego-sensor/sensor0$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat decimals 
1
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat driver_name 
lego-ev3-us
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat mode
US-DIST-CM
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat modes 
US-DIST-CM US-DIST-IN US-LISTEN US-SI-CM US-SI-IN US-DC-CM US-DC-IN
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
2550
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
362
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat value0
44
robot@ev3dev:/sys/class/lego-sensor/sensor0$ 
```

jabrena, 2018-03-25:
EV3 Gyro Sensor: Port?

```
robot@ev3dev:/sys/class/lego-sensor$ cd sensor1
robot@ev3dev:/sys/class/lego-sensor/sensor1$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat address 
ev3-ports:in1
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat commands 

robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat commands 

robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat driver_name 
lego-ev3-gyro
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat modes
GYRO-ANG GYRO-RATE GYRO-FAS GYRO-G&A GYRO-CAL TILT-RATE TILT-ANG
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat value0
0
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat mode
GYRO-ANG
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat value0
53
robot@ev3dev:/sys/class/lego-sensor/sensor1$ cat value0
-12
robot@ev3dev:/sys/class/lego-sensor/sensor1$  
```

jabrena, 2018-03-25:
**EV3 Touch Sensor:** Port?

```
robot@ev3dev:/sys/class/lego-sensor/sensor2$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat commands 

robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat modes 
TOUCH
robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat value0
0
robot@ev3dev:/sys/class/lego-sensor/sensor2$ cat address 
ev3-ports:in1
robot@ev3dev:/sys/class/lego-sensor/sensor2$ 
```

jabrena, 2018-03-25:
**EV3 Color Sensor:** Port?

``` 
robot@ev3dev:/sys/class/lego-sensor$ cd sensor4
robot@ev3dev:/sys/class/lego-sensor/sensor4$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@ev3dev:/sys/class/lego-sensor/sensor4$ cat modes 
COL-REFLECT COL-AMBIENT COL-COLOR REF-RAW RGB-RAW COL-CAL
robot@ev3dev:/sys/class/lego-sensor/sensor4$ cat value0
0
robot@ev3dev:/sys/class/lego-sensor/sensor4$ cat driver_name 
lego-ev3-color
robot@ev3dev:/sys/class/lego-sensor/sensor4$ cat decimals 
0
robot@ev3dev:/sys/class/lego-sensor/sensor4$ 
```

jabrena, 2018-03-25:
**EV3 Color Sensor:** Port?

```
robot@ev3dev:/sys/class/lego-sensor/sensor5$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat driver_name 
lego-ev3-ir
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat dec
cat: dec: No such file or directory
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat decimals 
0
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat mode
IR-PROX
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat modes
IR-PROX IR-SEEK IR-REMOTE IR-REM-A IR-S-ALT IR-CAL
robot@ev3dev:/sys/class/lego-sensor/sensor5$ cat value0 
23
robot@ev3dev:/sys/class/lego-sensor/sensor5$ 
```

jabrena, 2018-03-25:
**Lego Ports:** Ports?

``` 
robot@ev3dev:/sys/class/lego-port/port0$ ls
address  device  driver_name  ev3-ports:in1:ev3-uart-host  mode  modes  power  set_device  status  subsystem  uevent
robot@ev3dev:/sys/class/lego-port/port0$ cat address 
ev3-ports:in1
robot@ev3dev:/sys/class/lego-port/port0$ cat driver_name 
ev3-input-port
robot@ev3dev:/sys/class/lego-port/port0$ cat mode
auto
robot@ev3dev:/sys/class/lego-port/port0$ cat modes
auto nxt-analog nxt-color nxt-i2c other-i2c ev3-analog ev3-uart other-uart raw
robot@ev3dev:/sys/class/lego-port/port0$ 
```

jabrena, 2018-03-31:
EV3 Sensors:

- Battery: Ok
- Touch: Ok
- Color: Ok but it was necessary to upgrade the Switch Delay. [In Testing]

jabrena, 2018-03-31:
What is the usage of:

`root      9193  0.0  0.9   1768   548 ?        Ss   09:43   0:00 /usr/sbin/ldattach 29 /dev/tty_ev3-ports:in1`

JakubVanek, 2018-03-31:
It maintains the communication link between the UART sensor and the brick. You can even install the drivers on a Linux PC and use the sensors from there.
https://github.com/ev3dev/lego-linux-drivers-dkms

jabrena, 2018-03-31:
Improved logging support for Motors:

```
ev3dev#88|Starting motors on A
ev3dev#88|2018-03-31 13:15:58 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#88|2018-03-31 13:15:58 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/power_supply/lego-ev3-battery
ev3dev#88|2018-03-31 13:15:58 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/lego-ev3-battery
ev3dev#88|2018-03-31 13:15:58 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: EV3BRICK
ev3dev#88|2018-03-31 13:15:58 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outA
ev3dev#88|2018-03-31 13:15:58 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#88|2018-03-31 13:15:58 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#88|2018-03-31 13:15:59 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/lego-port
ev3dev#88|2018-03-31 13:15:59 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#88|2018-03-31 13:15:59 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port6/address
ev3dev#88|2018-03-31 13:15:59 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outC
ev3dev#88|2018-03-31 13:15:59 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outC.
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.hardware.EV3DevDevice - Skipped device
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 2:
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port4/address
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outA.
ev3dev#88|2018-03-31 13:16:00 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port4/address, ev3-ports:outA
ev3dev#88|2018-03-31 13:16:00 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.utils.Sysfs - echo tacho-motor > /sys/class/lego-port/port4/mode
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.hardware.EV3DevDevice - Retrieving devices in path: 
ev3dev#88|2018-03-31 13:16:00 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/tacho-motor
ev3dev#88|2018-03-31 13:16:01 [main] TRACE ev3dev.hardware.EV3DevDevice - Checking devices on: /sys/class/tacho-motor
ev3dev#88|2018-03-31 13:16:01 [main] TRACE ev3dev.hardware.EV3DevDevice - Device 1:
ev3dev#88|2018-03-31 13:16:01 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/tacho-motor/motor11/address
ev3dev#88|2018-03-31 13:16:01 [main] TRACE ev3dev.utils.Sysfs - value: ev3-ports:outA
ev3dev#88|2018-03-31 13:16:01 [main] TRACE ev3dev.hardware.EV3DevDevice - Port expected: ev3-ports:outA, actual: ev3-ports:outA.
ev3dev#88|2018-03-31 13:16:01 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor11/address, ev3-ports:outA
ev3dev#88|2018-03-31 13:16:01 [main] TRACE ev3dev.utils.Sysfs - echo reset > /sys/class/tacho-motor/motor11/command
:deployAndRun
```

jabrena, 2018-03-31:
Pending doubt about how set Volume for EV3 Speaker.

jabrena, 2018-03-31:
Solved doubt with the way to manage the volume.

jabrena, 2018-03-31:
https://stackoverflow.com/questions/953598/audio-volume-control-increase-or-decrease-in-java

jabrena, 2018-03-31:
https://coderanch.com/t/492931/java/Adjusting-master-volume-Windows-XP

jabrena, 2018-03-31:
https://www.ntu.edu.sg/home/ehchua/programming/java/J8c_PlayingSound.html

jabrena, 2018-03-31:
https://www.raspberrypi.org/forums/viewtopic.php?t=99920

jabrena, 2018-03-31:
http://www.javazoom.net/javalayer/javalayer.html

jabrena, 2018-03-31:
https://mvnrepository.com/artifact/javazoom/jlayer/1.0.1

jabrena, 2018-04-01:
http://killradio.org:8000/
http://sumerki.su:8000/

jabrena, 2018-04-05:
It is necessary to review Sound support to close this issue.

jabrena, 2018-04-06:
Internal refactor process for Sound support:

- Play samples with a pure Java solution
- Try to use javax.sound.* to increase/decrease volume (ev3dev Stretch not provide /volume endpoint)

jabrena, 2018-04-06:
It is necessary to create a Linear interpolation method to pass from 0-100 scale to db scale 0.0f -1.0f

**Theory:**
http://wwwprof.uniandes.edu.co/~gprieto/classes/compufis/interpolacion.pdf

``` java
    /**
     * x    y
     * -    -
     * 100  1.0
     * 30   x
     * 0.0  0
     *
     */
```

jabrena, 2018-04-07:
It is necessary to go back with changes in Sound class.

Executing the Regression tests the latest code although it run nice in Desktop in EV3 don´t run in the same way. It is necessary to interact with ALSA.

``` 
ev3dev#29|2018-04-03 02:58:07 [main] ERROR ev3dev.actuators.Sound - Unsupported control type: Volume
ev3dev#29|java.lang.IllegalArgumentException: Unsupported control type: Volume
ev3dev#29|	at java.desktop/com.sun.media.sound.AbstractLine.getControl(Unknown Source)
ev3dev#29|	at ev3dev.actuators.Sound.setVolume(Sound.java:169)
ev3dev#29|	at examples.SoundDemo.main(SoundDemo.java:27)
ev3dev#29|2018-04-03 02:58:07 [main] ERROR ev3dev.actuators.Sound - Unsupported control type: Volume
ev3dev#29|java.lang.IllegalArgumentException: Unsupported control type: Volume
ev3dev#29|	at java.desktop/com.sun.media.sound.AbstractLine.getControl(Unknown Source)
ev3dev#29|	at ev3dev.actuators.Sound.setVolume(Sound.java:169)
ev3dev#29|	at examples.SoundDemo.main(SoundDemo.java:27)
ev3dev#29|2018-04-03 02:58:10 [main] ERROR ev3dev.actuators.Sound - Unsupported control type: Volume
ev3dev#29|java.lang.IllegalArgumentException: Unsupported control type: Volume
ev3dev#29|	at java.desktop/com.sun.media.sound.AbstractLine.getControl(Unknown Source)
ev3dev#29|	at ev3dev.actuators.Sound.getVolume(Sound.java:217)
ev3dev#29|	at examples.SoundDemo.main(SoundDemo.java:28)
ev3dev#29|2018-04-03 02:58:10 [main] ERROR ev3dev.actuators.Sound - Unsupported control type: Volume
ev3dev#29|java.lang.IllegalArgumentException: Unsupported control type: Volume
ev3dev#29|	at java.desktop/com.sun.media.sound.AbstractLine.getControl(Unknown Source)
ev3dev#29|	at ev3dev.actuators.Sound.getVolume(Sound.java:217)
ev3dev#29|	at examples.SoundDemo.main(SoundDemo.java:28)
ev3dev#29|Exception in thread "main" java.lang.RuntimeException: Something goes wrong
ev3dev#29|	at ev3dev.actuators.Sound.getVolume(Sound.java:229)
ev3dev#29|	at examples.SoundDemo.main(SoundDemo.java:28)
Failed command ev3dev#29 with status 1: java -jar /home/robot/ev3dev-lang-java-2.1.0-SNAPSHOT-all.jar 
```

jabrena, 2018-04-07:
Fixed

jabrena, 2018-04-07:
CI Passed:
https://travis-ci.org/ev3dev-lang-java/ev3dev-lang-java/builds/363618251

jabrena, 2018-04-07:
Snapshot generated:
https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.1.0-SNAPSHOT/build.log
https://jitpack.io/#ev3dev-lang-java/ev3dev-lang-java/2.1.0-SNAPSHOT

jabrena, 2018-04-23:
Fixed

closed 2018-04-23 by jabrena