ev3dev-lang-java/ev3dev-lang-java #432: BrickPi 2.8 not recognized by installer on Raspberry Pi 3 B+

woolfel, 2018-03-20:
When I run the installer with the newest Pi 3 B+ and BrickPi 2.8, I get platform not recognized.

jabrena, 2018-03-20:
Can you paste the log traces from your last example. In my case, I received the board but I could install in a brickpi or brickpi 3.

woolfel, 2018-03-21:
which version of Raspbian are you using? I was trying to use Stretch, but that had problems. I double checked my BrickPi on Pi 3B with Wheezy and it worked fine with BrickPiJava. I'm thinking Stretch is the issue.

jabrena, 2018-03-21:
Hi @woolfel, 

At the moment, the whole project continue using Debian Jessie to interact with EV3Dev.
Debian Stretch has some API changes that I am trying to clarify in order to provide a new version of the library. Can you paste here the log messages with Stretch please?

Do you have time to test with Jessie?

Cheers

Juan Antonio

jabrena, 2018-03-21:
At the moment, I am adding `Unit testing` in the whole project in order to migrate to Debian Stretch easily.

woolfel, 2018-03-21:
I was going to test with Jessie today, since it looked like Stretch is DOA for BrickPi right now. I realized that last night when Stretch just wouldn't cooperate.

jabrena, 2018-03-21:
Hi @woolfel, 

I have a question for you about the support about `PivotPi`. Is it easy to integrate in BrickPi 2/3 in some way? What is your opinion? I have a real curiosity because you have a real expertise with that device.

Cheers

Juan Antonio

jabrena, 2018-03-21:
Take your time mate. I will create an Issue about the POC later.

Cheers

woolfel, 2018-03-23:
Update for you. Last night tried multiple times to get Wheezy and Jessie to boot on Pi 3 B+ without any luck. I was able to boot with Stretch and get BrickPi working with the python drivers. It did have some issues with BrickPi 2.8 and would give serial read errors, so I think it's a Raspbian issue.  The BrickPi 3 should arrive tomorrow, so I will try Pi 3B with BrickPi3 on Jessie this weekend.

woolfel, 2018-03-24:
I downloaded the official image and tried it on my Pi 3B with BrickPi 2.8. The boot up got stuck on 
[image]

the BrickPi 3 is supposed to arrive today, so I will test it with that tomorrow

jabrena, 2018-03-24:
Upss, but it is normal. It is a new board.
We should open an issue on ev3dev to review in detail.
I will open an issue about the case.

Juan Antonio

jabrena, 2018-03-24:
Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/431

jabrena, 2018-03-24:
I am going to close this issue. We will follow this support using this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/431

closed 2018-03-24 by jabrena