ev3dev-lang-java/ev3dev-lang-java #439: Release OpenJDK 9 JRE

jabrena, 2018-03-25:


JakubVanek, 2018-03-25:
https://github.com/ev3dev-lang-java/openjdk-ev3/releases/tag/v0.4.0 for Stretch

jabrena, 2018-03-25:
Testing on EV3 Brick

jabrena, 2018-03-25:
```
wget https://github.com/ev3dev-lang-java/openjdk-ev3/releases/download/v0.4.0/jre-ev3.zip
7z x jre-ev3.zip
robot@ev3dev:~$ cd jre-ev3
robot@ev3dev:~/jre-ev3$ ls
bin  conf  legal  lib  release
robot@ev3dev:~/jre-ev3$ cd bin/
robot@ev3dev:~/jre-ev3/bin$ ls
java  javac  javah  jdb  jrunscript  jshell  keytool  rmid  rmiregistry  serialver
robot@ev3dev:~/jre-ev3/bin$ ./java -version
openjdk version "9.0.4"
OpenJDK Runtime Environment (build 9.0.4+12)
OpenJDK Client VM (build 9.0.4+12, mixed mode)
robot@ev3dev:~/jre-ev3/bin$ vi HelloWorld.java
robot@ev3dev:~/jre-ev3/bin$ ./javac HelloWorld.java 
robot@ev3dev:~/jre-ev3/bin$ ls
HelloWorld.class  HelloWorld.java  java  javac  javah  jdb  jrunscript  jshell  keytool  rmid  rmiregistry  serialver
robot@ev3dev:~/jre-ev3/bin$ j      
-bash: j: command not found
robot@ev3dev:~/jre-ev3/bin$ ./java HelloWorld 
Hello, World
robot@ev3dev:~/jre-ev3/bin$ 
```

jabrena, 2018-03-25:
Testing finished with success on Iteration 1.
I will automate this version and I will execute more tests from the template but it run in a better way that the compilation with OpenJDK 10.

jabrena, 2018-03-25:
Closing this issue.
Next step will involve other components:

- Installer: To automate the installation on EV3 Brick
- Template project: To install JRE 9 and deploy examples

closed 2018-03-25 by jabrena