ev3dev-lang-java/ev3dev-lang-java #441: Jigsaw modules

JakubVanek, 2018-03-25:
JDK 9 introduces Jigsaw with JMODs. Support should be added to this library.

It might be even possible to modularize this library into multiple Jigsaw modules. OpenJDK source itself is a good example of modularization.

jabrena, 2018-03-25:
Totally agree with you.

Currently, I am reading this about this Java topic:
https://github.com/java9-modularity/examples
https://javamodularity.com/

jabrena, 2018-03-26:
Yes, the idea to create modules, is quite interesting and I think that it could be pretty interesting.
For the project, the first idea to implement Jigsaw, could be to decouple the API from the implementation. At the moment, we use on library for 4 platforms.

Using Modules, we could have one API and 4 implementations. In that way, we will avoid this kind of stuff:

https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/src/main/java/ev3dev/sensors/Battery.java

``` java
private void init(){
        LOGGER.debug("Init sensor");

        //TODO Create separator variable for the whole project
        BATTERY_PATH = ROOT_PATH + "/" + BATTERY;
        final EV3DevPlatform platform = this.getPlatform();
        if(platform.equals(EV3DevPlatform.EV3BRICK)) {
            BATTERY_PATH_LOCAL += BATTERY_PATH + "/" + BATTERY_EV3;
        } else if(platform.equals(EV3DevPlatform.PISTORMS)) {
            BATTERY_PATH_LOCAL += BATTERY_PATH + "/" + BATTERY_PISTORMS;
        } else if(platform.equals(EV3DevPlatform.BRICKPI)) {
            BATTERY_PATH_LOCAL += BATTERY_PATH + "/" + BATTERY_BRICKPI;
        } else if(platform.equals(EV3DevPlatform.BRICKPI3)) {
            BATTERY_PATH_LOCAL += BATTERY_PATH + "/" + BATTERY_BRICKPI3;
        } else if(platform.equals(EV3DevPlatform.UNKNOWN)) {
            throw new RuntimeException("Platform Unknown");
        }
    }
```

Or the scenarios where some platform don't support one feature:

``` java
/**
	 * Returns the current of the battery in amps.
	 * @return current
	 */
	public float getBatteryCurrent() {
	    //TODO Review units
    	if(this.getPlatform().equals(EV3DevPlatform.EV3BRICK)){
    		return Sysfs.readFloat(BATTERY_PATH + "/" + BATTERY_EV3 + "/" +  CURRENT);
    	}else {
            LOGGER.warn("This method is not available for {} & {}", EV3DevPlatform.PISTORMS, EV3DevPlatform.BRICKPI);
            return -1f;
        }
	}
```

Docs about ServiceLoader:
https://docs.oracle.com/javase/9/docs/api/java/util/ServiceLoader.html

jabrena, 2018-03-26:
Later, once the library has different implementation decoupled, it could be possible to have other modules for sensors and other stuff.

As you say, using JLink is the key element to create minimal runtime images. :)

jabrena, 2018-03-26:
Take a look this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/413

appear this point. :)

jabrena, 2018-04-15:
https://www.logicbig.com/tutorials/core-java-tutorial/java-se-api/service-loader.html
https://docs.oracle.com/javase/tutorial/ext/basics/spi.html

jabrena, 2018-05-07:
I am going to move this issue to next release.
In this iteration with the dual compatibility, I don´t want to add more changes in the library.

JakubVanek, 2018-08-13:
Hmmm, AppCDS for user modules is supported from Java 11 onwards.

*before:* http://openjdk.java.net/jeps/310:
> In this release, CDS cannot archive classes from user-defined modules (such as those specified in --module-path). We plan to add that support in a future release.

*and now:* https://bugs.openjdk.java.net/browse/JDK-8194812

closed 2019-08-18 by jabrena