ev3dev-lang-java/ev3dev-lang-java #442: Decouple logging

JakubVanek, 2018-03-25:
It would be a great idea to remove the dependency on logging features: https://github.com/ev3dev-lang-java/openjdk-ev3/issues/5#issuecomment-356647407

I don't remember the exact numbers, but I remember the speedup was quite interesting (60 -> 20 seconds, maybe?).

JakubVanek, 2018-03-25:
relates to #441

jabrena, 2018-03-25:
Tomorrow, I will explain the reasons to use current dependency.
Good point to discuss @JakubVanek 

Cheers

jabrena, 2018-03-26:
Good Morning @JakubVanek, current development for this library requires 2 dependencies:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/develop/build.gradle

``` groovy
dependencies {
    compile("org.slf4j:slf4j-api:1.7.25")
    compile("com.github.ev3dev-lang-java:lejos-commons:0.7.0")

    annotationProcessor("org.projectlombok:lombok:1.16.20")
    testCompile("org.projectlombok:lombok:1.16.20")

    testCompile("ch.qos.logback:logback-classic:1.2.3")

    testCompile("commons-io:commons-io:2.5")
    
    testCompile("junit:junit:4.12")
    testCompile("org.hamcrest:hamcrest-all:1.3")
}
```

jabrena, 2018-03-26:
How to use a concrete logging implementation?

If you observe the template example:
https://github.com/ev3dev-lang-java/template_project_gradle/blob/master/build.gradle

```
dependencies {
    compile("ch.qos.logback:logback-classic:1.2.3")
    compile("com.github.ev3dev-lang-java:ev3dev-lang-java:0.7.0")
    compile("com.github.ev3dev-lang-java:lejos-navigation:0.2.0")
}
```

The example add Logback and it is possible to configure the logging settings:

https://github.com/ev3dev-lang-java/template_project_gradle/blob/master/src/main/resources/logback.xml

``` xml
<?xml version="1.0" encoding="UTF-8"?>
<configuration>

    <jmxConfigurator />

    <appender name="STDOUT" class="ch.qos.logback.core.ConsoleAppender">
        <encoder class="ch.qos.logback.classic.encoder.PatternLayoutEncoder">
            <Pattern>
                %d{yyyy-MM-dd HH:mm:ss} [%thread] %-5level %logger{36} - %msg%n
            </Pattern>
        </encoder>
    </appender>

    <logger name="ch.qos.logback.core" level="OFF" />
    <logger name="ev3dev.hardware" level="DEBUG" />

    <root level="DEBUG">
        <appender-ref ref="STDOUT" />
    </root>

</configuration>
```

JakubVanek, 2018-03-26:
Well, the less classes we load the better.

I'll use an extreme case (impossible to reach on Linux) of how fast things can be.

When I need to write a program for a robot, I go for [EV3RT RTOS](http://ev3rt-git.github.io/about/). The boot process really takes only about _five seconds_ and if you use the dynamic loader, you can restart programs even faster. Yes, you need to know C (or C++) and μITRON and EV3RT APIs, but then you're fine. And it supports logging via Port S1 UART, Bluetooth and LCD (though I quite miss USB debug functionality and an on-brick shell).

The current ev3dev boot is about an order of magnitude slower, but that's OK. It would just be very nice thing if each JVM startup wouldn't take a long time.

Maybe we could even use native code and JNI/JNA to speed things up. Maybe this would make for another project (let's say, ev3dev-lang-jnijava).

JakubVanek, 2018-03-26:
Removing just Logback looks promising, that might be sufficient.

jabrena, 2018-03-26:
`Logback` is a User Decision. 
The library only provides the infrastructure for logging but not force any particular implementation.

Why I like Logback?
https://logback.qos.ch/reasonsToSwitch.html

Some iterations ago, I could integrate Robot's log messages on ELK but I need to publish the whole configuration.
https://www.elastic.co/elk-stack

Analyse Log execution from multiple robots is a nice feature, but not common, I recognise :)
But for general, logging is a good practice for any Java developer and this is one of the Goals of this project. Help people to use Java in General and Robotics with Java in particular.

closed 2018-03-28 by jabrena