ev3dev-lang-java/ev3dev-lang-java #443: Externalise EV3Dev constants to offer support for multiple EV3Dev Distros (Jessie & Stretch)

jabrena, 2018-03-28:


JakubVanek, 2018-03-28:
This might be done using Jigsaw and service providers.

jabrena, 2018-03-30:
Yes, but I was thinking in the properties to have support for jessie and stretch.

**Example:**

Jessie:
/sys/class/power_supply/legoev3-battery

Stretch:
/sys/class/power_supply/lego-ev3-battery

Using 2 property files it is possible to load the right properties:

```
robot@ev3dev:~$ cat /etc/os-release
PRETTY_NAME="ev3dev-stretch"
NAME="ev3dev-stretch"
ID=ev3dev
ID_LIKE=debian
HOME_URL="http://www.ev3dev.org"
SUPPORT_URL="http://www.ev3dev.org/support"
BUG_REPORT_URL="https://github.com/ev3dev/ev3dev/issues"
```

jabrena, 2018-04-14:
Reviewing the UML design the right class candidate to add a new method `getDistro` is **EV3DevPlatforms.java**

The new method could return:

- JESSIE
- STRETCH

The possible values will be stored in a Enum

If the class doesn´t detect one of the distro will raise a RuntimeException.

jabrena, 2018-04-14:
Later, it is necessary to analyse how to load in a property file all elements.

Related links:
https://stackoverflow.com/questions/12473537/naming-convention-for-properties-in-property-file

jabrena, 2018-04-14:
How to reset a static field? (Reflection problem)
https://stackoverflow.com/questions/26234742/resetting-static-fields-for-junit-tests
https://caffinc.github.io/2015/12/static-final-java-junit-test/
https://blog.box.com/blog/unit-testing-with-static-variables/

jabrena, 2018-04-14:
It is necessary to identify the changes between Jessie Support & Stretch Support.

- [ ] **Sound:** In this case, exist 2 different implementations. The main problem in Stretch, is the lack of a volume endpoint.
- [ ] LED: Different paths. Unify paths and the way to build the final paths.
- [ ] EV3DevPlatforms: Platform, Motor Ports & Sensor Ports
- [ ] BaseSensor: SwitchDelay
- [ ] Battery:
- [ ] EV3Key: event path
- [ ] FakeBattery:
- [ ] FakeLegoSensor:
- [ ] FakeUnregulatedMotor:
- [ ] FakeRegulatedMotor:
- [ ] FakeLargeRegulatedMotor:

Some tests:

- [ ] ShellTest
- [ ] EV3DevPlatformsTest
- [ ] EV3SensorDeviceTest

jabrena, 2018-05-04:
I will continue with the changes.

jabrena, 2018-08-05:
It is necessary to review some classes that raises some exceptions in Travis.

jabrena, 2018-10-28:
Added support.

jabrena, 2018-10-28:
Finally, the library in the last version only support Debian Stretch. Doesn't any value to maintain 2 distros.

closed 2018-10-28 by jabrena