ev3dev-lang-java/ev3dev-lang-java #444: Use preinstalled package instead of fat JAR

JakubVanek, 2018-03-28:
We should install the libraries in the installer and then only update them when needed. AppCDS then could speed up the loading phase.
Fat JARs are IMO a slow overkill.

jabrena, 2018-04-07:
Using the installer we could install the main library and others (dependencies) and set in the classpath, (old fashion way to interact with the JVM).

I need to review how to exclude some depdencies from the `fatjar` with gradle.

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/445

JakubVanek, 2018-04-10:
See https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/445#issuecomment-380164768 about how it could be done.

JakubVanek, 2018-04-10:
Blocked by #441.

jabrena, 2018-04-16:
It is necessary to know how to exclude some dependency.
http://unethicalblogger.com/2015/07/15/gradle-goodness-excluding-depends-from-shadow.html

JakubVanek, 2018-04-16:
This is a part of `build.gradle` from my removal-of-logging-libraries experiment:
```gradle
dependencies {
    compile files('../out/fakelog-0.1.0.jar', '../out/lejos-commons-0.6.0.jar')
    runtime files(  'libs/fakelog-0.1.0.jar',   'libs/lejos-commons-0.6.0.jar')

    testCompile("org.projectlombok:lombok:1.16.16",
                "ch.qos.logback:logback-classic:1.2.3",
                "junit:junit:4.12",
                "org.hamcrest:hamcrest-all:1.3")
}
```

jabrena, 2018-04-16:
hehehe, you don't like logging... :D 

I will test it. :)

JakubVanek, 2018-04-16:
Hmm, particularly I have a not-invented-here syndrome and premature optimizations mania... :smile: 

As I've written earlier, I didn't like how it was slowing down the class loading phase.

jabrena, 2018-05-05:
Hi @JakubVanek, 

In the Template project, we could have 2 ways to use the Gradle tasks:

- FatJar
- Preloading packages

In the second option, we could have some gradle tasks to install in the brick the dependencies used:

- ev3dev-lang-java-x.y.z.jar
- slf4j-api-1.7.25.jar
- lejos-commons-x.y.z.jar

that jars could be downloaded in some path:

/home/robot/installer/template

later, export some variable to use later in the classpath and run.

Is it Ok for you?

We could test the performance improvement.

Juan Antonio

jabrena, 2018-05-05:
Hi @JakubVanek,

The idea that you proposed:

``` gradle
dependencies {
    compile files('../out/fakelog-0.1.0.jar', '../out/lejos-commons-0.6.0.jar')
    runtime files(  'libs/fakelog-0.1.0.jar',   'libs/lejos-commons-0.6.0.jar')

    testCompile("org.projectlombok:lombok:1.16.16",
                "ch.qos.logback:logback-classic:1.2.3",
                "junit:junit:4.12",
                "org.hamcrest:hamcrest-all:1.3")
}
``` 

it is necessary to be reviewed from a programatic perspective. It is in the moment of the Jar generation the point to remove any dependency. I will review other alternatives on internet.

JakubVanek, 2018-05-05:
Prealpha launcher version:
`/opt/ev3dev-lang-java/create.sh`:
```
#!/bin/bash

if [ "$#" -ne "2" ]; then
    echo "Usage: " 1>&2
    echo "$0 \"</home/robot/program.sh>\" \"</home/robot/jars/program.jar>\" " 1>&2
fi

sed "s/%%TEMPLATE%%/$2/" "/opt/ev3dev-lang-java/launcher.sh" > "$1"
```
`/opt/ev3dev-lang-java/launcher.sh`:
```
#!/bin/bash

for f in /opt/ev3dev-lang-java/config.d/*; do
    source $f
done

"$JAVA_EXE" "$JAVA_FLAGS" "$JAVA_EXTRA" -jar "%%TEMPLATE%%" "$@"
```

and a part of the installer:
```
    echo "#!/bin/bash"                 >"/opt/ev3dev-lang-java/config.d/jdkver.sh"
    echo "JDK_EXE=\"$JAVA_REAL_EXE\"" >>"/opt/ev3dev-lang-java/config.d/jdkver.sh"
```

Additional configuration files will just append args to JAVA_FLAGS (extending class or module path, selecting AppCDS cache file etc.).

jabrena, 2018-05-19:
Moving out of the release

jabrena, 2018-05-20:
Another way to exclude a dependency is this example:
https://github.com/ev3dev-lang-java/examples/blob/master/opencv/build.gradle

``` gradle
dependencies {
    compileOnly("nu.pattern:opencv:2.4.9-7")
    //compileOnly("org.openpnp:opencv:3.2.0-0")

    compile("org.slf4j:slf4j-api:1.7.25")
    compile("ch.qos.logback:logback-classic:1.2.3")

    annotationProcessor("org.projectlombok:lombok:1.16.20")
    compileOnly("org.projectlombok:lombok:1.16.20")
    testCompileOnly("org.projectlombok:lombok:1.16.20")

    testCompile 'junit:junit:4.12'
}
```

with `compileOnly` you exclude any dependency to the process to generate the fatjar.
The same idea is used for `Lombok`.

```
    compile("org.slf4j:slf4j-api:1.7.25")
    compile("ch.qos.logback:logback-classic:1.2.3")
```

This dependencies are clear candidate to move this feature.

JakubVanek, 2018-12-02:
Delivered.

closed 2018-12-02 by JakubVanek