ev3dev-lang-java/ev3dev-lang-java #445: Preload essential classes

JakubVanek, 2018-03-28:
Essential leJOS classes should be preloaded. This would prevent the issue with having to wait a long time after first leJOS class use.
leJOS used a wrapper main to perform maintenance tasks, this might have caused the classes to get preloaded.
https://sourceforge.net/p/lejos/ev3/code/ci/master/tree/ev3classes/src/lejos/internal/ev3/EV3Wrapper.java
Idk if this is possible with Jigsaw.

jabrena, 2018-04-07:
Now we have the possibility to install OpenJDK 10 JRI from the installer.
Do you think that it is possible to download some dependencies in every brick and preload?

- How to preload that classes?
- How to add some path to preload some jars?
- What is the relation with: java -Xshare:dump?

jabrena, 2018-04-07:
From this question:
https://stackoverflow.com/questions/4692076/speed-up-application-start-by-adding-own-application-classes-to-classes-jsa

> As of Java 8u40 (and Embedded Java 8u51), Java now supports Application Class Data Sharing (AppCDS) (ie your own classes in the shared archive). On our embedded java, we've found a startup improvement of >40%! Pretty awesome for almost no work on our part...
https://blogs.oracle.com/thejavatutorials/jdk-8u40-released

jabrena, 2018-04-07:
https://docs.oracle.com/javase/8/docs/technotes/tools/unix/java.html

```
-Xshare:mode
Sets the class data sharing (CDS) mode. Possible mode arguments for this option include the following:

auto
Use CDS if possible. This is the default value for Java HotSpot 32-Bit Client VM.

on
Require the use of CDS. Print an error message and exit if class data sharing cannot be used.

off
Do not use CDS. This is the default value for Java HotSpot 32-Bit Server VM, Java HotSpot 64-Bit Client VM, and Java HotSpot 64-Bit Server VM.

dump
Manually generate the CDS archive. Specify the application class path as described in "Setting the Class Path".

You should regenerate the CDS archive with each new JDK release.
```

jabrena, 2018-04-07:
In the folder /home/installer/ we could create a new folder named libraries to add store the required libraries but how to set the libraries to execute later -Xshare:dump?

JakubVanek, 2018-04-10:
Hmm, why `/home/installer`? We can create `/opt/ev3dev-lang-java/lib` with library modules and add that to module path (see later). We should also create a Java wrapper (`/opt/ev3dev-lang-java/bin/java`) for supplying our own arguments (like `-Xshare:on`, `-XX:+UseAppCDS`, module path and so on).

To update cache, we can create another wrapper `/opt/ev3dev-lang-java/bin/java-refresh`. That would run a dump like there: https://mjg123.github.io/2017/10/04/AppCDS-and-Clojure.html. It needs to be done only once after installing the libraries or updating JVM.

jabrena, 2018-04-11:
Yes, once we have the installer upgraded, we will execute this Spike to review results.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/276

jabrena, 2018-05-12:
https://github.com/CodeFX-org/demo-java-10/blob/master/app-cds.sh

jabrena, 2018-05-19:
Moving out of the release

jabrena, 2018-06-22:
Good video about this matter:
https://www.youtube.com/watch?v=NE9euFxr2-A&feature=youtu.be

JakubVanek, 2018-12-02:
I'm closing this issue, as this probably was a problem even on leJOS. Replacement of Logback with Slf4j also lessened the impact of this issue.

closed 2018-12-02 by JakubVanek