ev3dev-lang-java/ev3dev-lang-java #453: Consider merging with leJOS

JakubVanek, 2018-04-01:
https://github.com/ev3dev/lms2012-compat
LMS2012 kernel compatibility layer makes some of the kernel functionality in the stock firmware available to ev3dev. This makes it easier to port the original leJOS to this platform.

I've succeeded with running leJOS menu, but without any input or sound output - these use interfaces that are not emulated. I've tried running some Pilot tests but I haven't succeeded with that. The EV3MotorPort kernel interfacing subclass was probably looping forever in startSynchronization. I didn't want to add more hacks today. It might be just that the compatibility layer is a more robust implementation than the original driver.

However, I think it is worth thinking about bringing all the features of leJOS to ev3dev-lang-java. We could gradually replace the compatibility layer-backed implementation of lejos.internal with our own ev3dev-specific one.

There could be two modes of operation:
* LeJOS with menu - the current model, AFAIK this makes RMI remote control available.
* Only LeJOS libraries - only take leJOS ev3classes and launch programs directly from brickman.

Personally, I like the first scenario more.

jabrena, 2018-04-02:
I replied here:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/413

jabrena, 2018-04-02:
It could be great to execute this idea, but we have to schedule the process.

closed 2018-04-15 by JakubVanek