ev3dev-lang-java/ev3dev-lang-java #464: Platform detection error with installer.sh

snaicken, 2018-04-07:
The platform.sh file executes the following to identify an EV3 device:

```
#1. Detect platform
#1.1 Detect if the brick is not using latest kernel
if [ -d "/sys/class/power_supply/lego-ev3-battery" ]; then
  echo "The user has a EV3 Brick"
  PLATFORM=$EV3
```

However, the appropriate system device is found elsewhere.

```
robot@ev3dev:~/installer$ uname -a
Linux ev3dev 4.4.87-22-ev3dev-ev3 #1 PREEMPT Sat Sep 9 14:45:55 CDT 2017 armv5tejl GNU/Linux
robot@ev3dev:~/installer$ ls /sys/class/power_supply/legoev3-battery
current_now  power  subsystem   type    voltage_max_design  voltage_now
device       scope  technology  uevent  voltage_min_design
```

jabrena, 2018-04-07:
Hi mate, 

Try to use the installer from master, you have Debian Jessie but in Develop branch, we are preparing the next release with Debian Stretch support.

In Debian Stretch, some ev3dev paths changed and this is one of them.

```
cd /home/robot
mkdir installer
cd installer
wget -N https://raw.githubusercontent.com/ev3dev-lang-java/installer/master/installer.sh
chmod +x installer.sh
sudo ./installer.sh help
sudo ./installer.sh
```

Cheers

snaicken, 2018-04-07:
Thanks so much for the quick response.  master version is working fine, but I had to change this line in the installer.sh, as it points to the develop branch.

`local domain="https://raw.githubusercontent.com/ev3dev-lang-java/installer/develop/modules"`

Cheers.

closed 2018-04-07 by jabrena