ev3dev-lang-java/ev3dev-lang-java #476: Certificate project with latest Snapshot Debian Stretch image for EV3 Brick(snapshot-ev3dev-stretch-ev3-generic-2018-03-16)

jabrena, 2018-04-14:
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/
https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2018-03-16/snapshot-ev3dev-stretch-ev3-generic-2018-03-16.img.xz

JakubVanek, 2018-04-14:
Hi, what do you mean by certification? The JCK tests?

jabrena, 2018-04-14:
Nope. It is the typical issue that I test the main library with the device.
In this iteration I will certificate for EV3 Brick for Debian Stretch.

jabrena, 2018-04-20:
Testing with better version of current Debian Stretch support:
https://jitpack.io/com/github/ev3dev-lang-java/ev3dev-lang-java/2.2.1-SNAPSHOT/build.log

jabrena, 2018-04-21:
Small detail about new logs:

```
ev3dev#60|2018-04-21 13:06:02 [main] DEBUG ev3dev.utils.Shell - Command: cat /etc/os-release | grep jessie | wc -l
ev3dev#60|2018-04-21 13:06:03 [main] DEBUG ev3dev.hardware.EV3DevDistros - Debian Stretch detected
ev3dev#60|2018-04-21 13:06:04 [main] DEBUG ev3dev.hardware.EV3DevFileSystem - Root Path: /sys/class
ev3dev#60|2018-04-21 13:06:04 [main] INFO  e.a.lego.motors.BaseRegulatedMotor - Configuring motor connected on Port: A
ev3dev#60|2018-04-21 13:06:04 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Detecting motor on port: ev3-ports:outA
ev3dev#60|2018-04-21 13:06:04 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/lego-port/port4/address, ev3-ports:outA
ev3dev#60|2018-04-21 13:06:04 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Setting port in mode: tacho-motor
ev3dev#60|2018-04-21 13:06:05 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected device on path: /sys/class/tacho-motor/motor5/address, ev3-ports:outA
ev3dev#60|2018-04-21 13:06:06 [main] DEBUG e.a.lego.motors.BaseRegulatedMotor - Motor ready to use on Port: A
```