ev3dev-lang-java/ev3dev-lang-java #478: Create a mini gui library an a configuration tool

JakubVanek, 2018-04-15:
Hi,

Some time ago, I've developed a standalone configuration editing tool for leJOS that used only the on-brick buttons.

https://gitlab.com/robovanek/robo2015-ss/tree/master/program/

* cfgtool is the configuration app
* microgui is the library that cfgtool uses
* program/src/main/Settings.java contains the client-side settings importer/exporter

I've also developed a compatibility layer (IIRC EV3LCD & others Swing implementation) that allows running the config tool directly on PC.

I can imagine that some people might find this program (and its library) helpful. We might want to integrate it into ev3dev-lang-java.

Some old development screenshots:
[image]
[image]
[image]

Current version running on Swing (it doesn't have on-screen keyboard, native input is supported):
[image]

The configurator is built on top of a `microgui` library that could be used by all programs. It is really not the best thing I've created, architectural changes would be welcome. I think there were some issues with how the window stack is implemented. Also the widgets are implemented in cfgtool, they should be refactored out.

I'm the only author of cfgtool and microgui. Matej Kripner was my team member and he developed parts of the main program, that's why he's mentioned as an author. I'm OK with relicensing my code.

Cheers

JakubVanek, 2018-04-15:
Hmmm, there was a time when I was writing something like a blog. This post describes the circumstances of this program's birth: http://www.jakubvanekpc.jecool.net/?p=410

jabrena, 2018-04-15:
I have to create the blog section with jekyll. Exist many good articles in the air:

- OpenJDK adventure
- ARM5 Support
- EV3Dev movement model vs LeJOS movement model
- Evolutionary Architecture
- ...

In terms of UX, any collaboration is always welcome but you are helping so much in **JVM** area but if you have energy, Ok :)

EV3 Brick has support for LCD. I am not very happy with the support but it is running. In ev3dev side, Python guys integrate with this library: https://github.com/ev3dev/grx
I think that in EV3Dev side, this is the way to evolve the Display infrastructure in the low level and later add the Java layer and finally the application layer.

Another idea could be try to run LeJOS menu on EV3Dev and use Installer to replace or run at least. :)

JakubVanek, 2018-04-15:
Well, these projects implement widgets and input stack. For the EV3 display they rely on leJOS internal LCD classes.

In the future, going The EV3Dev Way would be preferred. We'd remove redundant code from java gui code and add glue layers for EV3Dev.

jabrena, 2018-04-15:
I can´t say **No**, you are in the infrastructure side. :)

JakubVanek, 2018-04-15:
> * EV3Dev movement model vs LeJOS movement model

If this is about the ev3dev acceleration model vs leJOS acceleration model...

I don't know the difference. I _think_ that ev3dev relies purely on timing, whether lejos _maybe_ is using the currently travelled distance as an argument for the velocity function.

@gloomyandy can I ask about what the difference is about?

JakubVanek, 2018-04-29:
I thought it might be useful to have these features:
* a quicker way how to create GUIs on EV3, including a built-in on-screen keyboard,
  * basically a mini widget toolkit + window stack + human input + graphics output
* a possibility to run on-brick GUIs on PC (probably an edge case)
* an on-brick settings editor when you can't have a laptop next to the brick (maybe a FLL-like scenario)
  * modifying PID parameters, robot behaviour etc.

The story behind only the original tool:
I was bored and I was curious, whether I was able to write a configuration editor. Then I won't have to connect the brick to the laptop and edit it in a text editor.

Looking back at my old configuration file, I've exported these things to it:
* sensor and motor ports
* motor parameters (speed, acceleration, direction)
* physical robot parameters (like wheel diameter and track width)
* regulation parameters (PID)
* timing parameters (how long to wait after stop, etc.)
* sensor parameters (reflected light thresholds, etc.)
* playing field size and other movement parameters
* logic configuration (how far we trust the ultrasonic sensor)

jabrena, 2018-05-04:
It seems a user project not a core library.

Besides, it is necessary to review this issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/494

In latest Debian Stretch snapshot was released a new display driver and it is necessary to check if everything goes well again.

Juan Antonio

closed 2018-05-05 by jabrena